lvgl/lvgl #1381: Switch action not fired when mouse down in switch but mouse up outside of switch

GnUfTw, 2020-02-14:
**Description**

I am experiencing some unexpected behavior when working with the LVGL switch object. The issue is that if one mouses down with the switch in focus, drags the mouse outside of the switch and then releases the mouse; the switch changes state visibly but the registered switch action is not fired.

**To Reproduce**

1. Create a switch and register a action handler that is equivalent to the following:
```c
static lv_res_t foo(lv_obj_t* sw)
{
    if (lv_sw_get_state(sw))
    {
        printf("Switch on\n");
    }
    else
    {
        printf("Switch off\n");
    }
}
```
2. Launch application that contains switch
3. Mouse down inside of switch (keep mouse pressed down until step 5)
4. Drag mouse outside of the switch
5. Release mouse
6. _Bug: Switch changes state visibly but the action handler is never fired. Observe that neither "Switch on" nor "Switch off" appears on stdout._

**Expected behavior**

I would have expected the switches action handler to be fired or the switch to have not changed state.

**Screenshots**
NA

**Additional context**

- LVGL version: 5.2.0
- This behavior can be reproduced using simulator _and_ target hardware
- Updating to a new version of LVGL is not an option for me

embeddedt, 2020-02-14:
I took a look at the code and I think I know where the problem is. You can try this patch. Go into `lv_sw.c`, and add this to the `else if(sign == LV_SIGNAL_PRESS_LOST)` block within `lv_sw_signal`.

https://github.com/littlevgl/lvgl/blob/a9a55321c7fe5159834f417aaff92117a85dddcd/lv_objx/lv_sw.c#L253

GnUfTw, 2020-02-14:
Thanks for the patch, that addresses the issue. For the record, I added that line at the _end_ of the `else if (sign == LV_SIGNAL_PRESS_LOST) block.

closed 2020-02-14 by GnUfTw

embeddedt, 2020-02-14:
Sorry; I forgot to mention the exact location. You put it in the right place. :slightly_smiling_face: