lvgl/lvgl #1384: dev-7.0 new styles visual studio compilation error

barbiani, 2020-02-16:
@kisvegabor it is unbelievable how much is being achieved by littlevgl!

Tonight I pulled the lastest code, recreated a visual studio project with the new examples and the compiler complained about this cast to return '(value_type)'.

Error (active)	E0119	cast to type "lv_color_t" is not allowed lvgl\src\lv_core\lv_obj.h	1066	

I simply removed the cast and it compiled.

```
#define _LV_OBJ_STYLE_SET_GET_DECLARE(prop_name, func_name, value_type, style_type)          \
static inline value_type lv_obj_get_style_##func_name (const lv_obj_t * obj, uint8_t part)  \
{                                                                                           \
    return (value_type) _lv_obj_get_style##style_type (obj, part, LV_STYLE_##prop_name);     \
```

embeddedt, 2020-02-16:
> it is unbelievable how much is being achieved by littlevgl!

Indeed; I hardly recognized the teaser at first. :slightly_smiling_face: 

> I simply removed the cast and it compiled.

Removing the cast causes issues with C++, so I think this has to be solved upstream in a more complicated way.

For now, keep using your workaround, but it shouldn't be needed once #1383 is merged.

kisvegabor, 2020-02-17:
So happy to hear that you like the updates! :smile: 

I've just commented on @embeddedt PR, it can be merged soon.

barbiani, 2020-03-01:
This is with gcc-arm-9.2-2019.12-x86_64-arm-none-eabi (bare metal raspberry pi)

Took [circle](https://github.com/rsta2/circle) and updated the demo to dev-7.0.

[image]

embeddedt, 2020-03-01:
I can fix this one.

embeddedt, 2020-03-01:
The root problem here is that C/C++ don't like it when you implicitly cast a function pointer to/from `void *`. POSIX does mandate that this be required, but the C standard does not. As such, C compilers will output a warning that ISO C doesn't allow this. I think we can safely ignore that/add a pragma for GCC.

C++, on the other hand, outputs an error if you try this without `-fpermissive`. I've tried a whole bunch of combinations but they're either far too messy, or introduce other issues.

@kisvegabor How costly do you think it would it be if we didn't inline these helper functions in `lv_obj_style_dec.h`?

kisvegabor, 2020-03-03:
> How costly do you think it would it be if we didn't inline these helper functions in lv_obj_style_dec.h?

To be honest I don't know for sure. However, without `inline`ing two "lists" should be maintained (both in c and h file).

As it's only related to function pointers I think a local `#if __cplusplus` "hack" would be better. What do think about [this](https://www.kdab.com/how-to-cast-a-function-pointer-to-a-void/) possible solution?

embeddedt, 2020-03-03:
I think that solution on its own is not enough.

> > Even though this is guaranteed to work with all standard-compliant compilers, it’s not guaranteed to be safe on all platforms. On machines with Harvard architecture, the size of a function pointer often differs from the size of a `void*` (e.g. 16bit on the Arduino One). So, especially if you’re developing for micro controllers, double check before using this trick.

According to this article, for portability, we can't treat function pointers as regular pointers. (We can, however, safely [cast a function pointer to another one](https://stackoverflow.com/a/559671) for storage purposes. It must be cast back before being invoked.)

That means we need to introduce another set of style functions for storing and retrieving function pointers, in comparison to regular pointers.

Alternatively, we could come up with a new way of registering functions in styles that doesn't reference the function directly. I suggest this only because it might lend itself to easier serialization for things like GUI builders.

kisvegabor, 2020-03-04:
I see. I'd like to keep the existing 4 base types so I suggest the following. 
Create a type like:
```c
typedef union {
  void(*fptr)(void);
  void * dptr;
}lv_style_fptr_dptr_t;
```

And change the current `..._ptr` functions to accept `lv_style_fptr_dptr_t`.

And finally, create `inline` wrappers like 
```c
static inline void _lv_style_set_fptr(lv_style_t * style, lv_style_property_t prop, void (*p)(void))
{
  lv_style_fptr_dptr_t t;
  t.fptr = p;
  _lv_style_set_fptr(style, prop, t);
}

static inline void _lv_style_set_dptr(lv_style_t * style, lv_style_property_t prop, void *p)
{
  lv_style_fptr_dptr_t t;
  t.dptr = p;
  _lv_style_set_fptr(style, prop, t);
}
```

If you agree, I can add this.

embeddedt, 2020-03-04:
This looks portable to me. Can you also test it in C++ to make sure it works there? 

By the way, I was thinking that this whole issue seems like a good reason to add a C++ compilation test to CI.

kisvegabor, 2020-03-04:
> This looks portable to me.

Great!

> By the way, I was thinking that this whole issue seems like a good reason to add a C++ compilation test to CI.

Stupid question: I've changed `gcc` to `g++` in test's makefile but it compiled without warning. Have I missed something?

kisvegabor, 2020-03-04:
Thank you, now I see the warning. I'll add the mentioned fix tomorrow.

kisvegabor, 2020-03-05:
I've just pushed the mentioned fix. It's super ugly but it works. Later we can find a more elegant solution.

embeddedt, 2020-03-06:
I've tested it and I no longer see any warnings with C++.

@barbiani @fvanroie Is the issue fixed for you as well?

fvanroie, 2020-03-06:
I have removed the `-fpermissive` build flag and updated to the latest dev-7.0 code. Now my *Arduino* project builds without errors.

kisvegabor, 2020-03-06:
Thank you for the feedback. I close this issue then.

closed 2020-03-06 by kisvegabor