lvgl/lvgl #1387: Deleting an active screen  will  crash program

ShenRen, 2020-02-17:
**Describe the bug**

Deleting an active screen which will  crash program.

**To Reproduce**

code:
```c
lv_obj_t * screen = lv_obj_create(NULL,NULL); // create a screen
lv_disp_load_scr(screen );
lv_obj_del(screen );
 // Program will crash when displayer was refreshed.
```

**Expected behavior**

I expect it to activate other screens or default screen.

**Screenshots**
No Screenshots

**Additional context**

I find the code in `lv_obj_del()`.
[ lv_obj.c ](https://github.com/littlevgl/lvgl/blob/master/src/lv_core/lv_obj.c#L476)
```c
lv_res_t lv_obj_del(lv_obj_t * obj)
{
   //other code

    /*Remove the object from parent's children list*/
    lv_obj_t * par = lv_obj_get_parent(obj);
    if(par == NULL) { /*It is a screen*/
        lv_disp_t * d = lv_obj_get_disp(obj);
        lv_ll_rem(&d->scr_ll, obj);
        // There is no check that the screen is active. 
    } else {
        lv_ll_rem(&(par->child_ll), obj);
    }

   //other code
}
```

I think it should be solved like this:

```c
lv_res_t lv_obj_del(lv_obj_t * obj)
{
    //other code
    /*Remove the object from parent's children list*/
    lv_obj_t * par = lv_obj_get_parent(obj);
    if(par == NULL) { /*It is a screen*/
        lv_disp_t * d = lv_obj_get_disp(obj);
        lv_ll_rem(&d->scr_ll, obj);
        /*check if obj is an active screen*/
        if(d->act_scr == obj)
        {
            lv_disp_load_scr(lv_ll_get_head(&d->scr_ll));
        }
    } else {
        lv_ll_rem(&(par->child_ll), obj);
    }
   //other code
}
```

embeddedt, 2020-02-17:
This is expected behavior. You should call `lv_scr_load` with a new screen before you call `lv_obj_del` on the previous one.

We could add an assertion which will prevent it from actually crashing, but I don't think it needs to load a new screen automatically.

BesitzeRuf, 2020-02-17:
The `lv_obj_del(..)` function does what is expected from function, which shall to delete an object. The problem is in any other function (in your case in refresher, which doesn't check if the object is NULL or not.

kisvegabor, 2020-02-18:
I agree with @embeddedt and his suggestion. Its something you should handle manually rather than loading a "random" screen automatically. That wouldn't be deterministic from user's point of view. 

@BesitzeRuf 
> The problem is in any other function (in your case in refresher, which doesn't check if the object is NULL or not.

There is an `ASSERT` mechanism since v6.0 which is really not used during refresh because of performance reasons.

ShenRen, 2020-02-18:
Yes, it's true that random loading a screen is not a good solution. 
But at least remove its active screen status. At the same time, prevent the program from crashing.
```c

lv_res_t lv_obj_del(lv_obj_t * obj)
{
    //other code
    /*Remove the object from parent's children list*/
    lv_obj_t * par = lv_obj_get_parent(obj);
    if(par == NULL) { /*It is a screen*/
        lv_disp_t * d = lv_obj_get_disp(obj);
        lv_ll_rem(&d->scr_ll, obj);
        /*check if obj is an active screen*/
        if(d->act_scr == obj)
        {
            d->act_scr = NULL; //check if the screen is NULL or not on other place.
        }
    } else {
        lv_ll_rem(&(par->child_ll), obj);
    }
   //other code
}

```

kisvegabor, 2020-02-22:
> 
```
 if(d->act_scr == obj)
 {
       d->act_scr = NULL; //check if the screen is NULL or not on other place.
 }
```

I agree with that. Keeping an invalid pointer in `act_scr` is definitely not a good idea.

However, the default screen can be also deleted. So IMO we are fine the setting NULL to `act_scr`.

embeddedt, 2020-03-06:
What is supposed to happen when `d->act_scr` is `NULL`? Do we fill the screen with black, or simply not redraw anything?

kisvegabor, 2020-03-06:
I could imagine a warning or assert `lv_scr_act` or a similar error label to "No data" when the image source can't be found. What do you think?

Anyway, I think we shouldn't overcomplicate this case, because always should be an active screen.

embeddedt, 2020-03-07:
I think the best thing to do is just not redraw anything if there is no active screen. (In fact, this could be a useful feature for saving memory when showing a static picture. Deleting the active screen means a much smaller memory footprint while the static image is in place.)

kisvegabor, 2020-03-08:
Geat idea. Let's do this way!

Implemented in 1f2c772e.

embeddedt, 2020-03-25:
It seems that the original issue is fixed now so I'll close this issue.

closed 2020-03-25 by embeddedt

brightproject, 2024-11-19:
Hello @kisvegabor 
Unfortunately, I did not understand from the correspondence how to eliminate this error and such a manifestation of the event function

> scr_unloaded_delete_cb

I began to have a similar manifestation.
I have three widget screens, between which I switch left and right by pressing the touch screen of the screen.
If the code in the screen file is like this

```
void ui_Screen2_screen_init(void)
{
ui_Screen2 = lv_obj_create(NULL);
lv_obj_clear_flag(ui_Screen2, LV_OBJ_FLAG_SCROLLABLE); /// Flags
}
```
Then the `esp32` does not reboot and the transition from screen to screen is normal.
If the code is like this
```
void ui_Screen2_screen_init(void)
{
ui_Screen2 = lv_obj_create(NULL);
lv_obj_clear_flag(ui_Screen2, LV_OBJ_FLAG_SCROLLABLE); /// Flags
lv_obj_add_event_cb(ui_Screen2, scr_unloaded_delete_cb, LV_EVENT_SCREEN_UNLOADED, &ui_Screen2);
}
```
Sometimes when moving between widget screens, the `esp32` reboots with the following information being output to the serial port:

> [Warn]  (847.798, +64)   lv_obj_get_disp: No screen found       (in lv_obj_tree.c line #281)
>     [Warn]  (847.799, +1)    lv_obj_get_disp: No screen found       (in lv_obj_tree.c line #281)
>     Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 
>     Core  1 register dump:
>     PC      : 0x42058c0a  PS      : 0x00060e30  A0      : 0x82004b14  A1      : 0x3fcebe10
>     A2      : 0x00000000  A3      : 0x00060e20  A4      : 0x00000004  A5      : 0x00060e23
>     A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x109fefff  A9      : 0x3fcebe10
>     A10     : 0x00060e23  A11     : 0x00000000  A12     : 0x00060e20  A13     : 0x00000002
>     A14     : 0x3fcaff28  A15     : 0x00000064  SAR     : 0x00000013  EXCCAUSE: 0x0000001c
>     EXCVADDR: 0x109ff015  LBEG    : 0x40056f08  LEND    : 0x40056f12  LCOUNT  : 0x00000000
> 
> 
>     Backtrace: 0x42058c07:0x3fcebe10 0x42004b11:0x3fcebe30 0x4201039c:0x3fcebe60 0x42002935:0x3fcebe80 0x4202f100:0x3fcebea0 0x4037f6c6:0x3fcebec0
>     #0  0x42058c07 in _lv_obj_get_ext_draw_size at lib/lvgl/src/core/lv_obj_draw.c:385
>     #1  0x42004b11 in lv_obj_invalidate at lib/lvgl/src/core/lv_obj_pos.c:857
>     #2  0x4201039c in lv_img_set_offset_x at lib/lvgl/src/widgets/lv_img.c:170
>     #3  0x42002935 in loop() at src/main.cpp:115
>         (inlined by) loop() at src/main.cpp:76
>     #4  0x4202f100 in loopTask(void*) at .platformio/packages/framework-arduinoespressif32@src-98cb8087edcca5875bec152528484d34/cores/esp32
>     #5  0x4037f6c6 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kerne

I commented out the lines and use it without `scr_unloaded_delete_cb`.
Visually I can't see the difference, it seems that with it and without it the speed of work is the same.
I wanted to understand this issue more fully.

brightproject, 2024-11-25:
> Can you share it?

Yes, no problem, but there is nothing in this function

```
void scr_unloaded_delete_cb(lv_event_t * e)

{

    lv_obj_t ** var = lv_event_get_user_data(e);
    lv_obj_del(*var);
    (*var) = NULL;

}
```

There are very few examples and explanations on this topic.
I simply do not use this technique in my widgets.

commit 1f2c772e (2020-03-08): do not redraw anything if the active screen was deleted. Closes #1387