lvgl/lvgl #1389: Increase image rotation precision

rabbitsaviola, 2020-02-19:
Suggest to increase image rotation precison, e.g. 0.1 degree, such that the second hand on a watch face can be rotated more smoothly.

C47D, 2020-02-19:
OT: rabbitsaviola por el conejito Saviola?

kisvegabor, 2020-02-20:
It was already asked several times which is a clear indication of it really should be improved.

I a good way to add it to the code but not sure about the API. Simply allow 0...3600 deg as input (for 0.1 deg resolution)?

rabbitsaviola, 2020-02-20:
> It was already asked several times which is a clear indication of it really should be improved.
> 
> I see a good way to add it to the code but not sure about the API. Simply allow 0...3600 deg as input (for 0.1 deg resolution)?

0..3600 is ok for me. Our gpu inteface also uses this as input.

kisvegabor, 2020-02-20:
Okay, I'll try it today or tomorrow.

kisvegabor, 2020-02-20:
@rabbitsaviola 
I've just added it. See https://github.com/littlevgl/lvgl/commit/759c4ccb35d053e926e9aa7056614b9755ee2e15

rabbitsaviola, 2020-02-21:
@kisvegabor 
I've just tried it. It works fine. :+1:

kisvegabor, 2020-02-22:
Great!

closed 2020-02-22 by kisvegabor