lvgl/lvgl #1390: Could you provide an offline version of the image conversion tool?

imosts, 2020-02-20:
Hi
I need to dynamically send a BMP picture from the mobile app to the device at runtime, then I need to store it on the file system of the device, and display it on the device. So could you please give me an offline version of the image conversion tool? If not, could you please give me the specific format of the .bin file? In this way, I can convert BMP to .bin on the mobile app.

Thanks very much!

kisvegabor, 2020-02-20:
The offline converter is here: https://github.com/littlevgl/lv_utils/issues/31

But converting the image in the mobile app seems much easier as the converter is in PHP.
The bin format has a 4 byte header [like this](https://github.com/littlevgl/lvgl/blob/master/src/lv_draw/lv_img_decoder.h#L57-L70) followed by the colors of the pixels.

Putting together the header might be tricky and required some error and trial due to endianness differences and bitfields. I suggest converting an image with the online converter and use it as a reference.

You can read about the color formats here: https://docs.littlevgl.com/en/html/overview/image.html

excitedbox, 2020-03-09:
an`t you send an image to your device and do the conversion there as part of your storage process? Seems like that would be the easiest unless you are using a really low power micro.

embeddedt, 2020-03-09:
> Can't you send an image to your device and do the conversion there as part of your storage process?

By device, I assume you mean the embedded system. That wouldn't be straightforward either as you would still have to rewrite the converter in C (the current version uses PHP).

kisvegabor, 2020-03-09:
FYI, here is the documentation of the internal image format:
https://docs.littlevgl.com/en/html/overview/image.html#color-formats

embeddedt, 2020-03-15:
It seems that there hasn't been any activity from the OP for roughly a month so I'll close this issue. If you are still having problems feel free to comment on the issue.

closed 2020-03-15 by embeddedt