lvgl/lvgl #1391: Logic with use of return results in dead code

xennex22, 2020-02-20:
https://forum.littlevgl.com/t/coding-style-use-of-return-in-functions/1731

lv_checkbox.c(168)
lv_cont.c(259)
lv_list.c(630)
lv_win.c(507)
```
if(info->result != NULL) return LV_RES_OK;
    else return ancestor_signal(win, sign, param);
    return LV_RES_OK;
```

lv_bar.c(270)
```
int16_t lv_bar_get_value(const lv_obj_t * bar)
{
    LV_ASSERT_OBJ(bar, LV_OBJX_NAME);

    lv_bar_ext_t * ext = lv_obj_get_ext_attr(bar);
    return ext->cur_value;
    return LV_BAR_GET_ANIM_VALUE(ext->cur_value, ext->cur_value_anim);
}
```

embeddedt, 2020-02-20:
The one in `lv_bar.c` is a mistake; only the second statement should be there.

xennex22, 2020-02-23:
lv_roller.c(774)
```
if(res != LV_RES_OK) return res;
    return LV_RES_OK;
```

kisvegabor, 2020-02-23:
I've just fixed the mentioned issues. Thank you for pointing them out.

(Sorry for the noise with the commits, I desperately tried to fix the typo in the comment message :sweat_smile:)

xennex22, 2020-02-24:
lv_bar.c(610)
```
if(info->result != NULL) return LV_RES_OK;
        else return ancestor_signal(bar, sign, param);
        return LV_RES_OK;
```

kisvegabor, 2020-02-25:
Thanks, I've fixed both of them

xennex22, 2020-02-25:
lv_draw_mask.c(1042)
```
static lv_draw_mask_res_t lv_draw_mask_fade(lv_opa_t * mask_buf, lv_coord_t abs_x, lv_coord_t abs_y, lv_coord_t len, lv_draw_mask_fade_param_t * p)
{
    ...
    if(abs_y <= p->cfg.y_top) {
        ...
        return LV_DRAW_MASK_RES_CHANGED;
    } else if(abs_y >= p->cfg.y_bottom) {
        ...
        return LV_DRAW_MASK_RES_CHANGED;
    } else {
        ...
        return LV_DRAW_MASK_RES_CHANGED;
    }
    return LV_DRAW_MASK_RES_FULL_COVER;  <-- dead
}
```
Suggest either removing the returns in the conditional statements or using a stack variable for the return value.

lv_draw_blend.c(395)
lv_draw_blend.c(665)
```
LV_LOG_WARN("fill_blended: unsupported blend mode");
return;
break; <-- dead
```
I'd just remove the break but I suspect some compilers may not like a switch without breaks.

kisvegabor, 2020-02-26:
I've just run cppcheck and interestingly wasn't reported these. 

Anyway, I've fixed both.

> I'd just remove the break but I suspect some compilers may not like a switch without breaks.

Finally, I've left the `return` only as you suggested. A tend to believe it's not possible to make all the compilers happy at the same time. :smile:

commit 6b9fa980 (2020-02-23): elminate dead con on some returns. Fixes #1391
commit ea5af5cd (2020-02-23): elminate dead code on some returns. Fixes #1391
commit 52451636 (2020-02-23): elminate dead code on some returns. Fixes #1391
closed 2020-02-27 by xennex22