lvgl/lvgl #1393: Focusing next object in a group made of imgbtn objects corrupts frame buffer

simonemike95, 2020-02-27:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

When a screen is open, calling any lv_group_focus functions (e.g. lv_group_focus_next) will corrupt the frame buffer and render incorrect screens.

**To Reproduce**

```
lv_group_t *home_button_group;

void create_btns(lv_obj_t *parent) {
    // Default icons (black background)
    LV_IMG_DECLARE(imgbtn_download_ic);
    LV_IMG_DECLARE(imgbtn_recorder_ic);

    // Create buttons
    lv_obj_t *downloads_btn = lv_imgbtn_create(parent, NULL);
    lv_imgbtn_set_toggle(downloads_btn, true);
    lv_imgbtn_set_src(downloads_btn, LV_BTN_STATE_REL, &imgbtn_download_ic);
    lv_imgbtn_set_src(downloads_btn, LV_BTN_STATE_TGL_PR, &imgbtn_download_ic);
    lv_imgbtn_set_src(downloads_btn, LV_BTN_STATE_PR, &imgbtn_download_ic);
    lv_imgbtn_set_src(downloads_btn, LV_BTN_STATE_TGL_REL, &imgbtn_download_ic);

    lv_obj_t *recorder_btn = lv_imgbtn_create(parent, NULL);
    lv_imgbtn_set_toggle(recorder_btn, true);
    lv_imgbtn_set_src(recorder_btn, LV_BTN_STATE_REL, &imgbtn_recorder_ic);
    lv_imgbtn_set_src(recorder_btn, LV_BTN_STATE_TGL_PR, &imgbtn_recorder_ic);
    lv_imgbtn_set_src(recorder_btn, LV_BTN_STATE_PR, &imgbtn_recorder_ic);
    lv_imgbtn_set_src(recorder_btn, LV_BTN_STATE_TGL_REL, &imgbtn_recorder_ic);

   // in between here, position is set with x and y coordinates

    // Create group
    button_group = lv_group_create();

    // Add all our buttons to the group
    lv_group_add_obj(button_group, downloads_btn);
    lv_group_add_obj(button_group, recorder_btn);

    lv_task_create(task, 2000, LV_TASK_PRIO_MID, &admin_btn);
}

static void task(lv_task_t *task) {
    lv_group_focus_next(button_group);
}
```

**Expected behavior**

The next button in the group should be highlighted.

**Screenshots**
Cannot upload.

**Additional context**

This is on an embedded Linux platform running on ARM, but very reproducible for me. There is a lot of custom hardware involved so if it is not reproducible for others that could be why. This did not happen on my PC simulator version.

simonemike95, 2020-02-27:
Similar to issue #630 but entire screen scrambles similar to how the button does for them.

simonemike95, 2020-02-27:
Workaround that worked in issue #630 works here as well. Call lv_scr_load(lv_scr_act()) and the screen will not be incorrect.

closed 2020-02-27 by simonemike95

embeddedt, 2020-03-07:
This is a very strange issue, particularly since it doesn't seem to appear for everyone/all the time.

I believe the focusing logic has changed/will change in 7.0 so maybe the problem will be resolved then.