lvgl/lvgl #1396: Build Errors using most recent dev-7.0 source

fvanroie, 2020-03-02:
**Describe the bug**

I'm using the dev-7.0 to compile lvlg for ESP32 in platformio using the arduino toolchain.

Using the most recent dev-7.0 branch to build a test version, I get these errors:
```
In file included from lib\lvgl/src/lv_core/lv_obj.h:1050:0,
                 from lib\lvgl/lvgl.h:24,
                 from src\hasp.cpp:8:
lib\lvgl/src/lv_core/lv_obj_style_dec.h: In function 'void lv_obj_set_style_local_transition_path(lv_obj_t*, uint8_t, lv_state_t, lv_anim_path_cb_t)':
lib\lvgl/src/lv_core/lv_obj_style_dec.h:78:116: error: invalid conversion from 'lv_anim_path_cb_t {aka short int (*)(const _lv_anim_t*)}' to 'const void*' [-fpermissive]
         _lv_obj_set_style_local##style_type (obj, part, LV_STYLE_##prop_name | (state << LV_STYLE_STATE_POS), value);         \
                                                                                                                    ^
lib\lvgl/src/lv_core/lv_obj_style_dec.h:177:1: note: in expansion of macro '_LV_OBJ_STYLE_SET_GET_DECLARE'
 _LV_OBJ_STYLE_SET_GET_DECLARE(TRANSITION_PATH, transition_path, lv_anim_path_cb_t, _ptr, scalar)
 ^
In file included from lib\lvgl/lvgl.h:24:0,
                 from src\hasp.cpp:8:
lib\lvgl/src/lv_core/lv_obj.h:573:6: error:   initializing argument 4 of 'void _lv_obj_set_style_local_ptr(lv_obj_t*, uint8_t, lv_style_property_t, const void*)' [-fpermissive]
 void _lv_obj_set_style_local_ptr(lv_obj_t * obj, uint8_t type, lv_style_property_t prop, const void * value);
      ^
In file included from lib\lvgl/src/lv_core/lv_obj.h:1050:0,
Compiling .pio\build\esp12e-st7735\src\hasp_eeprom.cpp.o
                 from lib\lvgl/lvgl.h:24,
                 from src\hasp.cpp:8:
lib\lvgl/src/lv_core/lv_obj_style_dec.h: In function 'void lv_style_set_transition_path(lv_style_t*, lv_state_t, lv_anim_path_cb_t)':
lib\lvgl/src/lv_core/lv_obj_style_dec.h:94:102: error: invalid conversion from 'lv_anim_path_cb_t {aka short int (*)(const _lv_anim_t*)}' to 'const void*' [-fpermissive]
         _lv_style_set##style_type (style, LV_STYLE_##prop_name | (state << LV_STYLE_STATE_POS), value);                 \
                                                                                                      ^
lib\lvgl/src/lv_core/lv_obj_style_dec.h:177:1: note: in expansion of macro '_LV_OBJ_STYLE_SET_GET_DECLARE'
 _LV_OBJ_STYLE_SET_GET_DECLARE(TRANSITION_PATH, transition_path, lv_anim_path_cb_t, _ptr, scalar)
 ^
In file included from lib\lvgl/src/lv_core/lv_obj.h:20:0,
                 from lib\lvgl/lvgl.h:24,
                 from src\hasp.cpp:8:
lib\lvgl/src/lv_core/lv_style.h:360:6: error:   initializing argument 3 of 'void _lv_style_set_ptr(lv_style_t*, lv_style_property_t, const void*)' [-fpermissive]
 void _lv_style_set_ptr(lv_style_t * style, lv_style_property_t prop, const void * p);
      ^
```

**To Reproduce**

- Compiling succeeds with commit 914383eea352e2eca2dbb0098e2f71c992e669ed from 2 weeks ago in the dev-7.0  branch.
- Later commits fail to build in lv_core/lv_obj_style_dec.h

**Expected behavior**

To be able to build the development version of dev-7.0 for testing the new features.

**Screenshots**
[image]

Using commit 914383eea352e2eca2dbb0098e2f71c992e669ed works

**Additional context**

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_d32_pro.html
PLATFORM: Espressif 32 1.11.2 > WEMOS LOLIN D32 PRO
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
```

I understand this is a development branch, so hopefully this bug report helps to fix the build issues I experience. I would like to test the new styles features of lvgl dev7.0.

If needed, I can provide additional information or help test on the ESP platform.

embeddedt, 2020-03-03:
Thanks for volunteering to test `dev-7.0`! Much appreciated!

I'm working on the issue with C++ and the new style system in #1384. To keep the discussion in one place, is it alright if we close this issue?

fvanroie, 2020-03-03:
Thanks for the quick reply. I didn't realize there was already a thread for this issue in Visual Studio.
I have subscribed to issue #1384 and can follow the discussion on that thread. No problem to close this one!

fvanroie, 2020-03-03:
Duplicate of issue #1384

closed 2020-03-03 by fvanroie