lvgl/lvgl #1399: [dev 7.0] linemeter scale drawing incorrect with big sizelinemeter

xennex22, 2020-03-04:
If you draw a large linemeter (or gauge) then the scale drawing gets messed up. The critical width seemes about 278.
```
lv_obj_t * line = lv_linemeter_create(lv_scr_act(), NULL);
lv_obj_set_size(line, 278, 278);
lv_obj_align(line, lv_scr_act(), LV_ALIGN_CENTER, 0, 0);
```

The cause is an overflow in the x_out / x_in calculation.  If the angle is 90, then lv_trigo_sin(90) = 0x7FFF which is then multiplied by the radius and then right shifted by LV_TRIGO_SHIFT - 8 (15-8 = 7).  So it will overflow when the radius gets to 128 when using lv_coord_t of int16_t .  

If I change the lv_coord_t typedef from int16_t to int32_t the LV_COORD_MAX define gets messed up and chopped down to -1000.  I think this is another bug that can be dealt with in another issue.

I propose changing the fixed point rounding value (I think that is what it is) from 8 to something less.  The maximum radius is going to be 32768 >> rounding_shift.  So 4 would give a maximum radius of 2048.  5 = 1024. etc.  I'm not sure of how much precision is desirable in the rounding calc so probably the best thing is to make it a define, and default it to 6, which should be good for everyone using a 1024 pixel wide screen or less and then drawing a full screen gauge.  That would give 1/64 pixel fixed point rather than 1/256 pixel.

kisvegabor, 2020-03-04:
Hi,

Thank you for pointing it out.

I've fixed it by using `int32_t` for `in/out` variables.

Once I was there I've reduced the direction error of the lines too.
See e1b188f7

kisvegabor, 2020-03-04:
I've also fixed the `lv_coord_t int32_t` issue.

kisvegabor, 2020-03-04:
I've added further improvements to the precision:

Before:
[image]

After:
[image]

xennex22, 2020-03-04:
Great - problem is fixed.

Next thing I am going to look at is using a float for the gauge, or possibly using a scale factor.  eg range 80 - 180 with scale of 10 shows 8-18 and then value can have 0.1 unit precision.  That would mean the linemeter could stay the same.  I'll try each way and see what works best.

kisvegabor, 2020-03-04:
Okay, great!

I'll close this issue then.

closed 2020-03-04 by kisvegabor