lvgl/lvgl #1400: lv_mem_monitor can crash with DIV0 exception

fvanroie, 2020-03-04:
**Describe the bug**

I was debugging my app crashing at startup when implementing `lv_mem_monitor` into my logging function. It turns out this was because `lv_mem_monitor` throws a *Divide by 0 exception*, when called before `lv_init`and thus before `lv_mem_init`.

**To Reproduce**

1. Call `lv_mem_monitor` before `lv_init`.

*or*

2. There also seems to exist an edge case where the actual value of `mon_p->free_size` can be exactly zero if **all** the entries are in use:

```
    lv_init();

    // Check initial free_size:
    lv_mem_monitor_t mem_mon;
    lv_mem_monitor(&mem_mon);
    printf("initial free_size = %u\n", mem_mon.free_size);

    // Use all the available memory, free_size is now zero:
    void * test = lv_mem_alloc(mem_mon.free_size);
    lv_mem_monitor(&mem_mon);
    printf("actual free_size = %u\n", mem_mon.free_size);
    lv_mem_free(test);
```
returns
```
initial free_size = 12208
actual free_size = 0
```
The second call to `lv_mem_monitor` crashes too if this edge case is not handled.

**Expected behavior**

Gracefully return even when  `mon_p->free_size` is zero:

https://github.com/littlevgl/lvgl/blob/7f0677917972c0c1782fa66dd641b91de8d1b300/src/lv_misc/lv_mem.c#L359-L362

Could add a check like:
```
    mon_p->total_size = LV_MEM_SIZE;
    mon_p->used_pct   = 100 - (100U * mon_p->free_size) / mon_p->total_size;
    if(mon_p->free_size != 0) {
        mon_p->frag_pct = (uint32_t)mon_p->free_biggest_size * 100U / mon_p->free_size;
        mon_p->frag_pct = 100 - mon_p->frag_pct;
    }
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

Adding a check on `mon_p->free_size != 0` before the division solved my issue.

I know that using `lv_mem_monitor` before `lv_init` is definitely not wise, and I considered implementing a flag to check it in my app. But looking at the source code I see there is also another rare edge case where there is a risk of a divide by zero.

It seems best to handle this exception gracefully inside the function.

embeddedt, 2020-03-04:
Thanks for the report!

closed 2020-03-04 by embeddedt

fvanroie, 2020-03-05:
Thank you for the fast response.

commit c5139775 (2020-03-04): Fix #1400: divide-by-zero error in `lv_mem_monitor`