lvgl/lvgl #1402: possible NULL dereference

tluio, 2020-03-06:
**Describe the bug**
File: lvgl/src/lv_draw/lv_img_decoder.c
Function: lv_img_decoder_built_in_open
On line 391, **user_data->palette** may be NULL, but on line 408 and 421, **user_data->palette** is dereferenced. So does **user_data->opa**. I guess LV_LOG_ERROR on line 394 does nothing but outputs the log. But I am not sure if **palette_size** must be 0 when **user_data->palette** is NULL.

**To Reproduce**

Sorry, I used a static code analysis tool so I don't known how to reproduce it.

**Expected behavior**

Sorry, but I don't known what to expect.

embeddedt, 2020-03-06:
Thanks for reporting this. There is definitely a bug here.

In debug mode, `LV_ASSERT_MEM` will stop execution if these pointers are NULL. However, this doesn't happen in release mode.

@kisvegabor What do you think we should do here? One solution is to add `return LV_RES_INV` statements after all the `LV_ASSERT_MEM` calls. Another is to always enable `LV_ASSERT_MEM`, but only have it log in debug mode.

tluio, 2020-03-06:
> Thanks for reporting this. There is definitely a bug here.
>
> In debug mode, `LV_ASSERT_MEM` will stop execution if these pointers are NULL. However, this doesn't happen in release mode.
>
> @kisvegabor [...]

Sorry, I have found there was a bug but I don't know how to fix it because I am not very familiar with the code.

kisvegabor, 2020-03-06:
@tluio thank you for the report. May I ask which static code analyzer have you used? I've just run cppcheck a few days ago but it doesn't show a warning for it.

Anyway, it's really a bug. So far, I returned after every failed `LV_MEM_ASSERT` but seemingly I've forgotten to do so here.

@embeddedt 
Do you agree to simply add the missing returns here and close the open files, free other resources etc if any?

embeddedt, 2020-03-06:
Yes; that should fix it.

tluio, 2020-03-07:
> @tluio thank you for the report. May I ask which static code analyzer have you used? I've just run cppcheck a few days ago but it doesn't show a warning for it.
>
> Anyway, it's really a bug. So far, I [...]

You're welcome. I used a static code analyzer written by myself.

kisvegabor, 2020-03-08:
I've fixed it here 8880326a
Can you confirm if it's ok now?

> I used a static code analyzer written by myself.

Great work! :)

tluio, 2020-03-08:
> Great work! :)
Thank you! But there are still many limits such as excessive memory usage which makes it unscalable. Now, I am trying to optimize it.

kisvegabor, 2020-03-10:
I've just tried Facebook's linter: https://github.com/facebook/infer  
You might be also interested in it. It showed some other NULL dereference issues.

I didn't know that Facebook has such useful open source projects. In a project, we just used their [compressor library](https://github.com/facebook/zstd).

Anyway, the original issue seems to be solved, so I close it now.

closed 2020-03-10 by kisvegabor

tluio, 2020-03-10:
> I've just tried Facebook's linter: https://github.com/facebook/infer
> You might be also interested in it. It showed some other NULL dereference issues.

Thank you very much.