lvgl/lvgl #1403: How to port display function with LDTC controller?

imliubo, 2020-03-07:
Hi,
I try port the display function(disp_flush),I referenced this [project](https://github.com/littlevgl/lv_port_stm32f746_disco_sw4stm32), but the display is incorrect, but I don't know what went wrong,I will show you a video!
[Vedio Link](https://makingfun.oss-cn-qingdao.aliyuncs.com/STM32/littleVGL_STM32_LTDC_DMA2D.mp4)

Here is my disp_flush function:
```c
static void disp_flush(lv_disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p)
{
    /*The most simple case (but also the slowest) to put all pixels to the screen one-by-one*/
		
	SCB_InvalidateICache();

	SCB_CleanInvalidateDCache();

	Copy_Buffer((const uint32_t *)color_p, my_fb, area->x1, area->y1, lv_area_get_width(area), lv_area_get_height(area));

    /* IMPORTANT!!!
     * Inform the graphics library that you are ready with the flushing*/
  lv_disp_flush_ready(disp_drv);
}
```
And the Copy_Buffer function:
```C
void Copy_Buffer(const uint32_t *pSrc, uint32_t *pDst, uint16_t x, uint16_t y, uint32_t xSize, uint32_t ySize) 
{
	uint32_t x_address = (hLtdcHandler.LayerCfg[ActiveLayer].FBStartAdress) + 2*(BSP_LCD_GetXSize()*y + x);
	uint32_t source      = (uint32_t)pSrc;
	
	/* Register to memory mode with ARGB8888 as color Mode */ 
  hDma2dHandler.Init.Mode         = DMA2D_R2M;
  if(hLtdcHandler.LayerCfg[ActiveLayer].PixelFormat == LTDC_PIXEL_FORMAT_RGB565)
  { /* RGB565 format */ 
    hDma2dHandler.Init.ColorMode    = DMA2D_RGB565;
  }
  else
  { /* ARGB8888 format */
    hDma2dHandler.Init.ColorMode    = DMA2D_ARGB8888;
  }
  hDma2dHandler.Init.OutputOffset = BSP_LCD_GetXSize() - xSize;      
  
  hDma2dHandler.Instance = DMA2D;
  
  /* DMA2D Initialization */
  if(HAL_DMA2D_Init(&hDma2dHandler) == HAL_OK) 
  {
    if(HAL_DMA2D_ConfigLayer(&hDma2dHandler, ActiveLayer) == HAL_OK) 
    {
      if (HAL_DMA2D_Start(&hDma2dHandler, source, x_address, xSize, ySize) == HAL_OK)
      {
        /* Polling For DMA transfer */  
        HAL_DMA2D_PollForTransfer(&hDma2dHandler, 10);
      }
    }
  } 
}
```
I'm not sure if something is missing, but it looks like the controls are all created, but the colors are not quite right.

I'm use this [screen](http://www.waveshare.net/shop/7inch-Capacitive-Touch-LCD-F.htm) and this [board](http://www.waveshare.net/shop/Core746I.htm)

I am looking forward to your solution or suggestions.

Have a nice day!

embeddedt, 2020-03-07:
If you are looking to reuse the `flush_cb` function from that STM32 project, I think you are looking for this piece of code.

https://github.com/littlevgl/lv_port_stm32f746_disco_sw4stm32/blob/fe9f47cefadbafb292f0c3a58f848ffc3e165aa9/hal_stm_lvgl/tft/tft.c#L187-L231

imliubo, 2020-03-08:
Hi @embeddedt,
I solved this problem with a very inefficient method,but it works for me.
```C
    int32_t x;
    int32_t y;
    for(y = area->y1; y <= area->y2; y++) {
        for(x = area->x1; x <= area->x2; x++) {
            /* Put a pixel to the display. For example: */
            /* put_px(x, y, *color_p)*/
	   BSP_LCD_PutPixel(x, y, color_p->full);
            color_p++;
        }
    }
```
I hope that I can continue to study it in the future. Now use this method. thank you for your help.

Have a nice day!

closed 2020-03-08 by imliubo