lvgl/lvgl #1404: lv_roller animation seems wrong

Bozog, 2020-03-07:
When updating the roller index with lv_roller_set_selected with the mode set to LV_ROLLER_MODE_INIFINITE the animation doesn't seem correct, it looks like the roller spins all the way round when either incrementing or decrementing the index. Works fine with the mode set to LV_ROLLER_MODE_NORMAL

Its is pretty hard to see with a small roller, but becomes much more apparent with a larger one.

Thanks

```c
lv_obj_t *scr = lv_cont_create(NULL, NULL);
lv_disp_load_scr(scr); 

lv_obj_t *roller = lv_roller_create(scr, NULL);
lv_obj_align(roller, scr, LV_ALIGN_CENTER, 0, 0);
lv_roller_set_options(roller,"00\n01\n02\n03\n04\n05\n06\n07\n08\n09", LV_ROLLER_MODE_INIFINITE);
int count = 9;

while(1) 
{
	for(int i=0;i<200;i++)
	{
		lv_tick_inc(5);
		lv_task_handler();
		usleep(5000);
	}
	if(++count > 9) count = 0;
	lv_roller_set_selected(roller, count, true);
}

embeddedt, 2020-03-07:
I think you have to increment the count to `10` for it to wrap around the way you want. Otherwise it has no way of knowing which way to go.

embeddedt, 2020-03-07:
Then you should be able to set `count = lv_roller_get_selected(roller)` and it will return the normalized value.

kisvegabor, 2020-03-08:
It's a little bit tricky and really not so intuitive. The infinite roller is emulated by adding the options 3 times on the top and 3 times on the bottom of the original options list. When you roll the roller you roll to the extra options and when the dragging stops the position is normalized to the middle option list. It's the same when you set an option manually. 

So is you have 10 options the possible IDs are in [0,69] range and after initialization, you stand on ID=30.

To see this try this code. You would expect to step only 2, but it will scroll a lot (from 30 to 2):
```c
    lv_roller_set_anim_time(roller1, 8000);
    lv_roller_set_selected(roller1, 2, LV_ANIM_ON);
```

I've fixed it `dev-7.0` by interpreting the new ID relative to the current "page".

Bozog, 2020-03-08:
Awesome, just what I was looking for, a trivial matter to make it work correctly then.
Thanks for the replies, and thanks for all your hard work ;)

kisvegabor, 2020-03-10:
Great! Thank you for the report! :)

commit f2d67bd4 (2020-03-08): roller: fix lv_roller_set_selected in INFINITE moed. Fixes #1404
closed 2020-03-10 by kisvegabor