lvgl/lvgl #1405: GPU示例的错误，GPU  BUG

970205011, 2020-03-10:
当我使用
`#define LV_USE_GPU              1`
时候，而且没有修改
`static void gpu_fill(lv_disp_drv_t *disp_drv, lv_color_t *dest_buf, lv_coord_t dest_width, const lv_area_t *fill_area, lv_color_t color)`
函数，出现了这个问题
[image]
我知道，我应该自己写`gpu_fill`代码。但是我认为这是一个BUG。
当我把程序
```
/*It's an example code which should be done by your GPU*/
int32_t x, y;
dest_buf += dest_width * fill_area->y1; /*Go to the first line*/
for (y = fill_area->y1; y < fill_area->y2; y++)
{
for (x = fill_area->x1; x < fill_area->x2; x++)
{
dest_buf[x] = color;
}
dest_buf += dest_width; /*Go to the next line*/
}
```
修改为
```
/*It's an example code which should be done by your GPU*/
int32_t x, y;
dest_buf += dest_width * fill_area->y1; /*Go to the first line*/
for (y = fill_area->y1; y <= fill_area->y2; y++)
{
for (x = fill_area->x1; x <= fill_area->x2; x++)
{
dest_buf[x] = color;
}
dest_buf += dest_width; /*Go to the next line*/
}
```
的时候，错误消失了，正常显示为
[image]
，希望可以帮助到你。

barbiani, 2020-03-10:
Is this a driver you downloaded from the the project repositories? For which chip?

970205011, 2020-03-10:
是的，昨天刚下载的，STM32F407IG

970205011, 2020-03-10:
> 
> 
> Is this a driver you downloaded from the the project repositories? For which chip?

是的，昨天刚下载的，STM32F407IG

kisvegabor, 2020-03-10:
Thank for the report. You are right,  I've just fixed it: https://github.com/littlevgl/lvgl/commit/c26635ed3ef609e03828a477a8c59278085d601f

(Next time, please comment in English, to make it easier to understand for other)

closed 2020-03-10 by kisvegabor