lvgl/lvgl #1406: lv_bar: wired condition

kisvegabor, 2020-03-10:
**Describe the bug**
I've noticed that the bar and slider animation doesn't work properly in some cases. 

It seems [this line](https://github.com/littlevgl/lvgl/blob/5ad1cb76c2d7a0cc0d17138589bc0833208aa883/src/lv_widgets/lv_bar.c#L491) causes the issue which was added in https://github.com/littlevgl/lvgl/commit/59fd1ad5f299173a2fbe4ab4670ea5157161b477 by @embeddedt.

I think `anim_start_value_start_x` should be always added to `anim_start_value_x` to place it relative to the current animation range.

I've tried to simply comment out the condition and it fixed the slider animation issue and lv_switch still works fine.

@embeddedt , what do you think?

**To Reproduce**
```c
    g = lv_group_create();

    lv_indev_drv_t kb_drv;
    lv_indev_drv_init(&kb_drv);
    kb_drv.type = LV_INDEV_TYPE_KEYPAD;
    kb_drv.read_cb = keyboard_read;
    lv_indev_t * kb_indev = lv_indev_drv_register(&kb_drv);
    lv_indev_set_group(kb_indev, g);

   lv_obj_t * slider = lv_slider_create(parent, NULL);
   lv_group_add_obj(g, slider);
```

Use the arrows to change the slider's value.

[image]

embeddedt, 2020-03-10:
There is an issue with the animation from a higher to lower value which that condition addressed. I will take a look.

embeddedt, 2020-03-14:
It should be fixed now. Sorry for the delay!

closed 2020-03-14 by embeddedt

kisvegabor, 2020-03-16:
I've just tested and it's working here too.

Thank you very much!

commit 82d4c8d6 (2020-03-14): lv_bar: fix #1406 (animation issue)