lvgl/lvgl #1410: use 2 framebuffers

RRPRIME, 2020-03-17:
Is there a way to use 2 framebuffers?

embeddedt, 2020-03-17:
Have a look at the [display buffer documentation](https://docs.littlevgl.com/en/html/porting/display.html#display-buffer).

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2020-03-17 by embeddedt

RRPRIME, 2020-03-17:
hey @embeddedt not really helpful  I want to use 2 framebuffers not use 2 buffers for the display. I literally want to init /dev/fb0 and /dev/fb1 and then draw on the 2 screens.

C47D, 2020-03-17:
Maybe this (https://forum.littlevgl.com/t/linux-drm-raspberry-pi4b/1845) can help? I haven't used lvgl on linux based machines tho.

embeddedt, 2020-03-17:
Ah, in that case you would need to tweak the `fbdev` driver to handle multiple frame buffers. Then you can just register a second display with LittlevGL.

RRPRIME, 2020-03-17:
hey, nope, I am working on the RPI zero I do have a /dev/fb0 which is the original output and a tft display on fb1 and due to some limitations of some sofware I need to draw and both framebuffers

embeddedt, 2020-03-17:
I think what I suggested above will do what you want.

kisvegabor, 2020-03-19:
As @embeddedt mentioned you can treat the 2 frame buffers as 2 displays. So LittlevGL multi-display support could be used here: https://docs.littlevgl.com/en/html/overview/display.html#