lvgl/lvgl #1411: [dev7.0] gauge formatting function

xennex22, 2020-03-17:
https://github.com/littlevgl/lvgl/issues/1243#issuecomment-594320473

I have not done a pull request as there definitely are going to be changes and the code will not compile without some more changes.
https://github.com/xennex22/lvgl/tree/dev-7.0

The main change is to add a formatting function as outlined
https://github.com/littlevgl/lvgl/issues/1243#issuecomment-594599836

I added a function `lv_utils_num_to_str_fixed` in `lv_utils.c` to write fixed point values.  This works but has the decimal separator hard coded.  `localeconv()->decimal_point[0]` works on my compiler but I think we want a better solution.
Options:
- code the decimal separator in the `lv_conf.h`, but then people might wish to change it on the fly.
- add a callback `lv_utils_num_to_str_fixed`, but I think this is a little complex.
- add a global to `lv_utils.c` that the user must define in their own code.  Perhaps in combination with a default from `lv_conf.h` that can be overwritten.

Example
```
void gauge_format_fn(lv_obj_t * gauge, char buf[], int bufsize, lv_gauge_value_t value)
{
    int decimals = (int)gauge->user_data;
    lv_utils_num_to_str_fixed(value, decimals, buf);
}
```
The format function passes the buffer size.  It is a small addition that can be used to prevent buffer overruns.  Note that I don't in the example above - perhaps `lv_utils_num_to_str_fixed` should take the buffer size as a parameter for consistency.

To use fixed point in a gauge with floating point you have to calculate a multiplier
```
    dec_power = 1;
    for(int i=0; i<decimals; i++)
        dec_power *= 10;
```
Then `lv_gauge_set_range` `lv_gauge_set_value` `lv_gauge_set_critical_value` can take your floating point value multiplied by the base 10 exponent.
```
lv_gauge_set_critical_value(gauge, floating_value * dec_power);
```
If you are using your own fixed point type then you don't need to bother with the multiplication and your formatting function would need to know

Now with a large number of decimal places the gauge value & range type overflows, so I suggest changing them to a typedef in `lv_conv.h`
```
typedef int16_t lv_gauge_value_t;
```
Note that I have implemented this in the format function already.  I don't really see a downside to adding the typedef.
I did the same typedef for the linemeter as well but it is not critical.
```
typedef int32_t lv_linemeter_value_t;
```
So:
- work out how to handle the decimal separator
- determine if we want to add the `lv_gauge_value_t` and `lv_linemeter_value_t` typedefs

embeddedt, 2020-03-17:
Would you mind opening the PR but [marking it as a draft](https://github.blog/2019-02-14-introducing-draft-pull-requests/#tag-your-work-in-progress)? It makes it easier to see the changes and comment on them.

xennex22, 2020-03-24:
PR merged, thanks for this.

closed 2020-03-24 by xennex22

kisvegabor, 2020-03-25:
Thank you too! :slightly_smiling_face: