lvgl/lvgl #1414: hue shift on PNG loads

RRPRIME, 2020-03-20:
Hello all,
did anyone notice the strange hue shift on the png loads ?
https://blog.littlevgl.com/2018-10-05/png_converter

I used both:
https://github.com/lvandeve/lodepng
and 
https://github.com/littlevgl/lv_lib_lodepng

so the PNG is of 4 dices, the one at the bottom is yellow, when rendered it is cyan. I suspect this is related to the BGR format, but since I initialized my fb correctly with BGR, I suspect that somewhere in the code the images are treated as BGR (to render properly) and when the device is initialized properly instead of rendering RGB they still render RGB thus the hue shifts.

Event the official blog post is showing this issue.

RRPRIME, 2020-03-20:
aha, for the PNG fix:

./lvgl/src/lv_misc/lv_color.h

typedef union
{
    struct
    {
        uint8_t blue;
        uint8_t green;
        uint8_t red;
        uint8_t alpha;
    } ch;
    uint32_t full;
} lv_color32_t;

and a bunch of other structs are BGR.

to fix the hue shift just change the order

typedef union
{
    struct
    {
        uint8_t red;
        uint8_t green;
        uint8_t blue;
        uint8_t alpha;
    } ch;
    uint32_t full;
} lv_color32_t;

refactor should be done here because of the BGR default assumption, this means that this will mess all the renders and a new LV_COLOR_BGR should be added to lv_conf to treat the displays that really are BGR.

embeddedt, 2020-03-20:
I think you are expected to handle this within your display driver.

RRPRIME, 2020-03-20:
not really, all colors should be RGB default then you should handle the BGR in your display or wherever you want.

Assuming all displays are BGR by default is illogic and the official blog post proves it.

embeddedt, 2020-03-20:
Let's ask @kisvegabor how he thinks this should be handled.

RRPRIME, 2020-03-20:
look, this is not my library and don't care that much, I found my fix, keep it the way it is, but it seems really interesting since Gabor did the png load blog post and did not notice the BGR/RGB issue on the pc sim. almost all image formats are RGB and you expect that if you configured properly your display that you should get the proper image displayed, and you shouldn't have to dig around the code to find the issue, at least fix the fb and monitor to make the BGR to RGB conversion.

I get that this is MCU oriented hence the BGR but at least treat the rest properly.

@embeddedt - just my 2 cents, you are really hostile in your approach, polish those soft skills, same aggression as the using 2 framebuffers issue I posted earlier that you closed.

closed 2020-03-20 by RRPRIME

RRPRIME, 2020-03-20:
@embeddedt 
"By default the LittlevGL assumes RGB order, however it can ba swapped by setting LV_SUBPX_BGR  1 in lv_conf.h." - HAHAHAHAHAH

@embeddedt - slow clap, really nice my friend. See I knew there was something I read, the fonts are treated as normal RGB with a BRG fix as the display should. 

My advice to you, stop moderating and start reading the documentation.

embeddedt, 2020-03-21:
@RRPRIME I'm sorry; my goal was not to come across as being hostile or demeaning. Sorry if that's the impression you got. I do try to keep up-to-date with everything but sometimes I forget about a feature/implementation detail.

By the way, the only reason your other issue got closed is because it was a question, which (per [project policy](https://github.com/littlevgl/lvgl/blob/master/docs/CONTRIBUTING.md#where-can-i-write-my-question-and-remarks)) should be directed to the forum and not to our issue tracker.

kisvegabor, 2020-03-24:
@RRPRIME 
@embeddedt is clearly the greatest contributor of LittlevGL. He helps a lot of people here and in the forum who are very grateful for him.  Fortunately, our user base is growing and therefore the number of questions is also growing. We do the support and development in our free time and not always have time to give long, super kind answers. 

Besides, you can't expect anybody (including me) to know all the features or immediately see the connection between a question and a seemingly unrelated feature. (It's really unrelated, I'll explain below)

Finally, it seems you are the one who should improve those soft skills. Telling the most active contributor what he should do (*"stop moderating and start reading the documentation"*) - after open 2 issues - is really rude and we don't welcome this style here.

--- 

Let's talk about the original question hopefully in a more pleasant style.

> may the people struggle with the BGR !!!

I don't understand this. Are the other colors correct in your display or is it any issue only for PNG images?

> "By default the LittlevGL assumes RGB order, however it can ba swapped by setting LV_SUBPX_BGR 1 in lv_conf.h."

It's related only to fonts. Subpixel rendered fonts are 3 times wider and the color of every color channel (R,G,B) is described. `LV_SUBPX_BGR` can be used if the order of the color channels physically inside a pixel is in BGR order. It might be the same or different from in which order the display needs the color information. (E.g. after rotating the by 180° the physical order of color channels is swapped but the display still needs the pixels in the original format.).

GISED-Link, 2020-06-03:
Hi,

I'm using lvgl 6.1 and freeRTOS on a LPC1788 (olimex MOD-LCD4.3" board). I use the 2 full size buffer method for the display driver and a touche pad. 

Everything is looking really nice and working greate. But... my LCD (or LCD peripheral ?) is BGR and not RGB. I searched how to change that in the lcd peripheral but didn't found anything. Olimex just deal with it by swaping red and blue during pixel writing. But I cannot use the write_pixel callback with the 2 full size buffer.

I have tried to add `LV_SUBPX_BGR 1` in the lv_conf.h file, but it do not the trick. Maybe it is only available on version 7.x of lvgl ?

Thank's for the work anyway 😃

kisvegabor, 2020-06-04:
Is it working if you swap `red` and `green` in [lv_colot_t](https://github.com/lvgl/lvgl/blob/master/src/lv_misc/lv_color.h#L215) and the relates [LV_COLOR_SET/GET](https://github.com/lvgl/lvgl/blob/master/src/lv_misc/lv_color.h#L158) macros?

What is your color depth? I can help to make it work and even add this as a config option in  lv_conf.h``

GISED-Link, 2020-06-04:
Thank you for the quick reply. My color depth is 32 bit. 

I've tried to swap red and blue in lv_colot_t but it does not the trick (certainly because I use 32 bit depth). The macro LV_COLOR_SET/GET only get the channel `.red` and `.blue` so I've not tried to swap them.

I've joined my lv_conf.h and my porting in case you need more information.
[lvgl_conf_and_port_olimex_mod_lcd.zip](https://github.com/lvgl/lvgl/files/4728094/lvgl_conf_and_port_olimex_mod_lcd.zip)

At the moment, to handle the color swap I'm using the 1st buffer method. I declare 3 screen size buffers. The first is used by lvgl to render the graphics. The 2 others are used by the LCD. When flush_cb is called by lvgl I "copy and swap red and blue" of the area from the lvgl render buffer (1st) to the hidden lcd buffer. When the copy is finished, I simply swap the lcd pointer (like the 3rd buffer method).

Because of the buffer area copy, it's not as efficient as possible. Because lvgl doesn't always render all the screen when it calls flush_cb, I guess I cannot use DMA.

But the performance when I use the first buffer method seems more than enough. Even when I tell the task that call lv_task_handler to sleep periodically for 10ms, it shows everything smoothly and render the animations beautifully. That means people here have done a really good job 👍 

When using freeRTOS, I use 2 task like that:
```c
/**
 * @brief update lvgl tick
 */
void update_lvgl_tick()
{
    while(1)
    {
        lv_tick_inc(1);
        vTaskDelay( 1 / portTICK_PERIOD_MS );
    }
}

/*
* @brief littlevGL update screen task
*/
void littlvGL_screen_update()
{
    while(1)
    {
        lv_task_handler();
        // vTaskDelay( 10 / portTICK_PERIOD_MS ); // just to test lvgl speed
        // no vTaskDelay is called because we use the lowest prio
        // and systick is at 1ms
    }
}
```
That means the fewer other tasks are working, the better the render will be.

kisvegabor, 2020-06-04:
I've also played with it and concluded the same: the easiest thing is to modify `LV_COLOR_MAKE`.