lvgl/lvgl #1416: lv_canvas seems won't auto refresh display

aodzip, 2020-03-23:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

When I use a task to dynamiclly add text to canvas, canvas remain black. 

**To Reproduce**
[TerminalEmulator.zip](https://github.com/littlevgl/lvgl/files/4367231/TerminalEmulator.zip)
```c
    static terminalemulator_t te = {0};
    te.config.column = 10;
    te.config.line = 5;
    lv_style_copy(&te.config.style_canvas_text, &lv_style_plain);
    te.config.style_canvas_text.body.grad_color = LV_COLOR_BLACK;
    te.config.style_canvas_text.body.main_color = LV_COLOR_BLACK;
    te.config.style_canvas_text.text.font = &lv_font_roboto_12;
    te.config.style_canvas_text.text.color = LV_COLOR_WHITE;
    strcpy(te.config.exe, "./a.out");
    TerminalEmulator_open(win_app, &te);
```
**Expected behavior**

Canvas should output some text

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

uncomment //Ugly Workaround will refresh canvas.

kisvegabor, 2020-03-24:
You are right. The canvas wasn't invalidated in draw functions.
Should be fixed here: https://github.com/littlevgl/lvgl/commit/437169243b8fb921742a4ccc256e395633d5f1bc

Is it ok now?

aodzip, 2020-03-24:
tested, it is ok now. no more ugly workaround
[image]

embeddedt, 2020-03-24:
It looks very nice! I do notice that the spacing of some characters is too small/too large but that's a minor issue.

Since the original problem is resolved I'll close this issue now.

closed 2020-03-24 by embeddedt

kisvegabor, 2020-03-24:
It really looks nice!

> I do notice that the spacing of some characters is too small/too large but that's a minor issue.

It seems all characters have the same box width and the shorter ones are aligned to the left.