lvgl/lvgl #1417: [dev7.0] Latest lv_draw_blend.c introduces artifacts

fvanroie, 2020-03-24:
**Describe the bug**

I have started testing the dev-7.0 local style system and have achieved good results. But commit https://github.com/littlevgl/lvgl/commit/f9edf0b341984dd28b1c3e162220b430cf6ee9ac introduces vertical artifacts in some of the the widgets. 

**To Reproduce**

I updated from 2a1db9ee to 523c9793 which introduced visible vertical artifacts. I traced it back to file `lv_draw\lv_draw_blend.c` and reverting to the previous version of that file fixes the issue.

**Expected behavior**

Get rid of the vertical artifacts.

**Screenshots**
On top is the the previous version 2a1db9ee. And at the bottom the latest 523c9793 with the issue.
[image]

**Additional context**

I experienced the issue on ESP8266 and ESP32. It seems there is bug in the latest lv_draw_blend.c update. I haven't verified in the simulator, but is needed I can run a test there as well.

barbiani, 2020-03-24:
Simulator @16bpp shows the same here.

kisvegabor, 2020-03-24:
Thank you for the detailed report. I've just fixed it in ec57e921

Does it work well for you too?

fvanroie, 2020-03-24:
I confirm the problem is indeed resolved now.
Thank you for the fast response @kisvegabor.

kisvegabor, 2020-03-24:
Great! Thank you for the feedback!

commit ec57e921 (2020-03-24): fix belnding issue. Fixes #1417
closed 2020-03-24 by kisvegabor