lvgl/lvgl #1418: [dev7.0] arc angle data type

xennex22, 2020-03-24:
The arc angle data type is `lv_coord_t` but the setter and getter functions use `int16_t`

I started to add some additional getter functions when I noticed.

Since an arc is only going to be 0..360 then perhaps `int16_t` would be best data type to use.

kisvegabor, 2020-03-25:
I agree with `int16_t` could you send a PR?

kisvegabor, 2020-03-25:
Or why not uint16_t?

xennex22, 2020-03-26:
There is existing code to cope with angles < 0, which would be redundant.  If we change to uint16_t it should be ok as long as user code doesn't cast from int16_t to uint16_t and pass values of 0x1000 - angle.  Probably I need to test a little which I can do tomorrow, then I'll do the PR.

kisvegabor, 2020-03-27:
Great! Thank you!

xennex22, 2020-03-27:
uint16_t works.
There is a lot of code like this
`
if(start > 360) start -= 360;
if(start < 0) start += 360;
`
But in some place it is
`
if(start_angle >= 360) start_angle -= 360;
`
I've replaced it with
`
start = start % 360;
`
The drawing calculations use lv_coord_t but some of the calculations subtract values and they currently work so I don't feel like changing the type to unsigned.

Found and fixed a bug where the invalidation would not work for the background arc when it was within 180 degrees of the end.

Now I need to work out how to add a set_rotation function to set the angle origin.

xennex22, 2020-03-27:
Rotating the whole arc.

`void lv_arc_set_rotation(lv_obj_t * arc, uint16_t rotation_angle);`

Works ok.  The invalidation tests in the setters and the drawing needs code like.

`
uint16_t bg_start = (ext->bg_angle_start + ext->rotation_angle) % 360;
uint16_t bg_end = (ext->bg_angle_end + ext->rotation_angle) % 360;`

Should I make another PR or use the same PR?

kisvegabor, 2020-03-30:
I was also thinking about adding `lv_arc_set_rotation`. Glad to hear that you implemented it.

You can use the same PR, thank you!

kisvegabor, 2020-03-30:
Please be sure to test the new API function with the following special cases:
- start = 0; end = 360: full circle
- start = 90; end = 450: full circle
- start = 0; end = 0: empty erc
- start = 90; end = 90; empty arc
- start = 90; end = 80; almost full arc

closed 2020-03-31 by xennex22