lvgl/lvgl #1424: [dev7.0] lv_dropdown_add_option does not handle utf8

xennex22, 2020-03-28:
Adding drop down options with utf8 text does not work.  `lv_dropdown_add_option()` passes a byte position to `lv_txt_ins()` which then uses a character position for the insert, so the insert goes past the end of the existing string.

Also I wonder about iterating through a string byte by byte looking for '\n' which might fail if a utf8 string has a 0x0A in it.  eg in `lv_dropdown_set_options`.  Not sure of what utf8 would have 0x0A in it but it is possible.

Seems like all `strlen` should be `lv_txt_get_encoded_length` and iteration should use `lv_txt_encoded_next`.  For some things, like buffer copies, `strlen` works on utf8 but for other things, like inserting, `strlen` is no good.

I'll start to look at it with the view of doing a PR.

xennex22, 2020-03-29:
Panic over.  UTF8 always has the highest bit set, so it is not going to give a premature null or mistaken 0x0A.  Therefore strlen and strcpy should be ok for allocation and copying strings, and even for copying out a label value.

So I just need to convert the byte position to a character position so that `lv_txt_ins` can convert it back to a byte position again.

kisvegabor, 2020-03-31:
So finally, do you think is there anything to improve here?

puzrin, 2020-03-31:
Note about fonts: make sure to support option "compression without prefilter". This may be a tradeoff between size and performance.

Bin spec has this option https://github.com/littlevgl/lv_font_conv/blob/master/doc/font_spec.md#table-head-font-header (Compression alg ID = 2).

kisvegabor, 2020-03-31:
Do you mean in lvgl in general? Or is it related to this issue?

xennex22, 2020-03-31:
The original issue is fixed and merged.

closed 2020-04-02 by xennex22