lvgl/lvgl #1426: Drag TabView On Any View？

ytm52513, 2020-04-01:
Tab3 has a lmeter view. It is now in the red area, You cannot slide to tab2. You can only drag outside the red area. How to set it to be draggable in the red area?
[image]

ytm52513, 2020-04-01:
The red area is the size of the lmeter view.

ytm52513, 2020-04-01:
[image]

embeddedt, 2020-04-01:
`lv_obj_set_click(lmeter, false)` should work.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2020-04-01 by embeddedt