lvgl/lvgl #1427: Refresh task, mono display

apapkovskiy, 2020-04-01:
For 1 bit displays [that line](https://github.com/littlevgl/lvgl/blob/e3e130c54ca29f6298adc1eec25d5365fa8d6821/src/lv_core/lv_refr.c#L199) writes outside the buffer. It assumes that the pixel size is `sizeof(lv_color_t)` but for mono display it is `sizeof(lv_color_t)/8`

kisvegabor, 2020-04-02:
Hi,
Thank you for the report.
Actually if `LV_COLOR_DEPTH 1` `lv_color_t` is still `uint8_t` for performance reasons. 
See [here](https://github.com/littlevgl/lvgl/blob/master/src/lv_misc/lv_color.h#L201-L210) and [here](https://github.com/littlevgl/lvgl/blob/master/src/lv_misc/lv_color.h#L253-L255).
However, if you have a `set_px_cb` callback in your display driver, lvgl doesn't know the bit size of the pixels. In this case, it really should cause issues. Did you mean this? Or have I missed something?

apapkovskiy, 2020-04-02:
I use Zephyr port of lvgl and it uses `set_px_cb` for the display driver. For the monochrome display the buffer size is w*h/8. That is the root of the problem ((((. It make sense (size w*h/8) due to it enables copying the whole buffer to the display without making any changes on the buffer.

apapkovskiy, 2020-04-02:
May be it make sense also introduce a callback to sync buffers?

If I don't use double-buffering I have to update the whole screen always due to lvgl flush vdb for each area separately. It is too expensive for e-ink display. In case of double-buffers lvgl renders all areas and only then flush vdb.

Also it can be solved by the flag the VDB has screen size and flush vdb after rendering all areas are done.

embeddedt, 2020-04-02:
> If I don't use double-buffering I have to update the whole screen always to have only one area to update due to lvgl flush vdb for each area separately. It is too expensive for e-ink display.

Depending on how your eInk display is set up, you may be able to use [`monitor_cb`](https://docs.littlevgl.com/en/html/porting/display.html#display-driver) for this purpose (search for it on that page). It gets called once all the areas have been sent. In my case, I update the underlying framebuffer each time `flush_cb` is called and tell the display to update in `monitor_cb`.

apapkovskiy, 2020-04-02:
@embeddedt Thanks a lot I think it should work

closed 2020-04-02 by apapkovskiy

kisvegabor, 2020-04-03:
You can also use the `rounder_cb` to force lvgl to always invalidate (redraw) the whole screen. This way, with 1 screen-sized buffer you will get the whole rendered screen in `flush_cb`. 

Anyway, it shouldn't crash so I fixed it by skipping the copy if `set_px_cb` is set.
See 5d8798b5

apapkovskiy, 2020-04-03:
@kisvegabor Thanks. I think it is even better !!!