lvgl/lvgl #1428: lv_chart_set_x_tick_texts

harteware07, 2020-04-02:
Hello!
when using the chart functions I noticed a strange behaviour. I need to have different text on the x axis, depending on the zoom level. 

It seems, as if the num_tick_marks value of the lv_chart_set_x_tick_texts (..) function is cropped somewhere.
I attach some photos where you can see the error:

num_tick_x = 17;
[image]

num_tick_x = 19;
[image]

num_tick_x = 15;
[image]

kisvegabor, 2020-04-02:
Hi,

Please send a code snippet to reproduce the issue. The data of the series is not important now, only the size of the chart and settings of the ticks.

harteware07, 2020-04-03:
here is a complete code example to reproduce the error
`void lv_ex_chart_1(void)
{
	static lv_style_t styleStats;
	lv_style_copy(&styleStats, &lv_style_pretty);
	styleStats.text.color 		= LV_COLOR_YELLOW;

    /*Create a chart*/
    lv_obj_t * chart;
    chart = lv_chart_create(lv_scr_act(), NULL);
    lv_obj_set_size(chart, 400, 150);
    lv_obj_align(chart, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_chart_set_type(chart, LV_CHART_TYPE_POINT | LV_CHART_TYPE_LINE);   /*Show lines and points too*/
    lv_chart_set_series_opa(chart, LV_OPA_70);                            /*Opacity of the data series*/
    lv_chart_set_series_width(chart, 4);                                  /*Line width and point radious*/
    lv_chart_set_margin(chart, 50);
    lv_chart_set_point_count(chart, 31);
    lv_chart_set_range(chart, 0, 100);

    /*Add two data series*/
    lv_chart_series_t * ser2 = lv_chart_add_series(chart, LV_COLOR_GREEN);

    lv_chart_set_style(chart, LV_CHART_STYLE_MAIN, &styleStats);

    int num_tick_x = GUI_STATS_CHART_NUM_POINTS_MAX;
	const char *s = "0\n1\n2\n3\n4\n5\n6\n7\n8\n9\n10\n11\n12\n13\n14\n15\n16\n17\n18\n19\n20\n21\n22\n23\n24\n25\n26\n27\n28\n29\n30";
	for(int i = 0; i < num_tick_x; i++) {
		ser2->points[i] = i;
	}
	lv_chart_set_x_tick_texts(chart, s, num_tick_x, LV_CHART_AXIS_DRAW_LAST_TICK);
    lv_chart_refresh(chart); /*Required after direct set*/
}
`
It should show 31 numbers, but shows only 6:
[image]

kisvegabor, 2020-04-06:
Thanks. I've tested and found that setting `LV_CHART_AXIS_TICK_LABEL_MAX_LEN 128` in `lv_conf.h` solves the issue.

kisvegabor, 2020-04-24:
In lack of feedback, I close this issue.

Let us know if it's still not working.

closed 2020-04-24 by kisvegabor

harteware07, 2020-05-06:
it is still not working for all numbers of num_tick_x.
num_tick_x = 23 -> 11 markers are displayed
num_tick_x = 24 -> 2  markers are displayed

so a very strange behaviour. Any ideas?

kisvegabor, 2020-05-06:
Why do you want to set so many ticks?
As the [docs](https://docs.littlevgl.com/en/html/object-types/chart.html#tick-marks-and-labels) says:
> If list_of_values is set then num_tick_marks tells the number of ticks between two labels. If list_of_values is NULL then it specifies the total number of ticks.

So with 23 you will get 23 small lines between each number. Do you really want this?

kisvegabor, 2020-05-06:
These settings were really not working due to buffer overflow.
I fixed it here: 18010c2e

harteware07, 2020-05-19:
> So with 23 you will get 23 small lines between each number. Do you really want this?

no :) i thougt something wrong. Thx for pointing this out, it is working for me now!

kisvegabor, 2020-05-19:
Great! :slightly_smiling_face: