lvgl/lvgl #1432: [dev7.0] creating imgbtn call img decoder with null image source

xennex22, 2020-04-03:
Code - create an image button
```
lv_obj_t * btn = lv_imgbtn_create(screen, NULL);
```
Warning output
```
Image get info found unknown src type
```
Call stack 
```
lv_imgbtn_create(par, NULL)
lv_theme_apply(imgbtn, LV_THEME_IMGBTN)
lv_theme_material_apply(imgbtn, lv_theme_style_t name)
lv_obj_refresh_style(imgbtn, LV_STYLE_PROP_ALL)
lv_imgbtn_signal(imgbtn, LV_SIGNAL_STYLE_CHG, NULL)
refr_img(imgbtn)
lv_img_decoder_get_info(const char * src = NULL, lv_img_header_t * header)
```
When `refr_img()` gets the image src it can check for a null and return, but I wonder if either I am doing something wrong or there is a better way to fix this.
I'd rather catch the null image src before calling the decoder.
```
const void * src = ext->img_src_mid[state];
if(src == NULL) return;
```
But I'd like warnings about missing button images at that point.  I can add a creating flag to the lv_imgbtn_ext_t, but it seems like there must be a better way.
This is with the creating flag
```
if(src == NULL) {
    if(!ext->creating)
        LV_LOG_WARN("Image btn src is NULL");
    return;
}
```

xennex22, 2020-04-04:
This is what I changed for my code.  There probably is a better way.
https://github.com/xennex22/lvgl/commit/dfe2f54417b6e5cea336de5f23c5681d55da6bcb#diff-3ddd9e0c98f1e44b2922282b79b9c22a
https://github.com/xennex22/lvgl/commit/f3439fa6eb9fe2061d89c3a7b39bcd122ca65b57#diff-701b57bbde7dfd327e9ecbe963e4fadd

kisvegabor, 2020-04-06:
Hi,

Thank you for the detailed report.
It seems you are doing it correctly.

> When refr_img() gets the image src it can check for a null and return, but I wonder if either I am doing something wrong or there is a better way to fix this.

I think this solution is fine. I've added it here: d75ea5f4

Does it work well for you too?

xennex22, 2020-04-06:
Your solution will prevent a null src being passed to the child functions, which I think is necessary, but it will not warn about missing button images.

kisvegabor, 2020-04-06:
> Your solution will prevent a null src being passed to the child functions, which I think is necessary

In which case do you think it's necessary to pass `NULL` src?

> but it will not warn about missing button images.

There is really no warning here but we can add one in `else` branch here:
```c
   if(info_res == LV_RES_OK) {
        ext->act_cf = header.cf;
        if(ext->tiled) lv_obj_set_height(imgbtn, header.h); /*Keep the suer defined width*/
        else  lv_obj_set_size(imgbtn, header.w, header.h);
    }
    else {
        ext->act_cf = LV_IMG_CF_UNKNOWN;
    }
```

Or did you mean something else?

xennex22, 2020-04-06:
I think it is necessary to prevent a NULL src being passed to the child functions.  Your solution does that.

As for the NULL check for an imgbtn source it is not that important; perhaps we should leave it alone.

kisvegabor, 2020-04-06:
> I think it is necessary to prevent a NULL src

Ah sorry, I've misread your previous comment. 

>  perhaps we should leave it alone.

Okay, then I close it. Thank you for reporting this issue!

closed 2020-04-06 by kisvegabor