lvgl/lvgl #1433: Arabic [or Persian] words do not print right.

HamidSaffari, 2020-04-04:
Hi, and salute to the professional creators.
I've tried to print some Arabic words but it prints wrongly.
based on your wiki page we can print this Arabic words:
[image]

but the code:

    static lv_style_t style1;
    lv_style_copy(&style1, &lv_style_plain);
    style1.text.font = &arabic_simpbdo_20; 
    lv_label_set_style(btn_label, LV_LABEL_STYLE_MAIN, &style1);
    lv_label_set_text(btn_label, "الکویت");  

and result:

[image]

Now there is two problems:

1. Starting characters from left to right instead of RTL. no matter what the configuration is [ LV_USE_BIDI 1  LV_BIDI_BASE_DIR_DEF  LV_BIDI_DIR_RTL ].

2. In these languages depend on next character the current character might change it's shape in some cases, but here just print all with space band.

HamidSaffari, 2020-04-04:
I realized that this issue discussed here before:

https://forum.littlevgl.com/t/how-to-use-bidi-lv-use-bidi-used-for-arabic-any-demo-examples/1290/17

And Anda gave some example code for printing Arabic correctly.
Can we edit it and will include it inside your src?
here is that example code:

bool Arabic_GetLigatrueFirstState(u16 now)
{
	for(int index = 0; index < ARRAY_COUNT(Arabic_CharSet1); index ++){
	  if( now == Arabic_CharSet1[index] ) return true;
	}
	return false;
}

bool Arabic_GetLigatureLastState(u16 now)
{
	for(int index = 0; index < ARRAY_COUNT(Arabic_CharSet2); index ++){
	   if( now == Arabic_CharSet2[index] ) return true;
	}
	return false;
}

bool Arabic_GetLigatureSpecCharState(u16 now, u16 next)
{
	if( now != 0x0644 ) return false;

	if( next == 0x0622 || next == 0x0623 || next == 0x0625 || next == 0x0627 ) return true;

	return false;
}

u16 Arabic_GetChar(u16 prev, u16 now, u16 next)
{
	if( now <= 0x0621 || now >= 0x064A ) return now;

	u32 offset = now - 0x0621;

	bool first_flag = Arabic_GetLigatrueFirstState(prev);
	bool last_flag = Arabic_GetLigatureLastState(next);

	u8 state = (first_flag << 0) | (last_flag << 1);

	u16 code;

	switch( state ){
		case 0:
		{
			now = Arabic_PositionTable[offset][3];
		}
		break;
		case 1:
		{
			now = Arabic_PositionTable[offset][0];
		}
		break;
		case 2:
		{
			now = Arabic_PositionTable[offset][1];
		}
		break;
		case 3:
		{
			now = Arabic_PositionTable[offset][2];
		}
		break;
		default:
		{
			now = Arabic_PositionTable[offset][3];
		}
		break;
		}

	return now;
}

u16 Arabic_GetLigatureSpecChar(u16 prev, u16 next)
{

	u16 code  = 0;
	u8  state = Arabic_GetLigatrueFirstState(prev);

	switch( next ){
	case 0x0622:
	{
		code = Arabic_CharSpecs[0][state];
	}
	break;
	case 0x0623:
	{
		code = Arabic_CharSpecs[1][state];
	}
	break;
	case 0x0625:
	{
		code = Arabic_CharSpecs[2][state];
	}
	break;
	case 0x0627:
	{
		code = Arabic_CharSpecs[3][state];
	}
	break;
	}

	return code;
}

void Arabic_Process(u16 *in, u32 in_count, u16 *out, u32 *out_count)
{

	if( in_count == 0 ){
		*out_count = 0;
		return;
	}
	if( in_count == 1 ){
		*out = *in;
		*out_count = 1;
		return;
	}
	u32 index = 0;
	u32 count = 0;
	while( index < in_count ){
		if( index == 0 ){
			*out = Arabic_GetChar(0, *in, *(in + 1));
			out ++;
			in ++;
			index ++;
			count ++;
		}else if( index == (in_count - 1) ){
			*out = Arabic_GetChar(*(in - 1), *in, 0);
			index ++;
			count ++;
		}else{
			if( Arabic_GetLigatureSpecCharState(*in, *(in + 1)) ){
				*out = Arabic_GetLigatureSpecChar(*(in - 1), *(in + 1));
				out ++;
				in += 2;
				index += 2;
			}else{
				*out = Arabic_GetChar(*(in - 1), *in, *(in + 1));
				out ++;
				in ++;
				index ++;
			}
			count ++;
		}
	}

	*out_count = count;
}

void Arabic_Test(void)
{
	// origin text: “مرحبا”;

	u16 arabic_text_unicodes[] = {
		0x645,
		0x631,
		0x62D,
		0x628,
		0x627,
	};
	u32 arabic_text_count = ARRAY_COUNT(arabic_text_unicodes);
	u16 arabic_result_unicodes[32];
	u32 arabic_result_count = 0;
	Arabic_Process(arabic_text_unicodes, arabic_text_count, arabic_result_unicodes, &arabic_result_count);
	// reverse(arabic RTL)
	for(int index = arabic_result_count - 1; index >= 0; index --){
		u16 letter       = arabic_result_unicodes[index];
		u16 letter_next  = (index == 0) ? 0 : arabic_result_unicodes[index - 1];
		u16 letter_width = lv_font_get_glyph_width(font, letter, letter_next);
		lv_draw_letter(&point, mask, font, letter, color, opa);
		if( letter_width ){
			 point.x += letter_width + style->text.letter_space;
		}
	}
}

embeddedt, 2020-04-04:
There is an open PR here: #1409

Those changes need some refactoring before they can be included upstream but it appears that they do work.

kisvegabor, 2020-04-06:
@HamidSaffari 
Let's continue in the PR @embeddedt mentioned.

closed 2020-04-06 by kisvegabor

HamidSaffari, 2020-04-06:
Hey I'm working in one re-shaper now and just facing one problem:
Is there any way in your text input functions you accept ( wchar_t* ) too? 
since your input texts only accept (const char*) how can I put wchar_t (double-char) type in that. 
in other word since I have to manipulate characters in word so I had to define words as wchar_t to get it's 16-bits unicodes like 0xFEB7 but after reshaping when I trying to feed it into your function it prints something else when I convert it by  wcstombs() or by recasting to (const char*) .
here is my code:

	const wchar_t* wstr = L"ﺷﮏ"; 
       // reshape(wstr, wcslen(txt1), true);// lets skip this
	char* ascii = new char[wcslen(wstr)+1];
	wcstombs (ascii, wstr, strlen(wstr)); 

        lv_label_set_text(btn_label, ascii);     /*print btn label*/

Also I've got the reshaper funcs from:   
https://github.com/Naheel-Azawy/c-arabic-reshaper

embeddedt, 2020-04-06:
You'll need to convert your wide string to UTF-8. We don't directly support 16-bit Unicode.

HamidSaffari, 2020-04-06:
I'm trying to convert UTF_16 wchar_t* UTF_16 to UTF-8 by:

size_t wcstombs (char* dest, const wchar_t* src, size_t max);

but it does not give me the right UTF-8 char*.