lvgl/lvgl #1435: set_px_cb is not invoked

Wielebny666, 2020-04-08:
Hi,
Probably "set_px_cb" is not called when there is no data to display.
The problem is that flush_cb sends the previous buffer content to the display.

Buffer size
static lv_color_t buf3_1[LV_HOR_RES_MAX * LV_VER_RES_MAX / 8];

**If buffer size is LV_HOR_RES_MAX * LV_VER_RES_MAX everything is OK.**
```
static void lv_vdb_wr_bpp1(lv_disp_drv_t *disp_drv, uint8_t *buf,
		lv_coord_t buf_w, lv_coord_t x, lv_coord_t y, lv_color_t color,
		lv_opa_t opa)
{
	//ESP_LOGI(TAG, "buf_w- %d x- %d y- %d color- %d br- %d", (uint8_t)buf_w, x, y, color.full, lv_color_brightness(color));
	printf("%d", color.full);
	buf += buf_w * (y >> 3) + x;
	if (lv_color_brightness(color) == 0)//> 128)
	{
		(*buf) |= (1 << (y % 8));
	}
	else
	{
		(*buf) &= ~(1 << (y % 8));
	}
}
```

[image]

kisvegabor, 2020-04-09:
Hi,

I have a few questions:
- Do you use 2 `LV_HOR_RES_MAX * LV_VER_RES_MAX / 8` buffers?
- Could you show how you initialize `lv_disp_drv_t`?
- Do you use the latest lvgl version from `master`?
- Could you try what happens with `LV_HOR_RES_MAX * LV_VER_RES_MAX / 8 + 1` sized buffer?

Wielebny666, 2020-04-09:
Hi,
> * Do you use 2 `LV_HOR_RES_MAX * LV_VER_RES_MAX / 8` buffers?
I use only one.

> * Could you show how you initialize `lv_disp_drv_t`?
```
	static lv_disp_buf_t disp_buf_3;
	static lv_color_t buf3_1[LV_HOR_RES_MAX * LV_VER_RES_MAX / 8]; /*A screen sized buffer*/
	lv_disp_buf_init(&disp_buf_3, buf3_1, NULL, LV_HOR_RES_MAX * LV_VER_RES_MAX / 8); 

	lv_disp_drv_t disp_drv; /*Descriptor of a display driver*/
	lv_disp_drv_init(&disp_drv); /*Basic initialization*/

	disp_drv.hor_res = LV_HOR_RES_MAX;
	disp_drv.ver_res = LV_VER_RES_MAX;

	disp_drv.flush_cb = disp_flush;
	disp_drv.set_px_cb = lv_vdb_wr_bpp1;
	disp_drv.rounder_cb = rounder_cb;

	disp_drv.buffer = &disp_buf_3;
```

> * Do you use the latest lvgl version from `master`?
last "master"
> * Could you try what happens with `LV_HOR_RES_MAX * LV_VER_RES_MAX / 8 + 1` sized buffer?
i check at noon.

Wielebny666, 2020-04-09:
Maybe here is the problem?

#define LV_COLOR_SCREEN_TRANSP    0

```
LV_ATTRIBUTE_FLUSH_READY void lv_disp_flush_ready(lv_disp_drv_t * disp_drv)
{
    /*If the screen is transparent initialize it when the flushing is ready*/
#if LV_COLOR_SCREEN_TRANSP
    if(disp_drv->screen_transp) {
        memset(disp_drv->buffer->buf_act, 0x00, disp_drv->buffer->size * sizeof(lv_color32_t));
    }
#endif

    disp_drv->buffer->flushing = 0;
}
```

BTW, it should not be?

memset(disp_drv->buffer->buf_act, 0x00, disp_drv->buffer->size * sizeof(**lv_color_t**));

Wielebny666, 2020-04-09:
> * Could you try what happens with `LV_HOR_RES_MAX * LV_VER_RES_MAX / 8 + 1` sized buffer?
no change

Now I know that LV_COLOR_SCREEN_TRANSP is for LCD with 32bit color depth. ;-)

Resolve problem
```
LV_ATTRIBUTE_FLUSH_READY void lv_disp_flush_ready(lv_disp_drv_t * disp_drv)
{
    /*If the screen is transparent initialize it when the flushing is ready*/
#if LV_COLOR_SCREEN_TRANSP
    if(disp_drv->screen_transp) {
        memset(disp_drv->buffer->buf_act, 0x00, disp_drv->buffer->size * sizeof(lv_color32_t));
    }
#elif LV_COLOR_DEPTH == 1
    memset(disp_drv->buffer->buf_act, 0x00, disp_drv->buffer->size * sizeof(lv_color_t));
#endif
    
    disp_drv->buffer->flushing = 0;
}
```

kisvegabor, 2020-04-10:
See my comment in https://github.com/littlevgl/lvgl/pull/1437

closed 2020-04-10 by kisvegabor