lvgl/lvgl #1438: lv_font_get_bitmap_fmt_txt() corrupts LVGL heap when rendering compressed font

BillHargen, 2020-04-10:
**Describe the bug**

The lv_font_get_bitmap_fmt_txt() routine can corrupt the LVGL heap when rendering a glyph from a compressed font.

This routine computes the amount of memory required to store the decompressed glyph.  The computation divides the number of pixels by the pixels-per-byte (based on the bpp of the font) but this computation rounds *down*.

**To Reproduce**

Hitting the bug depends on a number of conditions to be met:
1. First, the number of pixels cannot be a multiple of the pixels per byte.  For a font with bpp=4 this means that the number of pixels must be an odd number, which means that both the width and height must be off.  It is easier to hit this case with bpp=2 or bpp=1.
2. Second, this overrun needs to cross the allocation boundary of the heap manager.  For example, with bpp=4, a pixel count of 15 is not a problem because although the byte count is miss-computed as 7 instead of 8, the memory allocator will round the allocation up to 8 bytes anyway so no harm will be done.  But a pixel count of 9 leads to allocating 4 bytes, actually getting 4 bytes, and then using 5 bytes.
3. Third, because the code reuses a prior buffer via lv_mem_realloc(), it is also necessary that this case is larger than any previously rendered glyph, otherwise the previously allocated buffer is larger than necessary and the overrun does no harm.

It is not practical to include sample code because it can be difficult to hit these conditions, and the heap corruption needs to impact continued operation.  My case was using Noto Sans Bold converted with bpp=4 and compressed.  The character that corrupted the heap was capital A (U+0041, 17 x 17 pixels, a total of 289 pixels, needing 145 bytes, but requesting and receiving 144).

I added code to lv_font_get_bitmap_fmt_txt() to report when the edge condition was occurring and also reporting the available LVGL heap space before and after the call to decompress().  The output confirmed that the heap was corrupted during the decompress() call when the edge condition was met.

I changed the computation as shown below to round up and this solved the problem, at least for the bpp=4 case.

```
        switch(fdsc->bpp) {
        case 1: buf_size = (gsize + 7) >> 3;  break;
        case 2: buf_size = (gsize + 3) >> 2;  break;
        case 3: buf_size = (gsize + 1) >> 1;  break;
        case 4: buf_size = (gsize + 1) >> 1;  break;
```

embeddedt, 2020-04-11:
Wow! Thank you for the detailed analysis and solution!

@kisvegabor Can you review this to make sure it's correct? It looks fine to me, but I know you're more familiar with this part of the codebase.

kisvegabor, 2020-04-12:
Thank you @BillHargen! Your fix looks perfect! Can you send a Pul request?

PR #1441 (2020-04-12): Bug #1438 - round up memory size for decompressing glyph
Closes #1438

Corrected computation of memory size needed to decompress a glyph. The prior code rounded down, leading to a memory overrun in certain cases.
commit 7e4e7498 (2020-04-12): Fix #1438 - round up memory size for decompressing glyph (#1441)
commit 4c2fcebe (2020-04-13): Fix #1438 - round up memory size for decompressing glyph (#1441)
closed 2020-04-12 by embeddedt