lvgl/lvgl #1439: Label does not work

Wielebny666, 2020-04-11:
A simple example does not work.
```
    lv_obj_t * label2 = lv_label_create(lv_scr_act(), NULL);
    lv_obj_set_width(label2, 50);
    lv_label_set_text(label2, "It is a text. ");
    lv_obj_align(label2, NULL, LV_ALIGN_CENTER, 0, 0);
```

Other objects like button, roller without a problem.

embeddedt, 2020-04-11:
Does it work with the PC simulator?

Wielebny666, 2020-04-11:
> Does it work with the PC simulator?
I do not use :-(

Inside Page work correctly

Wielebny666, 2020-04-11:
With PC Simulator is OK.

gui: C:/ESP/lvgl/src/lv_objx/lv_label.c 161 label created
but "flush" is not invoked

Wielebny666, 2020-04-11:
> With PC Simulator is OK.
> 
> gui: C:/ESP/lvgl/src/lv_objx/lv_label.c 161 label created
> but "flush" is not invoked
I'm sorry my mistake,"flush" is invoked.

embeddedt, 2020-04-11:
> With PC Simulator is OK.

It sounds like an issue with your target platform then. Are you calling `lv_task_handler` multiple times? Do you see the flush function getting invoked on your target platform?

Wielebny666, 2020-04-11:
> Are you calling lv_task_handler multiple times?

Yes

```
	while (1)
	{
		vTaskDelay(1);
		//Try to lock the semaphore, if success, call lvgl stuff
		if (xSemaphoreTake(xGuiSemaphore, (TickType_t)10) == pdTRUE)
		{
			lv_task_handler();
			xSemaphoreGive(xGuiSemaphore);
		}
	}
```
> Do you see the flush function getting invoked on your target platform?

My log:

I (676) app_main: app_main
I (676) gui: guiTask
I (686) lv_port_disp: lv_port_disp_init
I (686) lv_port_disp: disp_init
I (706) ssd1306_obj: CSsd1306 24
I (706) ssd1306: iot_ssd1306_create 419
I (706) ssd1306: iot_ssd1306_init 327
I (716) ssd1306: iot_ssd1306_clear_screen 470
I (736) gui: C:/ESP/lvgl/src/lv_core/lv_obj.c 236 Screen create ready
I (736) gui: C:/ESP/lvgl/src/lv_core/lv_obj.c 236 Screen create ready
I (746) gui: C:/ESP/lvgl/src/lv_core/lv_obj.c 236 Screen create ready
I (746) lv_port_disp: rounder_cb
I (756) lv_port_disp: rounder_cb
I (756) lv_port_disp: rounder_cb
I (756) lv_port_disp: rounder_cb
I (766) lv_port_disp: rounder_cb
I (776) lv_port_disp: rounder_cb
I (776) lv_port_disp: rounder_cb
I (776) lv_port_indev2: lv_port_indev_init
I (786) lv_port_indev2: encoder_init
D (796) intr_alloc: Connected src 48 to int 4 (cpu 1)
I (796) lv_port_indev2: button_init
D (796) intr_alloc: Connected src 22 to int 9 (cpu 1)
I (806) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (806) lv_port_disp: rounder_cb
I (816) lv_port_disp: rounder_cb
I (816) lv_port_disp: rounder_cb
I (826) lv_port_disp: rounder_cb
I (836) lv_port_disp: rounder_cb
I (836) lv_port_disp: rounder_cb
I (836) lv_port_disp: rounder_cb
I (836) lv_port_disp: rounder_cb
I (836) lv_port_disp: rounder_cb
I (856) gui: C:/ESP/lvgl/src/lv_objx/lv_label.c 161 label created
I (926) lv_port_disp: rounder_cb
I (926) lv_port_disp: disp_flush

Wielebny666, 2020-04-11:
OK, i found problem.

I put theme init before lv_disp_drv_register()
```
	lv_theme_mono_init(0, NULL);
	lv_theme_set_current(lv_theme_get_mono());
```
when move after, now work.

closed 2020-04-11 by Wielebny666