lvgl/lvgl #1440: Spinbox problems

Wielebny666, 2020-04-12:
**Describe the bug**
Errors in spinbox mechanics

**To Reproduce**
I can't reproduce the error on "PC emu" because "spinbox" does not support the mouse wheel.
I'm using an encoder.

1.
```
lv_spinbox_set_range(spinbox, 1, 480);
```
After initializing "spinbox" displays "+".
Disappears when the value changes.

2.
If we leave the default range (-99999 - 99999) and in lv_spinbox_set_range () digit_count for example 3.
Initial value 0. I see "+000" on display.
I increase the value of the first digit by 12, on display "+012". 
I press the encoder button.
"gui: C: /ESP/lvgl/src/lv_objx/lv_ta.c 250 Text area: line break ignored in one-line mode"
The cursor moves to the "+" sign. 
[image]
I turn the encoder to the right and the digit to the right of the cursor will increase. 
I see "+100" on the display.
Log from the event "LV_EVENT_VALUE_CHANGED"
"gui: Value: 10012"

**Expected behavior**
1.
lv_spinbox_set_range should invoke lv_spinbox_updatevalue()

2.
The "digit_count" setting in lv_spinbox_set_digit_format() should check or change the "range_min" and "range_max" values

kisvegabor, 2020-04-12:
Thank you for the report!
I've fixed it here: ccbf4373

Wielebny666, 2020-04-12:
3. 
After initialization, the cursor is displayed.
Disappears after focus on spinbox

kisvegabor, 2020-04-14:
Do you mean the cursor disappears when the spinbox is defocused? If not, can you send a minimal code to reproduce?

Wielebny666, 2020-04-14:
After loading gui, the focus is on another widget.  The cursor can be seen in the spinbox, but it is not focused.
The problem is only before its first use.
So the initial state is wrong, but only when the focus is not on it when loading the gui.

```
	lv_theme_mono_init(0, NULL);
	lv_theme_set_current(lv_theme_get_mono());

	lv_group_t *g = lv_group_create();

	lv_obj_t *roller1 = lv_roller_create(lv_scr_act(), NULL);
	lv_roller_set_options(roller1, "Audi\nMercedes\nOpel\nVolvo\nVW", LV_ROLLER_MODE_INIFINITE);
	lv_roller_set_visible_row_count(roller1, 1);
	//lv_roller_set_fix_width(roller1, 0);
	lv_obj_align(roller1, NULL, LV_ALIGN_IN_BOTTOM_LEFT, 5, -5);
	//lv_obj_set_event_cb(roller1, roller_event_cb);
	lv_group_add_obj(g, roller1);
//	lv_style_t *new_style = lv_roller_get_style(roller1, LV_ROLLER_STYLE_BG);

	lv_obj_t *spinbox = lv_spinbox_create(lv_scr_act(), NULL);
	lv_spinbox_set_digit_format(spinbox, 3, 0);
	lv_spinbox_set_value(spinbox, 433);
	//	new_style = lv_spinbox_get_style(spinbox, LV_SPINBOX_STYLE_BG);
	lv_spinbox_set_style(spinbox, LV_SPINBOX_STYLE_CURSOR, NULL);
	lv_group_add_obj(g, spinbox);

```

kisvegabor, 2020-04-14:
I see. The problem is the spinbox shows the cursor by default, and it should be hidden when it's added to group but not focused. However, the spinbox can't to notice that it's added to a group with `lv_group_add_obj`.

I suggest calling `lv_ta_set_cursor_type(spinbox, LV_CURSOR_HIDDEN);` when you created the spinbox.

> P.S. Why spinbox theme is not defined in MONO THEME?

The spinbox was missing form the mono theme. I've just added it.

kisvegabor, 2020-04-24:
The above mentioned workaround should solve the issue so I close it.

Wielebny666, 2020-04-24:
Dear kisvegabor,
second problem from the first post still wasn't resolved.

kisvegabor, 2020-04-24:
I see, sorry. I tried it but it worked normally for me.
Can you send a code snippet about how you created and configured the spinbox?

Wielebny666, 2020-04-25:
Maybe the problem occurs with encoder.

```
	lv_obj_t *spinbox = lv_spinbox_create(lv_scr_act(), NULL);
	lv_spinbox_set_digit_format(spinbox, 3, 0);
	lv_ta_set_cursor_type(spinbox, LV_CURSOR_HIDDEN);
```

kisvegabor, 2020-04-28:
@Wielebny666 
I've tested it and it happens because the range is larger than the max value that can be displayed.
Setting `lv_spinbox_set_range(obj, -999, 999);` solves the problem.
I'll update the docs to highlight it.

Wielebny666, 2020-04-28:
I suggested this at the end of the first post ;-)

kisvegabor, 2020-04-28:
Uhh, I should be more attentive. :sweat_smile: 

> The "digit_count" setting in lv_spinbox_set_digit_format() should check or change the "range_min" and "range_max" values

We should do the same in `lv_spinbox_set_range()` too.
Do you have time to implement it and send a PR?

Wielebny666, 2020-04-28:
The question is, which function should be given priority. Whether the last method invoked has priority.
Should information on parameter changes be in INFO, WARNING or TRACE? Should the change be hidden, as in the case of exceeding digit_count or separator_position?

kisvegabor, 2020-04-28:
A warning log is definitely a good idea. 

Let say the current digit format is `xx.y` and the user sets -99999..99999 range. It's not obvious how to change the digit format. Should it be `xxxx.y` (expand on the left) or `xx.yyy` (expand on the right) or anything between?

So it seems to me it'd be the best to always change range. What do you think?

kisvegabor, 2020-04-30:
Fixed by #1481 so this issue can be closed.

Thank you for the fix @Wielebny666

closed 2020-04-30 by kisvegabor