lvgl/lvgl #1442: Program crashes with window, LV_LAYOUT_GRID and multiple labels

tonystuart, 2020-04-12:
**Describe the bug**

The program reboots repeatedly when the window layout is set to grid and four or more rows that each contain a label and a text area are added to the window.

**To Reproduce**

The following program illustrates the problem:

```
    lv_obj_t * win = lv_win_create(lv_scr_act(), NULL);
    lv_win_set_title(win, "Window title");
    lv_win_set_layout(win, LV_LAYOUT_GRID);

    lv_obj_t * close_btn = lv_win_add_btn(win, LV_SYMBOL_CLOSE);
    lv_obj_set_event_cb(close_btn, lv_win_close_event_cb);
    lv_win_add_btn(win, LV_SYMBOL_SETTINGS);

    for (int i = 0; i < 4; i++) {
        lv_obj_t *name_label = lv_label_create(win, NULL);
        lv_label_set_long_mode(name_label, LV_LABEL_LONG_BREAK);
        lv_label_set_text(name_label, "Name");
        lv_obj_set_width(name_label, 140);

        lv_obj_t *value_ta = lv_ta_create(win, NULL);
        lv_obj_set_width(value_ta, 140);
        lv_ta_set_one_line(value_ta, true);
        lv_ta_set_text(value_ta, "Value");
    }
```

**Expected behavior**

Expected the window to contain four rows, each containing a label and a text area.

**Screenshots**

Screen is blank

**Additional context**

- I was able to work around the problem by setting the window layout to LV_LAYOUT_OFF prior to adding the widgets and LV_LAYOUT_GRID after adding the labels.

- I noticed that lv_cont_layout_grid is called frequently when the widgets are added. This makes sense (i.e. the parent needs to adjust its size to the children), but it may be worth adding a note to the documentation to turn container layout off prior to adding a large number of widgets.

- Ultimately, I decided to use LV_LAYOUT_PRETTY (with LV_PROTECT_FOLLOW on the text area). I set LV_LAYOUT_OFF prior to adding the widgets and then set it to LV_LAYOUT_PRETTY afterwards. This works really well for my use case!

- The target system is an ESP32 with 240x320 LCD touch screen. The log does not contain anything relevant at the point of error, but when the system reboots, it contains this:

```
W (248) boot: PRO CPU has been reset by WDT.
W (249) boot: WDT reset info: PRO CPU PC=0x4008abd6
```

- In the program where I originally encountered this problem, I was getting:

```
( Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c3f5  PS      : 0x00060d30  A0      : 0x80058b9e  A1      : 0x3ffb0ef0
A2      : 0x4038837c  A3      : 0x3f7e07d8  A4      : 0xffc28b9f  A5      : 0x00000080
A6      : 0xffc28b9f  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x4038837c
A10     : 0x000000ff  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000001
A14     : 0x00060021  A15     : 0x00ef0143  SAR     : 0x00000004  EXCCAUSE: 0x00000003
EXCVADDR: 0x4038837c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd

ELF file SHA256: f725847259c86abfe2b76ffb2cd6096b7266bfbdd1c42b7a55490f2297eb14f5

Backtrace: 0x4000c3f2:0x3ffb0ef0 |<-CORRUPTED
```

- The problem also occurs if I use two labels instead of a label and a text area. It does not occur if I use two buttons (without labels).

embeddedt, 2020-04-12:
Reproduced on the PC simulator.

embeddedt, 2020-04-12:
Interesting. I can reproduce this on the `dev-7.0` branch but not with the latest `master` branch. Which version are you using?

tonystuart, 2020-04-12:
Thanks... I just tried the latest master (at commit 7e4e7) and it looks like the problem still occurs for me on the ESP32. I don't have the PC simulator set up.

kisvegabor, 2020-04-13:
Hi, 

I've fixed it in 0c309e2
Does it work well for you too?

closed 2020-04-13 by kisvegabor

tonystuart, 2020-04-13:
It works... many thanks!

commit 0c309e22 (2020-04-13): fix grid layot. Fixes #1442