lvgl/lvgl #1446: label to show odd number Chinese characters

TANK-RICHARD, 2020-04-14:
# label to show odd number Chinese characters

When the number of Chinese characters is even, label will show them correctly, while the number is odd, the last one will not be showed, and a question mark "?" instead.
for example:
         If the text of label is "正在测试", it is OK;
         If the text of label is "测试中", it is not OK, will show "测试?"

TANK-RICHARD, 2020-04-14:
my used font could be downloaded: 
http://www.waveshare.net/study/portal.php?mod=attachment&id=2045

kisvegabor, 2020-04-14:
Hi,

I've tried it but saw that seemingly the characters of the `正在测试` is not included in the font because the font you have attached has only these Chinese characters:`你堂好字展微是显汉示课这里雪`

Are you sure you have sent the correct font file?

TANK-RICHARD, 2020-04-15:
> Hi,
> 
> I've tried it but saw that seemingly the characters of the `正在测试` is not included in the font because the font you have attached has only these Chinese characters:`你堂好字展微是显汉示课这里雪`
> 
> Are you sure you have sent the correct font file?

Hi,

sorry for that, the attached file is mine.

[Chinese_character_common.zip](https://github.com/littlevgl/lvgl/files/4478542/Chinese_character_common.zip)

TANK-RICHARD, 2020-04-15:
This question can be avoided through hex UTF-8 code temporarily, for example, "测试中", using lv_label_set_text(label, "\xE6\xB5\x8B\xE8\xAF\x95\xE4\xB8\xAD");

kisvegabor, 2020-04-17:
It works for me with 3 characters too. If using UTF-8 characters directly works then probably the problem is related to your IDEs encoding settings.

You can test the encoding with this small code:
```c
    uint8_t i;
    const char * txt = "测试中";
    for(i=0; txt[i] != '\0'; i++) {
        printf("%02x ", (uint8_t)txt[i]);
    }
    printf("\n");
```
I got: `e6 b5 8b e8 af 95 e4 b8 ad`

embeddedt, 2020-05-09:
I'm closing this as there was no response from the OP.

closed 2020-05-09 by embeddedt