lvgl/lvgl #1447: animation does not work in tabview

SomeMaker, 2020-04-14:
**Describe the bug**

When assigning the initial position of the animation to a negative value, the animation does not work. This appears when the parent of the object is the tabs. 

**To Reproduce**

I used an example from PC simulator: demo_create();. 
In the function "text_area_event_handler" I set a.start = -500; (file: demo.c)
I was expecting the keyboard to exit the top of the screen. But keyboard did not come out at all

Then I printed the “Y” coordinate to see where the “kb” keyboard is drawn and where the “tab” tab is drawn. The "Y" coordinate of the keyboard was supposed to increase from -500 to about 170. But the keyboard went the other way
[image]

It also appears in the x-axis animation when a.start = -500. But this does not happen when the parent of the object is not a tab.
Positive animation works as expected

kisvegabor, 2020-04-16:
Hi,
I've just fixed it with this: https://github.com/littlevgl/lv_examples/commit/8194fe567dd9a287739df7a629230cbd8ef9256a#diff-d7792eb18e8fd5d455030c5c5a28978aR138

SomeMaker, 2020-04-16:
thanks, now works

closed 2020-04-16 by SomeMaker