lvgl/lvgl #1449: Determination of pointer size is unreliable

dykeag, 2020-04-15:
In src/lv_misc/lv_types.h, the preprocessor logic used to determine the size of a pointer (and therefore set the typedef of `lv_uintptr_t`) is unreliable. The logic is based on preprocessor macros which are compiler dependant. It would be much more reliable to use the definitions provided in `<stdint.h>`.  Sadly, those are only available in C99 and newer, but we can check availability of this as well in the preprocessor. 

The code for determining architecture should probably look like this: 
```
#if __STDC_VERSION__ >= 199901L  // If c99 or newer...

  #include <stdint.h> 
 
   // Use the definition of uintptr_t directly
  typedef uintptr_t lv_uintptr_t

  #if UINTPTR_MAX > 0xFFFFFFFF
    #define LV_ARCH_64
  #endif 

#else 
  // existing compiler-based determination 
#endif
```

embeddedt, 2020-04-15:
Your solution looks good to me. Can you send a pull request?

dykeag, 2020-04-15:
Done
https://github.com/littlevgl/lvgl/pull/1450

closed 2020-04-15 by dykeag