lvgl/lvgl #1451: Despite documentation, body opacity determines arc opacity

dykeag, 2020-04-15:
I am not sure if it is a bug or if it is intentional, but when drawing an arc (lv_draw_arc), the opactity from `style->body.opa` is used to determine the opacity of the line drawn. I has assumed that `style->line.opa` would have been used since the other parameters for the arc are taken from `style->line`

The real problem is that the [documentation for arcs](https://docs.littlevgl.com/en/html/object-types/arc.html) does not mention that this is the behaviour. 

Either the code or the documentation should be changed.

embeddedt, 2020-04-15:
The style system is being rewritten in 7.0 and I'm not sure if this will still be an issue there, so I think updating the documentation makes more sense. @kisvegabor Do you agree?

kisvegabor, 2020-04-17:
@embeddedt Yes, I agree. 
I've just updated the docs.

closed 2020-04-17 by kisvegabor