lvgl/lvgl #1453: Image button from Filesystem issue

albertomercurio, 2020-04-15:
**Describe the bug**

I need to use image button through littlevgl filesystem. It reads the image from the spiffs, but a “No” word appears in the top of the image when i press it. (See this [video](https://streamable.com/otr08b))

**To Reproduce**

I insert only the significant part. For the drive implementation:
```
void spiffs_drv_init(){
  lv_fs_drv_init(&drv);                     /*Basic initialization*/

  drv.letter = 'D';                         /*An uppercase letter to identify the drive */
  drv.file_size = sizeof(File);   /*Size required to store a file object*/
  drv.open_cb = my_open_cb;                 /*Callback to open a file */
  drv.close_cb = my_close_cb;               /*Callback to close a file */
  drv.remove_cb = my_remove_cb;             /*Callback to remove a file */
  drv.read_cb = my_read_cb;                 /*Callback to read a file */
  drv.write_cb = my_write_cb;               /*Callback to write a file */
  drv.seek_cb = my_seek_cb;                 /*Callback to seek in a file (Move cursor) */
  drv.tell_cb = my_tell_cb;                 /*Callback to tell the cursor position  */

  lv_fs_drv_register(&drv);                 /*Finally register the drive*/
}

lv_fs_res_t my_open_cb(lv_fs_drv_t *drv, void *file_p, const char *fn, lv_fs_mode_t mode){
  (void) drv; /*Unused*/

  if (f) {
    return LV_FS_RES_OK;
  }

  char buf[100];
  sprintf(buf,"/%s",fn);

  f = SPIFFS.open(buf, mode == LV_FS_MODE_WR ? FILE_WRITE : FILE_READ);

  if(!f || f.isDirectory()){
    return LV_FS_RES_UNKNOWN;
  } else{
    return LV_FS_RES_OK;
  }
}

lv_fs_res_t my_close_cb(lv_fs_drv_t *drv, void *file_p){
  (void) drv; /*Unused*/

  f.close();
  return LV_FS_RES_OK;
}

lv_fs_res_t my_read_cb(lv_fs_drv_t *drv, void *file_p, 
                        void *buf, uint32_t btr, uint32_t *br){
  (void) drv; /*Unused*/

  *br = f.read((uint8_t*)buf, btr);
  Serial.println("READ");
  
  return LV_FS_RES_OK;
}

lv_fs_res_t my_write_cb(lv_fs_drv_t *drv, void *file_p, 
                        const void *buf, uint32_t btw, uint32_t *bw){
  (void) drv; /*Unused*/

  *bw = f.write((const uint8_t*)buf, btw);
  return LV_FS_RES_OK;
}

lv_fs_res_t my_seek_cb(lv_fs_drv_t *drv, void *file_p, uint32_t pos){
  (void) drv; /*Unused*/

  f.seek(pos);
  return LV_FS_RES_OK;
}

lv_fs_res_t my_tell_cb(lv_fs_drv_t *drv, void *file_p, uint32_t *pos_p){
  (void) drv; /*Unused*/

  uint32_t tmp = f.position();
  pos_p = &tmp;
  
  return LV_FS_RES_OK;
}

lv_fs_res_t my_remove_cb(lv_fs_drv_t *drv, const char *fn){
  (void) drv; /*Unused*/

  char buf[100];
  sprintf(buf,"/%s",fn);

  if(SPIFFS.remove(buf)){
    return LV_FS_RES_OK;
  } else{
    return LV_FS_RES_UNKNOWN;
  }
}
```

For the gui:
```
void set_styles(){
  lv_style_copy(&style_upper_bar, &lv_style_scr);
  style_upper_bar.body.main_color = LV_COLOR_BLACK;
  style_upper_bar.body.grad_color = LV_COLOR_BLACK;
  style_upper_bar.body.opa = LV_OPA_50;

  lv_style_copy(&style_img_pr, &lv_style_plain);
  style_img_pr.image.color = LV_COLOR_BLACK;
  style_img_pr.image.intense = LV_OPA_50;
  style_img_pr.text.color = lv_color_hex3(0xaaa);
}

void start_gui(){
  set_styles();

  // ##################################
  // Home button on the left side bar
  lv_obj_t *test = lv_imgbtn_create(scr, NULL);
  lv_imgbtn_set_src(test, LV_BTN_STATE_PR, "D:/background_small.bin");
  lv_imgbtn_set_src(test, LV_BTN_STATE_REL, "D:/background_small.bin");
  lv_imgbtn_set_src(test, LV_BTN_STATE_TGL_PR, "D:/background_small.bin");
  lv_imgbtn_set_src(test, LV_BTN_STATE_TGL_REL, "D:/background_small.bin");
  lv_imgbtn_set_src(test, LV_BTN_STATE_INA, "D:/background_small.bin");
  lv_imgbtn_set_style(test, LV_BTN_STATE_PR, &style_img_pr);
  lv_imgbtn_set_style(test, LV_BTN_STATE_TGL_PR, &style_img_pr);
  lv_imgbtn_set_toggle(test, true);
  lv_obj_align(test, NULL, LV_ALIGN_CENTER, 10, 0);
  lv_btn_set_state(test, LV_BTN_STATE_TGL_REL);
  //lv_btn_set_ink_in_time(test, 100);
  // ##################################
  
  upper_bar = lv_obj_create(scr, NULL);
  lv_obj_set_size(upper_bar, screenWidth, 30);
  lv_obj_set_style(upper_bar, &style_upper_bar);
  lv_obj_align(upper_bar, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 0);
  
  // Home button on the left side bar
  home_btn = lv_imgbtn_create(upper_bar, NULL);
  lv_imgbtn_set_src(home_btn, LV_BTN_STATE_PR, "D:/home.bin");
  lv_imgbtn_set_src(home_btn, LV_BTN_STATE_REL, "D:/home.bin");
  lv_imgbtn_set_src(home_btn, LV_BTN_STATE_TGL_PR, "D:/home.bin");
  lv_imgbtn_set_src(home_btn, LV_BTN_STATE_TGL_REL, "D:/home.bin");
  //lv_imgbtn_set_src(home_btn, LV_BTN_STATE_INA, "D:/home.bin");
  lv_imgbtn_set_style(home_btn, LV_BTN_STATE_PR, &style_img_pr);
  lv_imgbtn_set_style(home_btn, LV_BTN_STATE_TGL_PR, &style_img_pr);
  lv_imgbtn_set_toggle(home_btn, true);
  lv_obj_align(home_btn, NULL, LV_ALIGN_IN_LEFT_MID, 10, 0);
  lv_btn_set_state(home_btn, LV_BTN_STATE_TGL_REL);
  lv_btn_set_ink_in_time(home_btn, 100);

  // Clock on the top of the screen
  clock_btn_label = lv_label_create(upper_bar, NULL);
  //lv_obj_set_size(clock_btn_label, 80, 30);
  lv_label_set_recolor(clock_btn_label, true);
  clock_icon_update();

  // Wifi button
  wifi_btn = lv_btn_create(upper_bar, NULL);
  lv_btn_set_toggle(wifi_btn, true);
  lv_btn_set_state(wifi_btn, LV_BTN_STATE_TGL_REL);
  lv_obj_set_event_cb(wifi_btn, wifi_btn_event);
  lv_obj_set_size(wifi_btn, 30, 30);
  lv_obj_align(wifi_btn, NULL, LV_ALIGN_IN_RIGHT_MID, -100, 0);
  lv_btn_set_ink_in_time(wifi_btn, 100);

  /*Create a label on the object. The label's style is NULL by default*/
  wifi_btn_label = lv_label_create(wifi_btn, NULL);
  lv_obj_align(wifi_btn_label, NULL, LV_ALIGN_CENTER, 0, 0);
  lv_label_set_recolor(wifi_btn_label, true);
  lv_label_set_text(wifi_btn_label, "#ffffff " LV_SYMBOL_WIFI "#");

  battery_label = lv_label_create(upper_bar, NULL);
  lv_obj_set_style(battery_label, &lv_style_transp);
  lv_obj_align(battery_label, NULL, LV_ALIGN_IN_RIGHT_MID, -50, 0);
  lv_label_set_recolor(battery_label, true);
  
  battery_label_level = lv_label_create(upper_bar, NULL);
  lv_label_set_recolor(battery_label_level, true);
  battery_icon_update();
}
```

And the setup part:
```
void boot_func(){
  //Set boot background
  static lv_style_t boot_background;
  lv_style_copy(&boot_background, &lv_style_plain);
  boot_background.body.main_color = lv_color_hex(0x185a9d);
  boot_background.body.grad_color = lv_color_hex(0x43cea2);
  lv_obj_set_style(scr, &boot_background);

  // Start logo
  lv_obj_t * loader = lv_preload_create(scr, NULL);
  lv_obj_align(loader, NULL, LV_ALIGN_CENTER, 0, 50);

  lv_obj_t * boot_text = lv_label_create(scr, NULL);

  lv_label_set_text(boot_text, "Mounting the drive...");
  lv_label_set_align(boot_text, LV_LABEL_ALIGN_CENTER);
  lv_obj_align(boot_text, NULL, LV_ALIGN_CENTER, 0, 100);
  delay_ms(2000);
  if(!SPIFFS.begin(true)){
    lv_label_set_text(boot_text, "Mount failed...");
    lv_label_set_align(boot_text, LV_LABEL_ALIGN_CENTER);
    lv_obj_align(boot_text, NULL, LV_ALIGN_CENTER, 0, 100);
    while(true){
      delay_ms(1000);
    }
  }
  spiffs_drv_init();

  lv_obj_del(loader);
  lv_obj_del(boot_text);
}
```

**Expected behavior**

A clear image button without the "No" word.

**Screenshots**
See this [video](https://streamable.com/otr08b)

kisvegabor, 2020-04-17:
Where does `f` come from in the driver? 
In the `file_p` parameter you get a pointer to a `sizeof(File);` sized memory which you can use as a file descriptor. This way, you can keep only one file opened which is problematic because the last image is always kept open (for performance reasons) but meanwhile some data is read from the next image. However, if you can have only one image opened at the same time things mess up.

albertomercurio, 2020-04-17:
`f` is a global variable, declared with the SPIFFS and FS Arduino libraries with:
```c
#include <SPIFFS.h>
#include <FS.h>

File f;
```
So, can i open only one image per time?

kisvegabor, 2020-04-17:
> So, can i open only one image per time?

This way yes, but lvgl opens more images at the same time which causes the issues.

albertomercurio, 2020-04-17:
So, what can i do to open multiple images in the same time?

embeddedt, 2020-04-17:
I think your driver needs to be set up to handle more than one `File` object at a time.

barbiani, 2020-04-17:
How about using a sprite map?

albertomercurio, 2020-04-17:
> I think your driver needs to be set up to handle more than one `File` object at a time.

Do you know how to do that? However i niticed that using a custom background image from SPIFFS the problem disappears. (Instead using no background or using a native color background)

kisvegabor, 2020-04-20:
In `my_open_cb` (and other `lv_fs` functions too) the `file_p` parameter is pointer to a `File` variable. So you can think of it a `File * file_p`. 

I haven't tested it, but at first something like this should work:
```c
lv_fs_res_t my_open_cb(lv_fs_drv_t *drv, void *file_p, const char *fn, lv_fs_mode_t mode){
  (void) drv; /*Unused*/
  
  File * fp = file_p;
  if (*fp) {
    return LV_FS_RES_OK;
  }

  char buf[100];
  sprintf(buf,"/%s",fn);

  (*fp) = SPIFFS.open(buf, mode == LV_FS_MODE_WR ? FILE_WRITE : FILE_READ);

  if(!(*fp) || (*fp).isDirectory()){
    return LV_FS_RES_UNKNOWN;
  } else{
    return LV_FS_RES_OK;
  }
}
```

And play this in the other function too.

closed 2020-05-13 by kisvegabor