lvgl/lvgl #1454: Better GPU support

kisvegabor, 2020-04-16:
I'm thinking about how could we support GPU features better. I've realized that the current "callback in display driver" doesn't work well here because there are too many cases. (fill, fill with opacity, fill with mask, copy and clip an image, copy  and clip an image with opacity, copy  and clip with pixel format conversation, blend and clip, blend with ...). If we consider more advanced GPUs that can draw lines, arcs etc it gets even worse.

Although there are a lot of cases there not that many GPUs. I think the most common ones are ST's DMA2D and NXP's PXP. (probably there is a foe other I don't know but still important) So what if we support those GPUs in lvgl. I mean we can add GPU support in lvgl itself inside the drawing algorithms where and how it fits the best. The GPU to use could be selected in `lv_conf.h`.

I made some experiments with DMA2D:
- [lv_conf_templ.h](https://github.com/littlevgl/lvgl/blob/speed/gpu/lv_conf_template.h#L181)
- [lv_draw_blend.c](https://github.com/littlevgl/lvgl/blob/speed/gpu/src/lv_draw/lv_draw_blend.c#L333)

What do you think?

embeddedt, 2020-04-16:
It looks like a good idea, although having a stack of custom GPU functions inside the core library will quickly become unmanageable, in my opinion. Perhaps these could be extracted into dedicated functions (`stm32_gpu_fill_normal`, etc.). That way there are less lines of code added directly within the generic functions.

Have you gotten a chance to investigate performance with this method yet?

kisvegabor, 2020-04-16:
I agree. I was afraid that the functions will require too many parameters but seems manageable.
I've moved DMA2D to a separate file as you suggested. 6f87d29a

Regarding the performance: 
- fill: no real difference
- copy image (rgb) lvgl faster with 20%
- copy image with opacity: DMA2D is faster with 20%
- fill with mask (in DMA2D it's fill with A8 color format where the intensity is the mask): lvgl 2x faster.

These numbers may vary depending on the test case but it should be something like this.

In one important case DMA2D is still not used: blend ARGB8888 images. Lvgl needs some updates to allow passing ARGB8888 images even if the color depth is 16 bit. Here DMA2D should be significantly faster.

kisvegabor, 2020-04-16:
I've just added DMA2D support for the mentioned ARGB8888 case if `LV_COLOR_DEPTH == 32`.
I've moved 6 pcs 100x100 ARGB images and measured this:
- lvgl:
  - no opacity: 132 FPS
  - opacity: 92 FPS
- DMA2D:
  - no opacity: 200 FPS
  - opacity: 200 FPS

So DMA2D can be 50..100% faster in this case.

embeddedt, 2020-04-16:
I still find it strange that our software algorithm is faster than DMA2D for some cases. :confused: I would expect hardware to always be faster.

On average, though, it seems like DMA2D results in better performance.

One suggestion: it might be better to use `HAL_DMA2D_PollForTransfer` with a timeout of 0, and add a hook that people can use to allow the processor to sleep/yield. Otherwise, the function just [spins](https://android.googlesource.com/trusty/lk/common/+/8007cb550a559cb7e6aa347b8e0c30c30bcee1ee/external/platform/stm32f7xx/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_dma2d.c#616) in a loop consuming CPU cycles.

kisvegabor, 2020-04-17:
> I still find it strange that our software algorithm is faster than DMA2D for some cases. confused I would expect hardware to always be faster.

I also wonder it but it seems that the case.

> On average, though, it seems like DMA2D results in better performance.

Yes. I've disabled the use of DMA2D in the mentioned "fill with mask" case.

> One suggestion: it might be better to use HAL_DMA2D_PollForTransfer with a timeout of 0, and add a hook that people can use to allow the processor to sleep/yield. Otherwise, the function just spins in a loop consuming CPU cycles.

I was also thinking about it. The same hook can be used while lvgl waits to flush the buffer. However, I'm not sure what can the user does in that hook. Yielding the task is not a good idea in my opinion because if we lose even 0.1 ms on every yield, in summary, it could be easily 10-20 ms loss. So the task to perform here should be something really short, e.g. check some I/Os, quickly process some bytes from UART or so. What do you think?

embeddedt, 2020-04-17:
I think the user should decide. If a DMA2D operation takes at least 1-2ms I would expect that yielding wouldn't have a significant performance impact (especially since something else can be processed in parallel).

kisvegabor, 2020-04-17:
> I think the user should decide. If a DMA2D operation takes at least 1-2ms I would expect that yielding wouldn't have a significant performance impact 

It assumes that the tick period is < 1ms. For example, in ESP32 the default tick period is 10 ms. 
Anyway, we can really leave it up to the user. What do you think about a `wait_cb` in the display driver?

embeddedt, 2020-04-17:
> For example, in ESP32 the default tick period is 10 ms.

Wow; that seems long!

> What do you think about a `wait_cb` in the display driver?

That seems like a good way of doing it. It should also be fully compatible with MicroPython.

nacyzhouw, 2020-04-18:
hi, kisvegabor. Have you ever considered about supporting GLES since now more and more IOT chips have GPU to do more thing.

kisvegabor, 2020-04-18:
@embeddedt Great, I'll add it next week.

@nacyzhouw 
That really would be great! However unfortunately, I have no experience with it. 
Have you already used it?

nacyzhouw, 2020-04-20:
@kisvegabor  I have't used it already since it would be a lot of work and the littevgl might need to be restructed.

kisvegabor, 2020-04-22:
I mean, have you used GLES not only with lvgl but in general?
What part should be restructured? What problems and incompatibilities do you see?

excitedbox, 2020-04-23:
If LVGL already works with SDL shouldn´t you be able to use that to render with the GPU? Or maybe just make a port for using OpenGL or WebGL. WebGL would be interesting because it would allow for making web apps that run in the browser and you can easily interface with Javascript through WebGL. 

You could make an app in Javascript and use LVGL for your GUI so you keep your branding between different devices without having to redesign everything. Would also make it quite easy to build GUI builders for embedded systems.

embeddedt, 2020-04-23:
@excitedbox LittlevGL can already run in a browser using Emscripten. See https://littlevgl.com/live-demo and https://github.com/littlevgl/lv_sim_emscripten.

barbiani, 2020-04-23:
We have to keep in mind that low end hardware does not have opengl, that
usually means a full linux setup with much more expensive parts.

In my experience, the current code needs over 400mhz cpu clock to software
render the screens with enough speed to feel fluid.

kisvegabor, 2020-04-24:
> In my experience, the current code needs over 400mhz cpu clock to software
render the screens with enough speed to feel fluid.

At which resolution? I found that 200 MHz is usually enough for an average UI on 800x480 resolution.

kisvegabor, 2020-04-24:
I've added `wait_cb`. See [here](https://github.com/littlevgl/lvgl/blob/speed/gpu/src/lv_gpu/lv_gpu_stm32_dma2d.c#L206).

barbiani, 2020-04-24:
> > In my experience, the current code needs over 400mhz cpu clock to software
> > render the screens with enough speed to feel fluid.
> 
> At which resolution? I found that 200 MHz is usually enough for an average UI on 800x480 resolution.

320x240 in the esp32@240MHz and 480x272@80MHz. The simulator is spetacular.

I state this based in v7 and lv_demo_widgets. I see about 12 fps during the tab transitions.

Edge flash is another really slow performing effect.

I am not complaining 😁. Saying it just to help new users selecting their dev boards.

kisvegabor, 2020-04-24:
@barbiani 
A lot depends on how you drive the TFT and the configuration of lvg. With STM32F7 (200 MHz) and 800x480 TFT I never see less than 20 FPS. By giving more RAM to lvgl the FPS is always more than 25 for me.

Could you open a new issue to describe your experiences in more detail? I'd be very curious to hear how the new version performs.

embeddedt, 2020-04-27:
@kisvegabor I'm trying to get the `speed/gpu` branch working on my board. It looks like there is memory corruption happening somewhere, as global variables are ending up with weird values.

kisvegabor, 2020-04-27:
@embeddedt I think I've fixed it here. 843fa5e3

embeddedt, 2020-04-27:
Thanks! Unfortunately, it's now freezing in the middle of running test 8 (border rounded). I'm trying to find the cause.

embeddedt, 2020-04-27:
It's because the cache was disabled.

https://github.com/littlevgl/lvgl/blob/843fa5e30479d0e04eb62c452b74a54a1c863f9a/src/lv_gpu/lv_gpu_stm32_dma2d.c#L102

Maybe we should check whether the cache is enabled before we do any operations on it.

embeddedt, 2020-04-27:
Nothing is dropping below 20 FPS anymore, so it looks like there is a substantial boost to performance here. I am noticing some weird artifacts on the display however. Sometimes a row seems to get skipped during the update of the display, so you either see white or remnants of a prior object.

kisvegabor, 2020-04-28:
> Maybe we should check whether the cache is enabled before we do any operations on it.

Good point. Fixed here: a54d6a17

> Nothing is dropping below 20 FPS anymore, so it looks like there is a substantial boost to performance here. 

What caused the boost? only the better GPU usage, or have enabled caching too?

> I am noticing some weird artifacts on the display however. Sometimes a row seems to get skipped during the update of the display, so you either see white or remnants of a prior object.

Could you upload your project somewhere? I have a F746 board to reproduce it.

embeddedt, 2020-04-28:
> What caused the boost? only the better GPU usage, or have enabled caching too?

I enabled both. I'll disable caching now and see how much performance is lost.

> Could you upload your project somewhere? 

Sure. https://github.com/littlevgl/lv_port_stm32f746_disco_sw4stm32/tree/speed/gpu_f746

embeddedt, 2020-04-28:
The artifacts go away once caching is disabled... but so do the high frame rates I was previously seeing.

Nonetheless, the use of DMA2D still seems to improve frame rates when drawing images. There's essentially no difference between an opaque image and a translucent one (which I would expect, since this is what the hardware was made for).

kisvegabor, 2020-04-28:
@embeddedt 
I found the issue. `SCB_CleanInvalidateDCache();` needs to be called at the beginning of the flush function too.

I've noticed that the "old" gpu callbacks were enabled instead of enabling `LV_USE_GPU_STM32_DMA2D`. Was it only the last test case you tried?

> Nonetheless, the use of DMA2D still seems to improve frame rates when drawing images. There's essentially no difference between an opaque image and a translucent one (which I would expect, since this is what the hardware was made for).

It seems we are too good :laughing: 
But seriously, I also don't really understand it. You can try enabling [this](https://github.com/littlevgl/lvgl/blob/14814d535e1d319a809b3467677b391e42c14cd8/src/lv_draw/lv_draw_blend.c#L376) too. In this case DMA2D is much slower than lvgl.

kisvegabor, 2020-04-29:
I've merged `speed/gpu` to `dev-7.0` because it shouldn't affect global behavior and we are getting close o release candidate state.

kisvegabor, 2020-04-30:
Let's continue in https://github.com/littlevgl/lvgl/issues/1483 if something comes up.

closed 2020-04-30 by kisvegabor