lvgl/lvgl #1455: Setting up littlegvl on STM32CubeIDE

WoulaMaker, 2020-04-16:
Hi,

I' m trying to set up a new littlevgl project on STM32CubeIDE that target a STM32F429I-DISC1 board. I added the lvgl folder to my project. I also moved the lv_conf.h file next to the lvgl folder, as requested by the documentation. In the lv_conf.h. I also modified the file to set the screen resolution and enable the code. However, when i call the lv_init() function in my program, for initialisation, an error is send during build, saying "... undefine reference to 'lv_init' ... ". I also included lv_conf.h in my main.c file. So i don' t really understand the problem. 

Can you help me please ?

embeddedt, 2020-04-16:
It sounds like the source code in the `lvgl` folder is not being compiled. Try clicking on the project in the sidebar and then pressing F5 to refresh it (Eclipse requires this in order to pick up on the files).

You could also try our pre-made [STM32F429 project](https://github.com/littlevgl/lv_port_stm32f429_disco_sw4stm32), which is compatible with STM32CubeIDE.

WoulaMaker, 2020-04-16:
It works perfectly with the example. Thank you. I'll try to solve my problem from it.

kisvegabor, 2020-04-17:
>  I'll try to solve my problem from there.

I close this issue then.

closed 2020-04-17 by kisvegabor