lvgl/lvgl #1456: Examples folder

David-OConnor, 2020-04-17:
It would be nice to have an examples folder. The examples in the readme are excerpts that don't compile, ie are missing the boilerplate and setup. This makes it tough to get started.

embeddedt, 2020-04-17:
The examples in the [lv_examples](https://github.com/littlevgl/lv_examples) repository might be closer to what you're looking for. For example, the [button example](https://github.com/littlevgl/lv_examples/blob/master/lv_tests/lv_test_objx/lv_test_btn/lv_test_btn.c) can be used simply by calling `lv_test_btn_1`.

Initializing a display driver is platform-specific and is thus not mentioned in the README file or the examples (which are intended to be platform-agnostic). We do have a number of [pre-made projects](https://github.com/littlevgl) (notably the [PC simulators](https://docs.littlevgl.com/en/html/get-started/pc-simulator.html)), [porting templates](https://github.com/littlevgl/lvgl/tree/master/porting), and [documentation](https://docs.littlevgl.com/en/html/porting/index.html) that describe how to do that.

David-OConnor, 2020-04-17:
That's exactly what I was looking for; thank you!

closed 2020-04-17 by David-OConnor