lvgl/lvgl #1461: lv_list_add_btn is very slow

xennex22, 2020-04-18:
Adding 55 items to a list takes over 1 second.  About 4 days ago it was fine.

The time to add each item starts at 4ms (STM32F7 @ 216 MHz) and increases to 40ms at item 55.

Sorry, I can't narrow anything down at the moment.  When I get more time I'll try to see what it is.

embeddedt, 2020-04-18:
Educated guess: perhaps https://github.com/littlevgl/lvgl/commit/6563f2fa0d991f5834deabf936a744fb90f854ea is the cause.

xennex22, 2020-04-18:
I tried rolling back the mem defrag changes but it did not help.

Here's the code I'm using to test
```
lv_obj_t * list = lv_list_create(screen, NULL);
lv_page_set_scrlbar_mode(list, LV_SCRLBAR_MODE_AUTO);
lv_obj_set_style_local_size(list, LV_PAGE_PART_SCRLBAR, LV_STATE_DEFAULT, LV_DPI / 4);
lv_obj_set_style_local_border_opa(list, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, LV_OPA_TRANSP);
lv_obj_align(list, screen, LV_ALIGN_IN_TOP_LEFT, 20, 50);
	
for(int i=0; i<100; i++) {
    char buf[200];
    snprintf(buf, sizeof(buf), "Item %d", i);
    lv_obj_t * list_btn;
    list_btn = lv_list_add_btn(list, NULL, buf);
}
```

Here's the rough timings
```
676ms lv_btn_create 
  349ms lv_cont_create
    254ms lv_obj_create
      5ms lv_ll_ins_head
      254ms lv_theme_apply
        5ms lv_style_list_add_style
          250ms lv_obj_refresh_style
            250ms obj->parent->signal_cb(obj->parent, LV_SIGNAL_CHILD_CHG, NULL);
              1ms lv_list_btn_signal
              ? 
```

kisvegabor, 2020-04-18:
I've tested it and it seems it due to the new `margin` style property which needs to be asked several times for every object. I'm thinking about a solution.

kisvegabor, 2020-04-18:
By the way, how did you create this timing measurement?

xennex22, 2020-04-18:
Stepping through the hardware debugger and looking at one of the debug stopwatches (uVision). Not that accurate, but it can help isolate the problem.

xennex22, 2020-04-18:
About half the total time is the child loop in `lv_cont_refr_autofit`

To create one list button:

`lv_obj_create` calls `lv_cont_refr_autofit` five times via `lv_theme_apply` then another time in `signal_cb`

`lv_cont_create` calls `lv_cont_refr_autofit` twice.

'lv_btn_create' calls `lv_cont_refr_autofit` 10 times (`lv_obj_set_size` `lv_btn_set_layout` `lv_theme_apply`)

`lv_list_add_btn` calls `lv_cont_refr_autofit` 3 times.

Then another 16 times to create the label.  So about 38 times.  The issue is that `lv_cont_refr_autofit` loops through all children which is a little slow but is called almost 40 times.

I did not measure the calls for the second button creation, but did measure calls to the `lv_cont_refr_autofit`.

To create a list with 45 buttons `lv_cont_refr_autofit` gets called 5980 times.

To create a list with 90 buttons `lv_cont_refr_autofit` gets called 20020 times.

xennex22, 2020-04-18:
Here's the results of a quick sampling profile I did with debug off.
[image]
Calls to gets styles is eating up most of the cycles, but the real problem is the number of calls to align functions which then call style functions, as outlined above.

kisvegabor, 2020-04-18:
@xennex22  I've fixed it. https://github.com/littlevgl/lvgl/commit/b7cb6f9995fbcbac4c7c294218c120d66f690bc8
Before the margin update, it was 483 ms to create the list, now it's only 280 ms.

I love this profiler. Somehow it doesn't work for me in CubeIDE. :disappointed:

xennex22, 2020-04-19:
Tested with the changes:
To create a list with 45 buttons `lv_cont_refr_autofit` gets called 2092 times.
To create a list with 90 buttons `lv_cont_refr_autofit` gets called 5831 times.
And the time per item to create is about four times faster than before (45th item = 14ms), which is about the same change in time as the decrease in calls to `lv_cont_refr_autofit`.

So perhaps I should close this now but I feel that there are improvements to be made in the number of calls.

closed 2020-04-19 by xennex22

kisvegabor, 2020-04-20:
> So perhaps I should close this now but I feel that there are improvements to be made in the number of calls.

I agree, however at first look can't see how to optimize it. Anyway, I keep it in mind!