lvgl/lvgl #1462: gauge change position when changing needle value

gaugetester, 2020-04-19:
> # Important: issues that don't use this template will be ignored/closed.

If you create a new gauge widget it shows up nice and tidy in the middle of the screen.
As soon as you change the value of the needle the complete widget is moved to to left upper
border of the screen.

A clear and concise description of what the bug is.

**To Reproduce**

	scr = lv_disp_get_scr_act(NULL);

	static lv_color_t needle_colors[1];
	needle_colors[0] = LV_COLOR_BLUE;

	mgauge = lv_gauge_create(scr,NULL);

	lv_gauge_set_needle_count(mgauge, 1, needle_colors);
	lv_obj_set_size(mgauge, 220,220);
	lv_obj_align(mgauge, NULL, LV_ALIGN_CENTER, 0, 20);

That works okay,

lv_gauge_set_value(mgauge, 0, 60.0);

But now it jumps.

**Expected behavior**

Of course only the needle should change.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Additional context**

I used the latest software version from git

embeddedt, 2020-04-19:
Is this a custom driver? I see artifacts of the previous gauge which leads me to think there's something up with the way your driver places the image on the framebuffer.

gaugetester, 2020-04-19:
The artefact are the problem, because it means that the gauge is drawn on the wrong place and not in the middle. When it is created it is build correct at the middle, but when you move the needle 
it needs to be refreshed and it is drawn again and this did not work.

Perhaps a little bit explanation. :)

I wrote my own widget for my own hardware (Gecko+320x240TFT with ILI9341) and run in the problem. Of course I thought that the reason is my own widget. So I wrote an emulation layer with Qt. lvgl is now running on my PC inside a window with the same problem.
I did this because it is quite easy to debug on so powerful hardware. (quite cool!)

I found this must be some problem around lv_core/lv_refr.c  with lv_inv_area because the value of scr_area are strange. So later at line 350 of lv_ref.c

           vdb->area.x1 = area_p->x1;
            vdb->area.x2 = area_p->x2;

it becomes a wrong value and shift the widget. 

For testing purpose I switched of my new widget and use the lv_gauge inside of lvgl and found the same behavier. For this reason I think I am not guilty. :-)
Unfortunatly it is hard to understand the source if you are not the writer, because there are only very few comments and many pointers.

embeddedt, 2020-04-19:
Can you try it with the [PC simulator](https://github.com/littlevgl/lv_sim_eclipse_sdl)? That's our test bench for seeing whether it's a library issue or not.

kisvegabor, 2020-04-19:
It works well in the simulator for me:
[image]

Do you use the latest `master` branch?

gaugetester, 2020-04-19:
I dont think I can try it with the simulator because I am using Centos as operating system.

I did a git pull 1-2weeks ago. 

It is strange that it is working for you. I will do some more investigation on my side...

embeddedt, 2020-04-19:
@gaugetester Could you show us your `flush_cb` implementation? I'm suspicious the problem is there.

embeddedt, 2020-04-19:
It's also strange to me that the gauge needle has no antialiasing in your screenshot. Have you changed any settings relating to that?

gaugetester, 2020-04-19:
My flash cl:

//for qt
void my_disp_flush(lv_disp_drv_t * disp, const lv_area_t * area, lv_color_t * color_p)
{
  UNUSED(color_p);

  MyLCD.x1 = area->x1;
  MyLCD.x2 = area->x2;
  MyLCD.y1 = area->y1;
  MyLCD.y2 = area->y2;
  MyLCD.disp = disp;
  MyLCD.freshdata = true;
  return;

}

void interface_ready(void)
{
	  MyLCD.freshdata = false;
	  lv_disp_flush_ready(MyLCD.disp);
}

And this is for the microcontroller:

void hmi_disp_flush(lv_disp_drv_t * disp, const lv_area_t * area, lv_color_t * color_p)
{
  unsigned short x, y;

  //Ausgabefenster setzen                                                                           
  setAddrWindow(area->x1,area->y1,area->x2,area->y2);
                                                                    
  for(y = area->y1; y <= area->y2; y++)
    {
      for(x = area->x1; x <= area->x2; x++)
        {
          pushColor(ili9341_Color233(   lcdram[x][y].byte )  );
        }
    }

  lv_disp_flush_ready(disp);                          
}

But I dont think there is a problem with flush_cb because it is working first time and I can see that lvgl is changing the value for x1 and y1. If the problem is in my code I expect a problem with pointer that destroy something. But in this case I will need some time to find.

kisvegabor, 2020-04-22:
Where does `lcdram` come from?
I see that you save `disp` in `MyLCD.disp = disp;` but you should save `color_p` instead.

What happens if you create a button like this and try to drag it with your finger?
```c
    lv_obj_t * btn = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_set_drag(btn, true);
    lv_label_create(btn, NULL);
```

closed 2020-05-13 by kisvegabor