lvgl/lvgl #1464: [Dev7.0] cpicker rectangle does not draw over 360 px

xennex22, 2020-04-19:
A cpicker of type `LV_CPICKER_TYPE_RECT` does not draw correctly if it is wider than 360 pixels.

[image]

The problem is that the loop to draw the graduated colors makes each color rectangle `i_step` wide which is unrelated to the actual pixel width of the widget.
```
rect_area.x2 = rect_area.x1 + i_step;
```

We could make the quantization 1.
```
uint16_t i_step = LV_MATH_MAX(1, 360 / grad_w);
```
[image]

But probably the idea is to reduce the rectangles so the loop x2 calculation needs to be changed
```
lv_coord_t xi2 = ((i+i_step) * grad_w) / 360;
rect_area.x2 = LV_MATH_MIN(grad_area.x1 + xi2, grad_area.x1 + grad_w - i_step);
```
[image]

Also I think we need to look at the double click behavior, which I had to disable.

embeddedt, 2020-04-19:
Can you send a pull request with the second fix?

xennex22, 2020-04-19:
Ok.  Still learning with github and I'm trying to find a way to do this efficiently so I don't have to do endless offline diffs with the other changes I have made.

closed 2020-04-20 by xennex22