lvgl/lvgl #1469: SVG support

raveslave, 2020-04-23:
Any plans for supporting native SVG format?
Could be as a source format, parsed runtime or during pre-compilation.

embeddedt, 2020-04-23:
We don't have this on the roadmap at the moment, but we could add support for it as an external library (similar to how PNGs are handled today). I assume you're looking for the ability to store a vector image and then render it on-the-fly at any size, not just the ability to convert an SVG to a bitmap.

In the past I've looked at using [libsvgtiny](https://github.com/dunkelstern/libsvgtiny), but I never got around to actually writing any code with it.

raveslave, 2020-04-23:
got it, so lvgl isn't really vector based?

kisvegabor, 2020-04-23:
@embeddedt 
I was also checking some SVG libraries in the near past but haven't found any which can rasterize as well. `libsvgtiny` seems interesting, the question is what kind of drawing routines it needs to create the image.

@raveslave 
There are some drawing functions in lvgl (actually much more than in other mainstream libraries) but not all SVG primitives can be drawn by lvgl. For example, lvgl doesn't have a path stroke or path fill algorithm. It can draw lines and filled polygons but can draw a Bezier line which a basic thing in SVG.

raveslave, 2020-04-24:
thx for reply. any plans to do path stroke & fills?

kisvegabor, 2020-04-24:
>  any plans to do path stroke & fills?

It's not the roadmap now. It'd be best to find a library that can rasterize SVG as well. Or check `libsvgtiny` in more detail to see how it could work with lvgl.

closed 2020-05-22 by stale[bot]