lvgl/lvgl #1471: Dev-7.0 performance experiments

barbiani, 2020-04-24:
I own 2 dev boards that I have been experimenting with lvgl dev-7.0 branch and rework-7 lv_demo_widgets example. These are some results.

[https://www.mikroe.com/mikromedia-4-pic32mx7](https://www.mikroe.com/mikromedia-4-pic32mx7)

This board has a PIC32MX running at 80Mhz.
Cache enabled, PMP timing low.
SSD1963 display controller driven by 16bit DMA.
2x32 line buffers.
Resolution 480x272 16 bit color.

Picture showing internal processing time x display update of a gauge in the middle of the screen.
[image]

Generally, the experience is good for static screens and small widget updates. Tab scrolling and edge flashes are quite slower close to the 10 fps range. During this scroll event, the red scope trace is always high meaning 100% cpu usage and the screen refresh/fill is noticeable.

I believe that the PIC32MZ@200MHz with controllerless display would over perform this one by far.

Now to the [Wrover kit](https://www.espressif.com/en/products/hardware/esp-wrover-kit/overview).

Esp32 mcu running at 240Mhz.
SPI speed at 40Mhz.
Flash at 80Mhz QIO.
ILI9341 controller.
Resolution 320x240 16 bit color.
2x40 line buffers.
Code from [ESP32 port](https://github.com/littlevgl/lv_port_esp32).

Picture showing the chip select line during a lv_demo_widgets tab transition:
[image]

The display filling feels a bit slower than the PIC32, but this is probably due to the wider bus of the pic.

C47D, 2020-04-24:
Hi @barbiani, we're also working on porting the lv_esp_port to lvgl v7, the project is [here](https://github.com/C47D/lv_port_esp32_v7), but the "discussion" is being done in [here](https://github.com/littlevgl/lv_port_esp32/issues/101), would you like to post your progress in the last link?

barbiani, 2020-04-24:
@C47D 
Does it need porting? I think that it is the same driver I am using... took from the lvgl repositories.

C47D, 2020-04-24:
@barbiani It's ready to go, based on the lv_port_esp32 repository, both lvgl and lv_examples are pointing to the lvgl v7 branches (you can check it on the .gitmodules file). I'm using the ESP32 Wrover Kit v4.1

kisvegabor, 2020-04-24:
@barbiani 
I've just optimized edge flash animation. Is it better now?

barbiani, 2020-04-25:
@C47D 
> @barbiani It's ready to go, based on the lv_port_esp32 repository, both lvgl and lv_examples are pointing to the lvgl v7 branches (you can check it on the .gitmodules file). I'm using the ESP32 Wrover Kit v4.1

Are you going to refresh the main repository? I will be adding an esp32 adc resistive touch driver soon.

@kisvegabor 
> @barbiani
> I've just optimized edge flash animation. Is it better now?

lv_ex_tileview_1 actually does not show the edge flash anymore in the lastest commit. The red focus border does not flex, just like it is when disabled.

Using the default config file from dev-7.0. Lastest rework-7 examples.

kisvegabor, 2020-04-25:
@barbiani 
I've just fixed it.

barbiani, 2020-04-25:
Much faster. It was taking seconds to animate in and out and now it feels very good.

kisvegabor, 2020-04-25:
Great! Thank you for the feedback! :)

barbiani, 2020-04-25:
Insert random touch coordinates (>screen size) in the touch driver and lvgl wil crash at  lv_obj_get_state().

kisvegabor, 2020-04-26:
Should be fixed. 930332ed

barbiani, 2020-05-13:
Lastest commit with the new 16bpp color filler increased the automatic animated tab transition fps from 27 to 33.

kisvegabor, 2020-05-13:
Awesome! 22% improvement :sunglasses:

kisvegabor, 2020-06-04:
The performance of v7 seems good so far, let's close this issue.
Thank you for the feedback!

closed 2020-06-04 by kisvegabor