lvgl/lvgl #1472: Changing button disabled states give exception.

xennex22, 2020-04-24:
There's a problem with the changes made to button disabled states.  https://github.com/littlevgl/lvgl/commit/e31ac20d00fdc3ec403c2fb98cb60057b7729a5f

lv_btn.h
```
enum {
    LV_BTN_STATE_ACTIVE,
    LV_BTN_STATE_RELEASED,
    LV_BTN_STATE_PRESSED,
    LV_BTN_STATE_CHECKED_RELEASED,
    LV_BTN_STATE_CHECKED_PRESSED,
    LV_BTN_STATE_DISABLED = 0x80,
    _LV_BTN_STATE_LAST, /* Number of states*/
};
```
lv_imgbtn.h
```
const void * img_src_mid[_LV_BTN_STATE_LAST];
```
`_LV_BTN_STATE_LAST` is now 0x81, so img_src_mid ends up being a lot bigger than it should.

I'm getting an exception using `lv_obj_set_state` and once I found the size of `_LV_BTN_STATE_LAST` and rolled back the commit, the exception was fixed.  I can't see why using the large array of pointers would except, but I did not trace the exception any further until the array size is fixed.

embeddedt, 2020-04-25:
I *think* this should fix it: https://github.com/littlevgl/lvgl/blob/dev-7.0/src/lv_widgets/lv_btn.h

xennex22, 2020-04-25:
The issue is that `lv_state_t` is a bitmask and `lv_btn_state_t` is an enum.

`lv_btn_set_state` uses a switch to translate `lv_btn_state_t` to `lv_state_t`.  I have not drawn up a table of all the bitmasks to check the logic but it is not wrong as far as I can tell.

`lv_btn_get_state` tries to convert a bitmask into an enum, which I guess you can do if the masks match the enum behavior from `lv_btn_set_state`.  But then it ors the disabled enum state onto the other values, which I do not think it right.

I think the logic needs to be revisited and the mapping of states to images checked.  eg `LV_BTN_STATE_ACTIVE` is only used as a parameter to `lv_btn_set_state` to remove the disabled flag.  So it does not need to be in the list of images sources  used by `lv_imgbtn`.

How many image sources should there be?  5?  released, pressed, checked_released, checked_pressed & disabled?  But there are 8 states (released/pressed, checked/unchecked, active/disabled).

kisvegabor, 2020-04-25:
Hm, that's really not that trivial.
My thoughts were there should be 5 states: pressed, releases, check pressed, checked released, and disabled. But it really doesn't align with the new `lv_btn_state_t`.

However, `lv_btn_state_t` was modified to allow more complex states, like checked + disabled.
As I'm thinking about it again, I found these states might be reasonable:  
1. released
2. pressed
3. checked released
4. checked pressed
5. disabled
6 checked disabled

I think pressed (+ checked) + disabled is rather confusing than useful. What do you think?

xennex22, 2020-04-25:
https://github.com/littlevgl/lvgl/commit/9fe050dcb94d713de44981824054621e9e83e580 fails if you assign an image to the disabled state.  eg.
```
lv_imgbtn_set_src(left_btn, LV_BTN_STATE_DISABLED, image_leftarrow);
```
Also it has an unused image slot for `LV_BTN_STATE_ACTIVE`

At this point I think we've better off making `lv_imgbtn_set_src` use its own enum other than `lv_btn_state_t`.  That way at least the compiler will stop me doing horrible things.

Unless we are using the library to dynamically make UIs then `lv_imgbtn_set_src` is always going to be called with the second parameter as a constant.  So we could do something like:
```
typedef enum {
    LV_IMGBTN_IMAGE_RELEASED = 0,
    LV_IMGBTN_IMAGE_PRESSED,
    LV_IMGBTN_IMAGE_CHECKED_RELEASED,
    LV_IMGBTN_IMAGE_CHECKED_PRESSED,
    LV_IMGBTN_IMAGE_DISABLED,
    _LV_IMGBTN_IMAGE_COUNT, /* Number of images*/
} lv_imgbtn_image_t;
```
```
const void * img_src_mid[_LV_IMGBTN_IMAGE_COUNT];   /*Store center images to each state*/
```
```
void lv_imgbtn_set_src(lv_obj_t * imgbtn, lv_imgbtn_image_t image, const void * src);
```
```
lv_imgbtn_set_src(left_btn, LV_BTN_STATE_DISABLED, image_leftarrow);
```
But then we've need to change the draw functions to translate from the state to the image index.

xennex22, 2020-04-25:
Posted at the same time.
checked + pressed + disabled does not make any sense.  So I think 6 possible button image states.

Another option is to make a static function for `lv_imgbtn` that translates the states to the image index.  This still would require a lot of changes in 'lv_imgbtn.c'
```
static lv_imgbtn_image_t lv_imgbtn_get_img_index(lv_btn_state_t state)
{
    if(state & LV_BTN_STATE_DISABLED) {
        state &= ~LV_BTN_STATE_DISABLED;
        if((state == LV_BTN_STATE_CHECKED_RELEASED) || (state == LV_BTN_STATE_CHECKED_RELEASED))
            return LV_IMGBTN_IMAGE_CHECKED_DISABLED;
        return LV_IMGBTN_IMAGE_DISABLED;
    }
    if(state == LV_BTN_STATE_PRESSED)
        return LV_IMGBTN_IMAGE_PRESSED;
    if(state == LV_BTN_STATE_CHECKED_RELEASED)
        return LV_IMGBTN_IMAGE_CHECKED_RELEASED;
    if(state == LV_BTN_STATE_CHECKED_PRESSED)
        return LV_IMGBTN_IMAGE_CHECKED_PRESSED;
    return LV_IMGBTN_IMAGE_RELEASED;
}
```
This works but at that point we may as well change all the `lv_imgbtn_set_src` functions to use a new enum `lv_imgbtn_image_t` as outlined above.

My only reservation is that because `lv_btn_state_t` is not an enum then some compilers will happily assign an `uint8_t` to `lv_btn_state_t` without complaining.  So I'd put a range test in `lv_imgbtn_set_src` to make sure the right enum was being used.
```
LV_DEBUG_ASSERT(image< _LV_IMGBTN_IMAGE_COUNT, "Btn image out of range", image);
```

embeddedt, 2020-04-25:
If we're going to change the enum that gets passed to `lv_imgbtn_set_src` in such a way that passing `LV_BTN_STATE_XXX` no longer works, I suggest that we rename the API slightly (e.g. to `lv_imgbtn_set_img_src`) so existing code doesn't silently break.

kisvegabor, 2020-04-28:
If you are both like the [6 states](https://github.com/littlevgl/lvgl/issues/1472#issuecomment-619395141) for buttons and image buttons, I'm also happy with that.
The enum names in `lv_btn_state_t` already changed compared to v6, so we are free to change as it's required.

xennex22, 2020-04-28:
What is the planned solution?  
1. Change all `lv_imgbtn_set_src` and `lv_imgbtn_get_src` functions to use a new 6 state enum `lv_imgbtn_image_t`
2. As #1 but also change the names of the `lv_imgbtn_set_src` and `lv_imgbtn_get_src` functions to `lv_imgbtn_set_img_src` and `lv_imgbtn_get_img_src`
3. `lv_imgbtn_set_src` and `lv_imgbtn_get_src` keep using `lv_btn_state_t` and then use the 6 state enum `lv_imgbtn_image_t` internally to `imgbtn`
4. Change `lv_btn_state_t` to 6 states and don't or the disabled flag

All of #1, #2, #3 need the state to image translation function outlined above.  With my code I implemented #3 because existing code did not need to be altered.  `lv_btn_state_t` should be cleaned up too, moving `LV_BTN_STATE_ACTIVE` to near `LV_BTN_STATE_DISABLED` and removing `LV_BTN_STATE_DISABLED`

kisvegabor, 2020-04-28:
@xennex22 I vote for No3.
Just to be sure we are talking about the same thing I've added it here: https://github.com/littlevgl/lvgl/commit/4401ad874d3816e578122e36da587214bd336a6a

As I see imgbtn can stay as it. Or have I missed something?

xennex22, 2020-04-29:
That's No 4, `lv_btn_state_t` has been changed to 6 states and the disabled flag is not a bitmask.

https://github.com/littlevgl/lvgl/commit/4401ad874d3816e578122e36da587214bd336a6a works for me.

kisvegabor, 2020-04-29:
Okay, great!
I close this issue then. Thank you for your contribution!

commit ac1680c5 (2020-04-25): Fix #1472: Ignore DISABLED state when counting button states
closed 2020-04-29 by kisvegabor