lvgl/lvgl #1474: first drawn buffer has uninitialized widgets

curly-brace, 2020-04-25:
hello all,
i'm sure that it is trivial issue, but i'm completely frustrated about this. rechecked everything (as i think) but issue is still with me.
i have a buffer with 32px height. and i create fullscreen page. on display i see that first 32px block shows page with default width, but after that page is drawn fullscreen.
for me it seems like first draw to buffer always has unitialized widgets. and this persists while screen refreshes.
what do i miss here? may be some call of a function?
i'm using esp32 and lv_tick is called out of timer, but i'm almost sure that tick does not causes this.
lvTaskHandler is freertos task with 1ms interval, but is believe it should not cause this too.

embeddedt, 2020-04-25:
I see you are using FreeRTOS. Are you using a mutex as described in the [documentation](https://docs.littlevgl.com/en/html/porting/os.html#tasks-and-threads)? If not, it's possible that `lv_task_handler` is running in between the creation of the page and when you set its width.

Please show us the code you are using to create the page and synchronize the execution of LittlevGL functions.

curly-brace, 2020-04-25:
added mutex, but with no success.
here is the code (without header, but whatever)
https://pastebin.com/rbgbAi98

curly-brace, 2020-04-25:
and screen looks like this. may be it will explain something
[image]

embeddedt, 2020-04-25:
You need the mutex around **any** calls to LittlevGL functions (it should also wrap lines 26-30).

curly-brace, 2020-04-25:
right! that fixed the issue! thank you!

closed 2020-04-25 by curly-brace