lvgl/lvgl #1477: hope the main lib can add a T9 keyboard.

simonchen007, 2020-04-27:
continue issue #534.

I have made a T9 keyboard, and tested.
but I'm new to littlevgl. so, I do not understand how to add it into the main lib.
hope it cloud be added.
here is the code:

```c
/**
 * @file demo_minikb.c
 * simon@yeacreate.com
 * made date: 20200426
 * T9 keyboard for littlevgl
 */

/*********************
 *      INCLUDES
 *********************/
#include "demo_minikb.h"

/*********************
 *      DEFINES
 *********************/

/**********************
 *      TYPEDEFS
 **********************/

// /**********************
//  *  STATIC PROTOTYPES
//  **********************/
void text_area_event_handler(lv_obj_t *text, lv_event_t event);
void create_kb_h();
void event_handler2(lv_obj_t *obj, lv_event_t event);
void create_mini_kb(int8_t which);
void even_minikb(lv_obj_t *obj, lv_event_t event);

// /**********************
//  *  STATIC VARIABLES./
//  **********************/
static lv_obj_t *kb;
static lv_obj_t *kb_min;
static lv_obj_t *body;
bool kb_min_enable;
static lv_obj_t *text_input;

/**********************
 *      MACROS
 **********************/

/**********************
 *   GLOBAL FUNCTIONS
 **********************/

/**
 * Create a demo application
 */

void text_area_event_handler(lv_obj_t *text, lv_event_t event)
{
    if (event == LV_EVENT_CLICKED)
    {
        if (kb == NULL)
        {
            create_kb_h(text);
        }
    }
}

static const char *main_map[] = {"   1\nABC", "   2\nDEF", "   3\nGHI", LV_SYMBOL_BACKSPACE, "\n",
[... 275 lines cut ...]
    lv_obj_set_event_cb(kb, event_handler2);

    lv_obj_set_size(kb, lv_obj_get_width(body) - 2, lv_obj_get_height(body) / 2);
    lv_obj_align(kb, NULL, LV_ALIGN_IN_BOTTOM_MID, 0, 0);
}

void demo_create(void)
{
    lv_coord_t hres = lv_disp_get_hor_res(NULL);
    lv_coord_t vres = lv_disp_get_ver_res(NULL);

    lv_obj_t *scr = lv_obj_create(NULL, NULL);
    lv_disp_load_scr(scr);

    body = lv_cont_create(lv_disp_get_scr_act(NULL), NULL);
    lv_obj_set_pos(body, 0, 0);
    lv_obj_set_size(body, lv_disp_get_hor_res(NULL) - 2, lv_disp_get_ver_res(NULL) - 2);

    text_input = lv_ta_create(body, NULL);
    lv_obj_set_size(text_input, lv_obj_get_width(body), lv_obj_get_height(body) / 2);
    lv_ta_set_cursor_type(text_input, LV_CURSOR_BLOCK);
    lv_ta_set_text(text_input, "hello!");
    lv_obj_set_event_cb(text_input, text_area_event_handler);
}
```

kisvegabor, 2020-04-27:
Hi @simonchen007 ,

We have some option for how to make it public:
1. We can create turn it to [example in the docs](https://docs.littlevgl.com/en/html/object-types/btnm.html#example). It's much simpler as there are fewer rules to follow to create an example.
2. Add this feature to the [keyboard object](https://github.com/littlevgl/lvgl/blob/dev-7.0/src/lv_widgets/lv_keyboard.c). It's a more complicated process because we need to discuss the details of the implementation as we need to "live" with this feature and maintain it in the future.

Which one would you prefer?

simonchen007, 2020-04-28:
hope it can add into the keyboard object(selection 2), then , everyone can use it.

kisvegabor, 2020-04-28:
Okay! v7 will go to release candidate state on Friday, so we have no time to add it there. 
Anyway, have you already tried v7 from the `dev-7.0` branch?

simonchen007, 2020-04-29:
just take your time.
I haven't tried v7 yet. some new features ?

embeddedt, 2020-04-29:
There are quite a few new things in v7. They're summarized [here](https://github.com/littlevgl/lvgl/issues/1243#issue-513344087).

embeddedt, 2020-05-20:
@simonchen007 Have you gotten a chance to try v7 yet? What do you think about putting together a pull request that adds this feature? You can make it against `master` and we will merge it to the development branch when it's ready (there are [a few](https://github.com/lvgl/lvgl/pulls) PRs waiting).

simonchen007, 2020-05-20:
@embeddedt  , I haven't test it in v7, I'm using lv_arduino land lv_platformio , it is just v6.x

simonchen007, 2020-06-11:
@C47D  , this demo is tested in v6. it works.

simonchen007, 2020-06-11:
@embeddedt  , 
We tested the v7 for 6 days with different boost,eg: dual buffers,double LV_MEM_SIZE,different LV_TICK_PERIOD_MS and etc , but still can not reach the speed with v6 the same benchmark(same codes).
We doubted the driver lib(TFT_eSPI) before, but this lib can run up to 80Mhz stable for my TFT screen.

kisvegabor, 2020-06-11:
To find what causes regression please test these with v6 and v7. Be sure the settings are identical (driver, display orientation lvgl settings, display buffer size, UI element looks the same).
- empty screen
- empty screen + this text "01234556789\n01234556789\n01234556789\n01234556789\n"
- empty screen + 2 not overlapping rectangles with 100x100 size (no gradient, 2 px border, 10 ox radius)

Please also send images from all UI (6 in total).

Regarding the configuration of LVGL and the driver:
- 1 or 2 display buffers (if 2 do you use DMA to flush)
- size of display buffer(s)

simonchen007, 2020-06-12:
@kisvegabor  , today , I try the lv_arduino(v7) for testing. it runs up to 30 FPS in esp32. 
the problem seems between esp-idf and esp-arduino.

kisvegabor, 2020-06-14:
@simonchen007 
Huh, it's a big relief for us :slightly_smiling_face:

simonchen007, 2020-06-14:
@kisvegabor  , but another bad news is lv_port_esp32 also does not have a high speed. Only lv_arduino can have nice speed currently.
I open another topic here: https://github.com/lvgl/lv_arduino/issues/52

C47D, 2020-06-14:
For lv_port_esp32 maybe this comment can help you improve the performance: https://github.com/lvgl/lv_port_esp32/issues/101#issuecomment-633743516

kisvegabor, 2020-07-08:
I close this issue because there was no activity here for a while.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2020-07-08 by kisvegabor

bhupiister, 2021-11-30:
@simonchen007 Did you make any further development to the T9 keyboard? I was trying your above-mentioned code but a lot of it is outdated. Please let me know if you have any repo with the latest code. I need to try it on a small screen.

mistrjirka, 2023-09-08:
The T9 would be extremely helpful on smaller devices. Why was it abounded? Was it too slow?

BrazenCodebane, 2024-05-07:
this project would be extremely helpful for a 2.4 inch cyd... any updates on it?

kisvegabor, 2024-05-08:
If anyone have time and interest to work on the implementation I would be happy to help figuring out how to approach it.