lvgl/lvgl #1483: [v7.0] Release candidate discussion

kisvegabor, 2020-04-30:
Hi,

On 1 May v7 will enter to release candidate state.  After that, no new features are merged to v7.
(The release is planned for 18 May.)

During the release candidate state and the goal is to
- get feedback from the new version
- fix bugs 
- finalize documentation
- update the projects

I've added a [CHANGELOG](https://github.com/littlevgl/lvgl/blob/dev-7.0/CHANGELOG.md) to summarize the new features and changes.

It also includes some "legal" changes:
- the name of the library will be changed to LVGL
- we are developing a new website
- we are creating a company the have an official background for LVGL
- update the redirects from the old domains to the new ones

So in order to make the new version better, I encourage you to make some experiments with it. The easiest way to try it is using [dev-7.0 branch of the Eclips based simulator](https://github.com/littlevgl/lv_sim_eclipse_sdl/tree/dev-7.0).

Feel free to share your thoughts about the above-described things.

If you have time and interest to update one or more projects, please write a comment here.
Here is the list of the projects which needs to be updated:
- [ ] lv_port_esp32
- [ ] lv_port_linux_frame_buffer  (@kisvegabor)
- [ ] lv_port_stm32f769_disco (also update to CubeIDE) (@kisvegabor)
- [ ] lv_port_stm32f429_disco_sw4stm32 (also update to CubeIDE) (@kisvegabor)
- [ ] lv_port_stm32f746_disco_sw4stm32 (also update to CubeIDE)
- [ ] lv_port_stm32f469_disco_sw4stm32 (also update to CubeIDE)
- [ ] lv_port_maixduino
- [ ] lv_arduino
- [ ] lv_sim_eclipse_sdl  (@kisvegabor)
- [ ] lv_sim_visual_studio_sdl
- [ ] lv_sim_codeblocks_win
- [ ] lv_sim_emscripten
- [ ] lv_platformio
- [ ] lv_binding_micropython
- [ ] lv_micropython

embeddedt, 2020-04-30:
@kisvegabor I can update the usual projects (STM32F746, CodeBlocks, Emscripten).

embeddedt, 2020-04-30:
How are we planning to handle support for legacy versions of LittlevGL going forward? It would be good to have an official policy to refer to, especially as the user base grows. We are still getting occasional questions about 5.x on the forum (the most recent one being from the end of March). I must admit that I can hardly remember the old action APIs anymore (events are so much cleaner and easier to use).

C47D, 2020-04-30:
@kisvegabor I guess it's time to merge the changes done on the lv_port_esp32 project to make LVGL v7 work, I can do that tomorrow.

persello, 2020-04-30:
Hello, I don't know if I'm doing something wrong again, I had some difficulties in finding documentation on the mono theme. With some changes (B/W primary/secondary colors selected), the material theme works on my monochrome OLED, while the mono theme doesn't (the screen is completely white or black, depending on the fact that the display has reversed colors or not). Here are the results with their theme configurations.

[image]

```c
/*Always enable at least on theme*/
#define LV_USE_THEME_EMPTY       0   /*No theme, you can apply your styles as you need*/
#define LV_USE_THEME_TEMPLATE    0   /*Simple to the create your theme based on it*/
#define LV_USE_THEME_MATERIAL    1   /*A fast and impressive theme*/
#define LV_USE_THEME_MONO        1   /*Mono-color theme for monochrome displays*/

#define LV_THEME_DEFAULT_INCLUDE            <stdint.h>      /*Include a header for the init. function*/
#define LV_THEME_DEFAULT_INIT               lv_theme_mono_init
#define LV_THEME_DEFAULT_COLOR_PRIMARY      LV_COLOR_BLACK
#define LV_THEME_DEFAULT_COLOR_SECONDARY    LV_COLOR_WHITE
#define LV_THEME_DEFAULT_FLAG               LV_THEME_MATERIAL_FLAG_LIGHT // Not sure about the flag, mono theme seems to have no flags.
#define LV_THEME_DEFAULT_FONT_SMALL         &lv_font_montserrat_16
#define LV_THEME_DEFAULT_FONT_NORMAL        &lv_font_montserrat_16
#define LV_THEME_DEFAULT_FONT_SUBTITLE      &lv_font_montserrat_16
#define LV_THEME_DEFAULT_FONT_TITLE         &lv_font_montserrat_16
```
---
[image]

```c
/*Always enable at least on theme*/
#define LV_USE_THEME_EMPTY       0   /*No theme, you can apply your styles as you need*/
#define LV_USE_THEME_TEMPLATE    0   /*Simple to the create your theme based on it*/
#define LV_USE_THEME_MATERIAL    1   /*A fast and impressive theme*/
#define LV_USE_THEME_MONO        1   /*Mono-color theme for monochrome displays*/

#define LV_THEME_DEFAULT_INCLUDE            <stdint.h>      /*Include a header for the init. function*/
#define LV_THEME_DEFAULT_INIT               lv_theme_material_init
#define LV_THEME_DEFAULT_COLOR_PRIMARY      LV_COLOR_BLACK
#define LV_THEME_DEFAULT_COLOR_SECONDARY    LV_COLOR_WHITE
#define LV_THEME_DEFAULT_FLAG               LV_THEME_MATERIAL_FLAG_LIGHT
#define LV_THEME_DEFAULT_FONT_SMALL         &lv_font_montserrat_16
#define LV_THEME_DEFAULT_FONT_NORMAL        &lv_font_montserrat_16
#define LV_THEME_DEFAULT_FONT_SUBTITLE      &lv_font_montserrat_16
#define LV_THEME_DEFAULT_FONT_TITLE         &lv_font_montserrat_16
```

I'm not sure about the flag setting when the mono theme is selected. Leaving the default colors as primary and secondary breaks the material theme (I think it's quite logic, since mono, and not material, is intended for this use) but the mono theme behaves the same way. The only thing changed between the two tries is shown above, no other changes have been made.

The board is an ESP8266, programmed with Arduino. Display ST1306 over I2C.

embeddedt, 2020-04-30:
I've created a module which may be useful for trying to convert 6.x styles to v7 styles.

https://github.com/embeddedt/lv_style_compat

excitedbox, 2020-04-30:
Congrats guys. Great work. I am so excited to see it get released.

kisvegabor, 2020-05-01:
@embeddedt 
> How are we planning to handle support for legacy versions of LittlevGL going forward? It would be good to have an official policy to refer to, especially as the user base grows. We are still getting [...]

I'm also don't really remember. What we can do for sure is to leave the old versions in `release/vx` branches to make them easier to access. As probably we will have more frequent major releases from now I can't tell to support the last X major version, as I don't know now how often we will really have major releases. Maybe we can say that older than half year versions are not officially supported. What do you think?

@C47D 
> @kisvegabor I guess it's time to merge the changes done on the lv_port_esp32 project to make LVGL v7 work, I can do that tomorrow, one question tho: Will v7 be a new branch on the lvgl repo? Same for the lv_examples repo, what will be the name of the branch for v7?

I've added a line about it in the top comment:
> Please create a `dev-7.0` branch in the project which can be easily merged when v7 is released.

@persello 
I've added more comments about the theme settings in [lv_conf_templ.h](https://github.com/littlevgl/lvgl/blob/aad8667a9f9dce82f719aec24f64dbe42f729ad0/lv_conf_template.h#L392).  It works well for in the simulator.
Independently from the flag settings you still should see something but it should be 0 in case of the mono theme.
Could you send your whole `lv_conf.h`?

embeddedt, 2020-05-01:
> Maybe we can say that older than half year versions are not officially supported. What do you think?

Sounds good to me.

persello, 2020-05-01:
@kisvegabor [Here](https://github.com/open-agri/TractorKit-DevSim/blob/a2c25a82f9cc4b4bce2c0283322241fceffa172b/src/lv_conf.h) is my `lv_conf.h`. The diff from the template is the following:
```
10c10
< #if 1 /*Set it to "1" to enable content*/
---
> #if 0 /*Set it to "1" to enable content*/
23,24c23,24
< #define LV_HOR_RES_MAX          (128)
< #define LV_VER_RES_MAX          (64)
---
> #define LV_HOR_RES_MAX          (480)
> #define LV_VER_RES_MAX          (320)
32c32
< #define LV_COLOR_DEPTH     1
---
> #define LV_COLOR_DEPTH     16
50c50
< #define LV_ANTIALIAS        0
---
> #define LV_ANTIALIAS        1
59c59
< #define LV_DPI              149     /*[px]*/
---
> #define LV_DPI              130     /*[px]*/
86c86
< #  define LV_MEM_SIZE    (16U * 1024U)
---
> #  define LV_MEM_SIZE    (32U * 1024U)
179c179
< #define LV_USE_GPU              0
---
> #define LV_USE_GPU              1
269c269
< #define LV_USE_LOG      1
---
> #define LV_USE_LOG      0
278c278
< #  define LV_LOG_LEVEL    LV_LOG_LEVEL_TRACE
---
> #  define LV_LOG_LEVEL    LV_LOG_LEVEL_WARN
336c336
< #define LV_FONT_MONTSERRAT_12    1
---
> #define LV_FONT_MONTSERRAT_12    0
395c395
< #define LV_USE_THEME_MONO        1   /*Mono-color theme for monochrome displays*/
---
> #define LV_USE_THEME_MONO        0   /*Mono-color theme for monochrome displays*/
398,401c398,401
< #define LV_THEME_DEFAULT_INIT               lv_theme_mono_init
< #define LV_THEME_DEFAULT_COLOR_PRIMARY      LV_COLOR_WHITE
< #define LV_THEME_DEFAULT_COLOR_SECONDARY    LV_COLOR_BLACK
< #define LV_THEME_DEFAULT_FLAG               0
---
> #define LV_THEME_DEFAULT_INIT               lv_theme_material_init
> #define LV_THEME_DEFAULT_COLOR_PRIMARY      LV_COLOR_RED
> #define LV_THEME_DEFAULT_COLOR_SECONDARY    LV_COLOR_BLUE
> #define LV_THEME_DEFAULT_FLAG               LV_THEME_MATERIAL_FLAG_LIGHT
```

kisvegabor, 2020-05-01:
@persello 
I've found the issue and fixed it. Does it work now?

ValentiWorkLearning, 2020-05-01:
@kisvegabor I have my pet-project that uses LVGL and C++ with Nordic SDK. 

https://github.com/ValentiWorkLearning/GradWork

Do you need additional information/help/porting guide?

fvanroie, 2020-05-01:
I know it's the cut-off date today for v7 features, but would it be possible to add a lv_led_get_state() function to get the current state of a led object? Like lv_switch_get_state() function does for switches.

persello, 2020-05-01:
@kisvegabor Perfect, thanks!

embeddedt, 2020-05-02:
@fvanroie Use [`lv_led_get_bright`](https://github.com/littlevgl/lvgl/blob/1e59100ff2931651987594abff5c551fddfb14ab/src/lv_widgets/lv_led.c#L166-L177).

fvanroie, 2020-05-03:
Thanks @embeddedt. I forgot about that function.

ValentiWorkLearning, 2020-05-04:
Observed abnormal memory consumption increasing during library usage. Investigating. Increased difference from 4-5 kilobytes from Lv_memory pool  to 13 kilobytes without any changes of widgets structure. It's not a memory leak from C++ side.
https://github.com/ValentiWorkLearning/GradWork/tree/dev/lvgl_migration?files=1
Here I created branch for processing migration from LVGL 6 to LVGL7.
For building for Visual Studio  2019 just clone repository, execute git submodule update --init and open in Visual Studio folder Simulator. 
Active branch for testing the previous memory consumption is master or dev/develop. Both of them are ready for building.
I'll provide additional information in the morning.

kisvegabor, 2020-05-04:
@ValentiWorkLearning Could you try with [lv_theme_empty_init](https://github.com/littlevgl/lvgl/blob/adbc158a5c508935e4cc93a20b383fce9504dc99/lv_conf_template.h#L416)?

ValentiWorkLearning, 2020-05-05:
@kisvegabor tried. Memory consumption reduced to 10 kilobytes, but still more than in version 6.  Version 6 - about 5376 bytes, version 7- 10 kilobytes.

kisvegabor, 2020-05-05:
@ValentiWorkLearning 
If you are on Linux can you run `nm -S --size-sort YOUR_EXECUTABLE_HERE -t d | grep -P "^[0-9]+ [0-9]+ [bBdD]" | tail -30` to see what uses that much RAM? It looks pretty normal here.

ValentiWorkLearning, 2020-05-05:
@kisvegabor hmm. NM Linux can profile usage of `/* Size of the memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#  define LV_MEM_SIZE    (128U * 1024U)` ?

I mean the usage  of internal library memory pool.
[image]

Now usage about 13 kilobytes from allocated pool with size 128U * 1024U
On the previous library version the same view uses around 6 kilobytes:
[image]

kisvegabor, 2020-05-05:
Ah sorry, it's not for malloc-ed memory. 
Could you compare the memory usage:
1. without creating any objects
2. with 10 labels.

Can you also attach your `lv_conf.h`?

I've noticed extra pixels in the upper corner of the circles. Do you know where do they come from? If you send a code snippet to reproduce it, I take a look.
[image]

ValentiWorkLearning, 2020-05-05:
Thanks! I'll provide today more information that you've requested.

ValentiWorkLearning, 2020-05-05:
1. Without creating any objects:
`Used: 1544 , 2% fragmentation: 1, biggest free: 129520
Used: 1544 , 2% fragmentation: 1, biggest free: 129520
Used: 1544 , 2% fragmentation: 1, biggest free: 129520
Used: 1544 , 2% fragmentation: 1, biggest free: 129520
Used: 1544 , 2% fragmentation: 1, biggest free: 129520
Used: 1544 , 2% fragmentation: 1, biggest free: 129520
Used: 1544 , 2% fragmentation: 1, biggest free: 129520
Used: 1544 , 2% fragmentation: 1, biggest free: 129520
`

2. With 10 labels that use lv_font_montserrat_16
`
Used: 3616 , 3% fragmentation: 0, biggest free: 127456
Used: 3616 , 3% fragmentation: 0, biggest free: 127456
Used: 3616 , 3% fragmentation: 0, biggest free: 127456
Used: 3616 , 3% fragmentation: 0, biggest free: 127456
Used: 3616 , 3% fragmentation: 0, biggest free: 127456
Used: 3616 , 3% fragmentation: 0, biggest free: 127456
Used: 3616 , 3% fragmentation: 0, biggest free: 127456
`
3. With 1 label that uses just one glyph from LcdNova68px font, LVGL version 7
`
Used: 7344 , 6% fragmentation: 5, biggest free: 117864
Used: 7344 , 6% fragmentation: 5, biggest free: 117864
Used: 7344 , 6% fragmentation: 5, biggest free: 117864
Used: 7344 , 6% fragmentation: 5, biggest free: 117864
Used: 7344 , 6% fragmentation: 5, biggest free: 117864
Used: 7344 , 6% fragmentation: 5, biggest free: 117864
Used: 7344 , 6% fragmentation: 5, biggest free: 117864
`

4.With 1 label that uses just one glyph from LcdNova68px font, LVGL version 6:
`
Used: 3104 , 10% fragmentation: 0, biggest free: 29664
Used: 3104 , 10% fragmentation: 0, biggest free: 29664
Used: 3104 , 10% fragmentation: 0, biggest free: 29664
Used: 3104 , 10% fragmentation: 0, biggest free: 29664
Used: 3104 , 10% fragmentation: 0, biggest free: 29664
Used: 3104 , 10% fragmentation: 0, biggest free: 29664
Used: 3104 , 10% fragmentation: 0, biggest free: 29664
Used: 3104 , 10% fragmentation: 0, biggest free: 29664
Used: 3104 , 10% fragmentation: 0, biggest free: 29664
`

It seems like something strange has happed inside decoding compressed glyphs.

For both builds with library version 6 and 7 the same source and font files were used.

Attached font and generated file from LVGL font converter. Font converter args are:
-Enable font compression (reduces size, requires 6.1+) - checked
-Horizontal subpixel hinting (may improve font quality, requires 6.1+) - checked
-Bpp-2
-Font size - 68px.

[fonts.zip](https://github.com/littlevgl/lvgl/files/4584051/fonts.zip)

ValentiWorkLearning, 2020-05-05:
@kisvegabor attached lv_conf.h for version 7 and version 6 for Win Desktop simulator. Attached them as .txt files for satisfying github requierments of an attachement format.
[lv_confv7.txt](https://github.com/littlevgl/lvgl/files/4584075/lv_confv7.txt)
[lv_confv6.txt](https://github.com/littlevgl/lvgl/files/4584076/lv_confv6.txt)

MCF, 2020-05-06:
@kisvegabor I should be able to handle the update for lv_sim_visual_studio_sdl.  Not going to promise it will be out on time, but if not that it will be close after that.

kisvegabor, 2020-05-06:
@ValentiWorkLearning 
Ah, the decompression was really changed. Now it always decompresses to 8 bpp for performance reasons and the buffer used for decompression was always reused and not freed.

I fixed it in 12919453 by freeing the buffer after every display refresh cycle.

kisvegabor, 2020-05-06:
@mcf Okay, thank you! :slightly_smiling_face:

ValentiWorkLearning, 2020-05-06:
@kisvegabor thanks for your fix, memory consumption decreased. Now I'm investigating the memory consumption while theme changing mechanism (because I've implemented my own theme changing for several styles). I'll provide more information.

Also, I don't forget about your question about weird pixels that take place in the upper corner of the circles. I'll try to investigate and provide you minimal example for investigating and reproducing it.

ValentiWorkLearning, 2020-05-06:
@kisvegabor Also, as I mentioned early, I've ported LVGL for Nordic NRF52832 MCU.
Do you need an examle/article about this? NRF52832 doesn't have enough hardware resources for producing stable 30FPS output, but a simple GUI for low-power device is possible.

kisvegabor, 2020-05-06:
@ValentiWorkLearning 
> Also, as I mentioned early, I've ported LVGL for Nordic NRF52832 MCU.
Do you need an examle/article about this? NRF52832 doesn't have enough hardware resources for producing stable 30FPS output, but a simple GUI for low-power device is possible.

That would be awesome! 
In the docs of v7 I added [sections for some manufacturers](https://docs.littlevgl.com/v7/en/html/get-started/index.html) and use cases. We can have a section here for Nordic too. I should contain the basics to get started with LVGL and Nordic. Would you be interested in adding this section?

maxpromer, 2020-05-06:
Hello, I interest to update [lv_port_maixduino](https://github.com/littlevgl/lv_port_maixduino). I have Sipeed M1 dock suit and a leader of [IOXGD Platform](https://github.com/ioxgd)

ValentiWorkLearning, 2020-05-06:
@kisvegabor Created the minimal example for reproducing extra pixels  in the upper corner of the circles:
`
    static lv_style_t m_checkedPointStyle;
    static lv_style_t m_uncheckedPointStyle;

    static const lv_color_t MainDarkColor = lv_color_make(0x00, 0x00, 0x00);
    static const lv_color_t MainLightColor = lv_color_make(0xCC, 0xCC, 0xCC);

    lv_style_set_border_color(&m_checkedPointStyle, LV_STATE_CHECKED, MainLightColor);
    lv_style_set_bg_color(&m_checkedPointStyle, LV_STATE_CHECKED, MainDarkColor);
    lv_style_set_line_color(&m_checkedPointStyle, LV_STATE_CHECKED, MainLightColor);
    lv_style_set_line_width(&m_checkedPointStyle, LV_STATE_CHECKED, 2);
    lv_style_set_radius(&m_checkedPointStyle, LV_STATE_CHECKED, LV_RADIUS_CIRCLE);

    lv_style_set_border_color(&m_uncheckedPointStyle, LV_STATE_DEFAULT, MainLightColor);
    lv_style_set_bg_color(&m_uncheckedPointStyle, LV_STATE_DEFAULT, MainLightColor);
    lv_style_set_line_color(&m_uncheckedPointStyle, LV_STATE_DEFAULT, MainLightColor);
    lv_style_set_line_width(&m_uncheckedPointStyle, LV_STATE_DEFAULT, 10);
    lv_style_set_radius(&m_uncheckedPointStyle, LV_STATE_DEFAULT, LV_RADIUS_CIRCLE);

    lv_obj_t* _parentObject = lv_disp_get_scr_act( NULL );
    lv_obj_t* pArcFirst = lv_arc_create(_parentObject, NULL);

    static uint8_t ArcSize = 16;
    static uint8_t _displayWidth = 240;
    static uint8_t _displayHeight = 240;

	lv_obj_add_style(pArcFirst, LV_ARC_PART_BG, &m_checkedPointStyle);
	lv_obj_add_style(pArcFirst, LV_ARC_PART_BG, &m_uncheckedPointStyle);

	lv_arc_set_angles(pArcFirst, 0, 360 );
	lv_obj_set_size(pArcFirst, ArcSize, ArcSize );
	lv_obj_align(
            pArcFirst
		,	nullptr
		,	LV_ALIGN_IN_BOTTOM_MID
		,	-static_cast<int>( _displayWidth / 12 )
		,	-static_cast<int>( _displayHeight / 10 )
	);

    lv_obj_t* pArcSecond  = lv_arc_create( _parentObject, NULL);
	lv_obj_add_style( pArcSecond, LV_ARC_PART_BG, &m_checkedPointStyle );
	lv_obj_add_style( pArcSecond, LV_ARC_PART_BG, &m_uncheckedPointStyle );

	lv_arc_set_angles( pArcSecond, 0, 360 );
	lv_obj_set_size( pArcSecond, ArcSize, ArcSize );
	lv_obj_align(
			pArcSecond
		,	nullptr
		,	LV_ALIGN_IN_BOTTOM_MID
		,	0
		,	-static_cast<int>(_displayHeight / 10)
	);

    lv_obj_t* pArcThird = lv_arc_create( _parentObject, NULL);

	lv_arc_set_angles( pArcThird, 0, 360);
	lv_obj_set_size( pArcThird, ArcSize, ArcSize);
	lv_obj_add_style( pArcThird, LV_ARC_PART_BG, &m_checkedPointStyle);
	lv_obj_add_style( pArcThird, LV_ARC_PART_BG, &m_uncheckedPointStyle );
	lv_obj_align(
			pArcThird
		,	nullptr
		,	LV_ALIGN_IN_BOTTOM_MID
		,	static_cast<int>( _displayWidth / 12)
		,	-static_cast<int>( _displayHeight / 10)
	);
`

embeddedt, 2020-05-06:
@ValentiWorkLearning I've tried your code on my board, but I don't seem to see the extra pixels. I tweaked it slightly so that it would compile as C instead of C++.

One tip: you shouldn't need to create two separate style objects anymore. Style properties for different state combinations are stored separately, so a single `m_pointStyle` should be enough.

```c
static lv_style_t m_checkedPointStyle;
static lv_style_t m_uncheckedPointStyle;

static const lv_color_t MainDarkColor = LV_COLOR_MAKE(0x00, 0x00, 0x00);
static const lv_color_t MainLightColor = LV_COLOR_MAKE(0xCC, 0xCC, 0xCC);

lv_style_set_border_color(&m_checkedPointStyle, LV_STATE_CHECKED, MainLightColor);
lv_style_set_bg_color(&m_checkedPointStyle, LV_STATE_CHECKED, MainDarkColor);
lv_style_set_line_color(&m_checkedPointStyle, LV_STATE_CHECKED, MainLightColor);
lv_style_set_line_width(&m_checkedPointStyle, LV_STATE_CHECKED, 2);
lv_style_set_radius(&m_checkedPointStyle, LV_STATE_CHECKED, LV_RADIUS_CIRCLE);

lv_style_set_border_color(&m_uncheckedPointStyle, LV_STATE_DEFAULT, MainLightColor);
lv_style_set_bg_color(&m_uncheckedPointStyle, LV_STATE_DEFAULT, MainLightColor);
lv_style_set_line_color(&m_uncheckedPointStyle, LV_STATE_DEFAULT, MainLightColor);
lv_style_set_line_width(&m_uncheckedPointStyle, LV_STATE_DEFAULT, 10);
lv_style_set_radius(&m_uncheckedPointStyle, LV_STATE_DEFAULT, LV_RADIUS_CIRCLE);

lv_obj_t* _parentObject = lv_disp_get_scr_act( NULL );
lv_obj_t* pArcFirst = lv_arc_create(_parentObject, NULL);

static uint8_t ArcSize = 16;
static uint8_t _displayWidth = 480;
static uint8_t _displayHeight = 480;

lv_obj_add_style(pArcFirst, LV_ARC_PART_BG, &m_checkedPointStyle);
lv_obj_add_style(pArcFirst, LV_ARC_PART_BG, &m_uncheckedPointStyle);

lv_arc_set_angles(pArcFirst, 0, 360 );
lv_obj_set_size(pArcFirst, ArcSize, ArcSize );
lv_obj_align(
        pArcFirst
	,	NULL
	,	LV_ALIGN_IN_BOTTOM_MID
	,	( _displayWidth / 12 )
	,	( _displayHeight / 10 )
);

lv_obj_t* pArcSecond  = lv_arc_create( _parentObject, NULL);
lv_obj_add_style( pArcSecond, LV_ARC_PART_BG, &m_checkedPointStyle );
lv_obj_add_style( pArcSecond, LV_ARC_PART_BG, &m_uncheckedPointStyle );

lv_arc_set_angles( pArcSecond, 0, 360 );
lv_obj_set_size( pArcSecond, ArcSize, ArcSize );
lv_obj_align(
		pArcSecond
	,	NULL
	,	LV_ALIGN_IN_BOTTOM_MID
	,	0
	,	-(_displayHeight / 10)
);

lv_obj_t* pArcThird = lv_arc_create( _parentObject, NULL);

lv_arc_set_angles( pArcThird, 0, 360);
lv_obj_set_size( pArcThird, ArcSize, ArcSize);
lv_obj_add_style( pArcThird, LV_ARC_PART_BG, &m_checkedPointStyle);
lv_obj_add_style( pArcThird, LV_ARC_PART_BG, &m_uncheckedPointStyle );
lv_obj_align(
		pArcThird
	,	NULL
	,	LV_ALIGN_IN_BOTTOM_MID
	,	( _displayWidth / 12)
	,	-( _displayHeight / 10)
);
```

ValentiWorkLearning, 2020-05-06:
@embeddedt Thanks for your tip! I'll implement it as soon as possible. Also, extra pixels are available in Simulator:
[image]

embeddedt, 2020-05-06:
Hmm; I only saw two circles on my board. I'll have to look at that more closely.

embeddedt, 2020-05-06:
Also, as a clarification for my tip, styles automatically inherit properties from a lower precedence state if they're not found in the current state. So your style code for this code sample can be reduced to the following:

```c
lv_style_set_border_color(&m_pointStyle, LV_STATE_DEFAULT, MainLightColor);
lv_style_set_bg_color(&m_pointStyle, LV_STATE_DEFAULT, MainLightColor);
lv_style_set_line_color(&m_pointStyle, LV_STATE_DEFAULT, MainLightColor);
lv_style_set_line_width(&m_pointStyle, LV_STATE_DEFAULT, 10);
lv_style_set_radius(&m_pointStyle, LV_STATE_DEFAULT, LV_RADIUS_CIRCLE);
lv_style_set_bg_color(&m_pointStyle, LV_STATE_CHECKED, MainDarkColor);
lv_style_set_line_color(&m_pointStyle, LV_STATE_CHECKED, MainLightColor);
lv_style_set_line_width(&m_pointStyle, LV_STATE_CHECKED, 2);
```

`bg_color`, `line_color`, and `line_width` are the only properties that get changed, so only they need to be set for the checked state.

kisvegabor, 2020-05-06:
@maxpromer Great! Thank you!

@ValentiWorkLearning I see the issue in the simulator and also fond two minor issues in the example code:
1. `_displayWidth` and `_displayHeight` were overflowed. They should be `uint16_t`
2. Offsets should be negative for the first arc align:
```c
  lv_obj_align(
          pArcFirst
      ,   NULL
      ,   LV_ALIGN_IN_BOTTOM_MID
      ,   -( _displayWidth / 12 )
      ,   -( _displayHeight / 10 )
  );
```

I'm investigating the pixel error now. 

Just out of curiosity, why have you used arcs to draw circles? A simple `lv_obj` with `LV_RADIUS_CIRCLE` should do the job as well.

ValentiWorkLearning, 2020-05-06:
@kisvegabor overflow occurs only in @embeddedt example because he assigned  to width and height of a display value that equals 480 pixels. Suppose, that's the reason of why he sees only two circles.

Also about drawing with arcs. I have no idea, why I've decided to to this through using arcs instead of lv_obj_t. Anyway, thanks for your advice:)
@embeddedt thanks for your tips. Take a look on code lines, where you assigned display width and height. Probably it causes your error.

embeddedt, 2020-05-06:
Thanks. I changed it to `uint16_t` and now I see three circles as expected. I also see the extra pixels (once I changed my display background so the contrast was obvious).

kisvegabor, 2020-05-06:
I found the bug and fixed it in 27621e09

The pixel error occurred because the line width was larger than the radius. Now it's limited.

C47D, 2020-05-06:
Hi, have you consider having an lv_config "freeze"? Just realized that the lv_port_esp32 lv_conf file is pretty old and wanted to update it but I don't know if there will be further changes.

embeddedt, 2020-05-06:
I don't expect it to change significantly, since we aren't adding any new features. In any case, if you use a tool like [Meld](https://meldmerge.org/), it usually only takes a couple of seconds to apply any changes.

maxpromer, 2020-05-07:
Plz update [Logging doc](https://docs.littlevgl.com/v7/en/html/porting/log.html) because example `void my_log_cb(lv_log_level_t level, const char * file, int line, const char * dsc)` is incorrect.

Now, log callback needs "Function name" parameter.

https://github.com/littlevgl/lvgl/blob/bc994c60eee740f613f7ebe4822c4bf75570e603/src/lv_misc/lv_log.h#L47-L50

ValentiWorkLearning, 2020-05-07:
Implemntation for logger for LVGL can be as:
https://github.com/ValentiWorkLearning/GradWork/blob/855db69f8e4bfcfe78cac980bc0f77b72702ca8a/Firmware/Firmware/src/graphics/gs_lvgl_service.cpp#L38

maxpromer, 2020-05-07:
Now, i fork lv_port_maixduino and create a `dev-7.0` branch. I test on Sipeed M1 dock suit it work.

[image]

embeddedt, 2020-05-07:
@maxpromer I've fixed the documentation [here](https://github.com/littlevgl/docs/commit/2a4234ae3b61d87d2b2b796aed10f5d173aa19a3); the changes will be visible whenever the docs are rebuilt next.

kisvegabor, 2020-05-08:
@maxpromer 
Great! Thank you!

ValentiWorkLearning, 2020-05-08:
@kisvegabor I've tried to launch my project on target ARM board with Nordic MCU. Unfortunately, I got error while LVGL was trying to allocate a memory chunk for decompressed font internally. After several memory size increasing I tried to profile my firmware on PC. After that I found, that now LVGL7-0 needs about 100 kilobytes buffer for lv_mem_alloc function. Moreover, this memory consumption doesn't visible for user while using memory-monitor, because it allocates and deallocates   in short period of time( as you mentioned only for display loop). 

So, I increased memory pool to #  define LV_MEM_SIZE    (128U * 1024U) for working on Desktop, but for small-footprint memory device( Nordic 52832 with 512 kilobytes of FLASH and 64 kilobytes of RAM) it's impossible to launch and test LVGLv7.0. 

Do you have any ideas, how to deal with this behaviour of font decompressing memory consumption?

ValentiWorkLearning, 2020-05-08:
@kisvegabor Also, as you asked me about creating a page for Nordic MCUs on LVGL website- I would like to do this and support it. As I explored, LVGL article format is Markdown, suppose it will be easy to deal with this. Which kind of topics are preferable for LVGL? Can I choose either C or C++ language for this examples? Can I use my self-developed wrappers for SPI bus and display or I should consider of using only Nordic's SDK tools?

kisvegabor, 2020-05-08:
@ValentiWorkLearning 
**Memory usage**
That's strange. It should be much less than 100kB especially for a simple UI that you have shown earlier. For example, `lv_demo_widgets` runs with 24KB RAM on STM32F7 and it has a lot of widgets. 

Can you check what causes so large memory consumption by adding this to `lv_mem_alloc`:
```c
if(size > 1024) { //try some other values too
   NRF_LOG("large malloc");    //breakpoint here
}
```

**Nordic**
Regarding the docs I imagined a description to get started. E.g.:
- Which Nordic modules are suitable?
- Are there recommended dev boards? How to get or put together a basic setup?
- How to connect a display and how to configure it? (SPI I suppose)
- What can you expect from a nordic chip? (e.g. regarding FPS)
- How to integrate lvgl in a nordic project? (build system and porting)

If you are interested in a longer-term contribution, we can open a new repo for a nordic chip and we can upload there an example project. What do you think about it?

C would be preferred for compatibility. Are your wrappers using the SDK's functions or the peripheries directly?

ValentiWorkLearning, 2020-05-08:
@kisvegabor I investigated a little bit the memory consumption and found something that we need for further troubleshooting. 

Modified variant of lv_mem_alloc:
```cpp
void * lv_mem_alloc(size_t size)
{
    //if (size > 1024) { //try some other values too
    //    printf("Requested more than 1024 kilobytes from memory pool\n");
    //}
    if(size == 0) {
        return &zero_mem;
    }

#ifdef LV_ARCH_64
    /*Round the size up to 8*/
    if(size & 0x7) {
        size = size & (~0x7);
        size += 8;
    }
#else
    /*Round the size up to 4*/
    if(size & 0x3) {
        size = size & (~0x3);
        size += 4;
    }
#endif
    void * alloc = NULL;

#if LV_MEM_CUSTOM == 0
    /*Use the built-in allocators*/
    lv_mem_ent_t * e = NULL;

    /* Search for a appropriate entry*/
    do {
        /* Get the next entry*/
        e = ent_get_next(e);

        /*If there is next entry then try to allocate there*/
        if(e != NULL) {
            alloc = ent_alloc(e, size);
        }
        /* End if there is not next entry OR the alloc. is successful*/
    } while(e != NULL && alloc == NULL);
    
#else
    /*Use custom, user defined malloc function*/
#if LV_ENABLE_GC == 1 /*gc must not include header*/
    alloc = LV_MEM_CUSTOM_ALLOC(size);
#else                 /* LV_ENABLE_GC */
    /*Allocate a header too to store the size*/
    alloc = LV_MEM_CUSTOM_ALLOC(size + sizeof(lv_mem_header_t));
    if(alloc != NULL) {
        ((lv_mem_ent_t *)alloc)->header.s.d_size = size;
        ((lv_mem_ent_t *)alloc)->header.s.used   = 1;

        alloc = &((lv_mem_ent_t *)alloc)->first_data;
    }
#endif                /* LV_ENABLE_GC */
#endif                /* LV_MEM_CUSTOM */

#if LV_MEM_ADD_JUNK
    if(alloc != NULL) lv_memset(alloc, 0xaa, size);
#endif

    lv_mem_monitor_t mon;
    lv_mem_monitor( &mon);

    printf("lv_mem_alloc: used: %6d (%3d %%), frag: %3d %%, biggest free: %6d\n", (int)mon.total_size - mon.free_size,
        mon.used_pct,
        mon.frag_pct,
        (int)mon.free_biggest_size);

    const int usedMemory = mon.total_size - mon.free_size;
    assert( !(usedMemory > 40000));
    if(alloc == NULL) LV_LOG_WARN("Couldn't allocate memory");

    return alloc;
}
```

lv_mem_free modified almost closer to lv_mem_alloc, just simple added logging of current heap usage.

After that, I got the following result:

[image]

While running, library can use more than 40 kilobytes of RAM, but this usage is invisible for calls from
```cpp
        auto memoryMonitorTask = cbc::obtain_connector(
            [this]( lv_task_t* _param )
            {
                memoryMonitor( _param );
            }
        );

        lv_task_create(
                memoryMonitorTask
            ,   1 // don't care, output will be normal because this show only usage when a display refreshed
            ,   LV_TASK_PRIO_MID
            ,   nullptr
        );
```
memory monitor task because this memory allocates and deallocates during display-loop.

following screenshots are attached as output of abnormal memory usage.

[image]
[image]

The failed assertion callstack is:
[image]

Thanks for your attention.

ValentiWorkLearning, 2020-05-08:
Attached screenshot of memory cunsumption for v7 and v6 LVGL.
Maybe it will be helpful.
[image]

persello, 2020-05-08:
Hello, I have noticed a small detail when using a list with the mono theme: when a button separator is aligned with the bottom of the list, the list’s rounded borders are ignored and the bottom of the list appears without any rounding, as seen in this picture:

[image]

```c
void ds_gui_main_menu(lv_indev_t *indev)
{
    lv_group_t *group = lv_group_create();
    lv_indev_set_group(indev, group);

    lv_obj_t * list = lv_list_create(lv_scr_act(), NULL);
    lv_obj_set_size(list, 128, 64);
    lv_obj_align(list, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_group_add_obj(group, list);

    lv_group_focus_obj(list);

    lv_list_add_btn(list, NULL, "Button 1");
    lv_list_add_btn(list, NULL, "Button 2");
    lv_list_add_btn(list, NULL, "Button 3");
    lv_list_add_btn(list, NULL, "Button 4");
    lv_list_add_btn(list, NULL, "Button 5");
}
```

When it’s aligned with the top this does not happen.

Also, I have noticed a strange behavior when scrolling this list with an encoder, but I prefer to first implement a debounce system before trying to understand its repeatability, if the issue persists, I will let you know.

embeddedt, 2020-05-08:
@persello Can you try enabling `clip_corner` (I think it's a style property) and see if the issue persists?

persello, 2020-05-08:
@embeddedt I’m not sure about how properly setting the clip_corner variable,  here is what I tried:

```
void ds_gui_main_menu(lv_indev_t *indev)
{
    lv_group_t *group = lv_group_create();
    lv_indev_set_group(indev, group);

    lv_obj_t * list = lv_list_create(lv_scr_act(), NULL);
    lv_obj_set_size(list, 128, 64);
    lv_obj_align(list, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_obj_set_style_local_clip_corner(list, LV_LIST_PART_SCRL, LV_STATE_FOCUSED, true);
    lv_group_add_obj(group, list);

    lv_group_focus_obj(list);

    lv_list_add_btn(list, NULL, "Button 1");
    lv_list_add_btn(list, NULL, "Button 2");
    lv_list_add_btn(list, NULL, "Button 3");
    lv_list_add_btn(list, NULL, "Button 4");
    lv_list_add_btn(list, NULL, "Button 5");
}
```

The behavior remains the same.

Regarding the top part, _sometimes_ I can see its corners briefly flashing, so this could be happening even there, but it can’t be seen because there is never an exact alignment. This is just an idea, because it seems unlikely for the divider not to be aligned with the top, just by looking at it (sorry, didn’t count the pixels, quite difficult on an oled screen 😊).

embeddedt, 2020-05-08:
Try with `LV_STATE_DEFAULT`, as you want this to be enabled in all cases.

persello, 2020-05-08:
@embeddedt Same as before, even with `LV_STATE_DEFAULT`.

kisvegabor, 2020-05-11:
@ValentiWorkLearning Thank you for the debugging. I found a but in label drawing which allocated much more memory then it ought to be. Besides I optimized memory usage in general. How it work now?

kisvegabor, 2020-05-11:
@persello 
The corner of the background should be clipped like this:
```c
lv_obj_set_style_local_clip_corner(list, LV_LIST_PART_BG, LV_STATE_DEFAULT, true);
```

I also improved the mono theme so it clips the corner by default.

ValentiWorkLearning, 2020-05-11:
@kisvegabor thanks for your fix. Updated lvgl version with your new commits and got the access violation:
[image]
It happens while exectuting lv_font_clean_up_fmt_txt and de-allocating memory, that was alocated for decompr_buf.
```cpp

LV_ATTRIBUTE_FAST_MEM void lv_memset(void * dst, uint8_t v, size_t len)
{

    uint8_t * d8 = (uint8_t *) dst;

    uintptr_t d_align = (lv_uintptr_t) d8 & ALIGN_MASK;

    /*Make the address aligned*/
    if(d_align) {
        d_align = ALIGN_MASK + 1 - d_align;
        while(d_align && len) {
            *d8 = v;
            d8++;
            len--;
            d_align--;
        }
    }

    uint32_t v32 = v + (v << 8) + (v << 16) + (v << 24);

    uint32_t * d32 = (uint32_t *)d8;

    while(len > 32) {
        SET32(v32); // access violation here
        SET32(v32);
        SET32(v32);
        SET32(v32);
        SET32(v32);
        SET32(v32);
        SET32(v32);
        SET32(v32);
        len -= 32;
    }

    while(len > 4) {
        SET32(v32);
        len -= 4;
    }

    d8 = (uint8_t *)d32;
    while(len) {
        *d8 = v;
        d8++;
        len--;
    }
}

```

[42 more replies]

closed 2020-05-28 by kisvegabor