lvgl/lvgl #1484: Merged latest  Dev 7.0 today and now have compilation errors...

pete-pjb, 2020-05-01:
**Describe the bug**

All was well with yesterdays merge but today I can no longer compile the library,  I suspect it's something to do with the LV_COLOR_SCREEN_TRANSP define being set to 1.  Although I have always used Dev-7.0 with this enabled previously....

I would have taken a look at this my self but I am completely swamped at the moment with hardware design.  I had a quick look and it seems there are a number of macros defined which contain the variable names which are undeclared if this helps.

Here is the output from the compiler console...

**ERRORS:**
Description	Resource	Path	Location	Type
'disp_buf_tmp' undeclared (first use in this function); did you mean 'disp_buf'?	lv_draw_blend.c	/lvgl/src/lv_draw	line 78	C/C++ Problem
'disp_buf_tmp' undeclared (first use in this function); did you mean 'disp_buf'?	lv_draw_blend.c	/lvgl/src/lv_draw	line 95	C/C++ Problem
'disp' undeclared (first use in this function); did you mean 'disp_w'?	lv_draw_blend.c	/lvgl/src/lv_draw	line 363	C/C++ Problem
'map_buf_tmp' undeclared (first use in this function); did you mean 'map_buf'?	lv_draw_blend.c	/lvgl/src/lv_draw	line 95	C/C++ Problem
make: *** [src/lv_draw/subdir.mk:57: src/lv_draw/lv_draw_blend.o] Error 1	lvgl		 	C/C++ Problem
make: *** Waiting for unfinished jobs....	lvgl		 	C/C++ Problem

**WARNINGS:**
Description	Resource	Path	Location	Type
'lv_ap_get_char_index' declared 'static' but never defined [-Wunused-function]	lv_txt_ap.c	/lvgl/src/lv_misc	line 27	C/C++ Problem
unused variable 'opa_composed' [-Wunused-variable]	lv_draw_blend.c	/lvgl/src/lv_draw	line 672	C/C++ Problem

**INFOS:**
Description	Resource	Path	Location	Type
each undeclared identifier is reported only once for each function it appears in	lv_draw_blend.c	/lvgl/src/lv_draw	line 363	C/C++ Problem
in expansion of macro 'FILL_NORMAL_MASK_PX_SCR_TRANSP'	lv_draw_blend.c	/lvgl/src/lv_draw	line 414	C/C++ Problem
in expansion of macro 'MAP_NORMAL_MASK_PX_SCR_TRANSP'	lv_draw_blend.c	/lvgl/src/lv_draw	line 754	C/C++ Problem

**To Reproduce**

Attempt to compile library...

**Expected behavior**

Library to build okay.

 Here is my lv_conf.h as I suspect it is probably something I am doing that is breaking it....

```c
/**
 * @file lv_conf.h
 *
 */

/*
 * COPY THIS FILE AS `lv_conf.h` NEXT TO the `lvgl` FOLDER
 */

#if 1 /*Set it to "1" to enable content*/

#ifndef LV_CONF_H
#define LV_CONF_H
/* clang-format off */

#include <stdint.h>
#define LV_VGA_DDR_DMA_BASE		0x0F000000		// This is the DDR address the DMA hardware fetches VGA pixel data from (32MB is reserved)
/*====================
   Graphical settings
 *====================*/

/* Maximal horizontal and vertical resolution to support by the library.*/
#define LV_HOR_RES_MAX          (1024)
#define LV_VER_RES_MAX          (768)

/* Color depth:
 * - 1:  1 byte per pixel
 * - 8:  RGB233
 * - 16: RGB565
 * - 32: ARGB8888
 */
[... 651 lines cut ...]
/*Tileview (dependencies: lv_page) */
#define LV_USE_TILEVIEW     1
#if LV_USE_TILEVIEW
/*Time of slide animation [ms] (0: no animation)*/
#  define LV_TILEVIEW_DEF_ANIM_TIME   300
#endif

/*Window (dependencies: lv_cont, lv_btn, lv_label, lv_img, lv_page)*/
#define LV_USE_WIN      1

/*==================
 * Non-user section
 *==================*/

#if defined(_MSC_VER) && !defined(_CRT_SECURE_NO_WARNINGS)    /* Disable warnings for Visual Studio*/
#  define _CRT_SECURE_NO_WARNINGS
#endif

/*--END OF LV_CONF_H--*/

#endif /*LV_CONF_H*/

#endif /*End of "Content enable"*/

```

kisvegabor, 2020-05-01:
Hi Pete,

I've just pushed a fix! Is it ok now?

pete-pjb, 2020-05-01:
Hi Gabor,

Many thanks for the speedy response, there is an improvement but, unfortunately we still have an error as follows:

Description	Resource	Path	Location	Type
'disp' undeclared (first use in this function); did you mean 'disp_w'?	lv_draw_blend.c	/lvgl/src/lv_draw	line 363	C/C++ Problem
make: *** [src/lv_draw/subdir.mk:57: src/lv_draw/lv_draw_blend.o] Error 1	lvgl		 	C/C++ 
Problem
make: *** Waiting for unfinished jobs....	lvgl		 	C/C++ Problem
'lv_ap_get_char_index' declared 'static' but never defined [-Wunused-function]	lv_txt_ap.c	/lvgl/src/lv_misc	line 27	C/C++ Problem
each undeclared identifier is reported only once for each function it appears in	lv_draw_blend.c	/lvgl/src/lv_draw	line 363	C/C++ Problem

Kind Regards,

Pete

kisvegabor, 2020-05-02:
I've pushed a fix again. 1e59100f

pete-pjb, 2020-05-02:
Hi Gabor,

Okay it is now compiling okay with no errors or warnings, but I have just a black screen when I start the  application, the code is running okay with FreeRTOS tasks executing as expected on both cores of my CPU but nothing is actually getting to the screen....  I am not able to debug this to give you any more information right now.  I will be able to take another look on Monday morning if that is helpful?

Have a good weekend....

Kind Regards,

Pete

embeddedt, 2020-05-02:
For the record, I tried the latest `dev-7.0` snapshot and it seems to work without issues on my board, so it's something localized to @pete-pjb's setup.

pete-pjb, 2020-05-04:
Hi @kisvegabor ,

Hope you had a good weekend!

I am beginning to look at the issue in my system, it is tough going as I really don't know lvgl library  code that well, but what I can see so far is the screen data is being written to the VGA buffers but something is coming along after that operation and resetting the values in the buffer back to 0x00000000.  I don't know if that is a clue?  In the mean time I will continue to see if I can work out the code flow...

Kind Regards,

Pete

pete-pjb, 2020-05-04:
Hi again @kisvegabor ,

It seems if I change my lv_conf.h file as below:

#define LV_COLOR_SCREEN_TRANSP    0

All is well...

The transparency works with this setting, I have a background bitmap with text over on my start-up screen, I also have semi-transparent message boxes that pop up and disappear over my main GUI and the transparency appears to be correct for these.  So I am happy for now :smiley: but I am uncertain as to whether there is some kind of issue here.  The transparency define had always been 1 in my previous builds so I'm am not sure what has happened here...

Kind Regards,

Pete

kisvegabor, 2020-05-05:
That's strange. It works for me with `LV_COLOR_SCREEN_TRANSP`. 
Note that it should be enabled only if the screen is transparent (e.g. an OSD menu of a TV) because it makes rendering slower.

pete-pjb, 2020-05-05:
I am happy to close the issue if you are....   I have everything working as required without the parameter enabled so it's all good here.  One day I might have some time to delve into the library and see how it works!   Being over 50 years old I traditionally have worked with VT100 console based embedded systems until now, so I have little knowledge of the inner workings of rendering graphical screens etc.  Hence my use of your excellent library! :smiley: 

Kind Regards,

Pete

closed 2020-05-05 by pete-pjb

kisvegabor, 2020-05-06:
Okay, we will the bug sooner or later :slightly_smiling_face: 

> I traditionally have worked with VT100 console based embedded systems until now

Ahh, a piece of history. Even my parent were children when VT100 appeared :smile: